JMRI/JMRI #7311: Signal Mast resetting to "lit" does not relight the signal

Fire-Power, 2019-08-12:
When you set a signal mast to dark by unticking "lit" it correctly extinguishes any lit aspect.
But when you tick "lit" in the signal mast table nothing happens.
If you then reselect current aspect or another aspect  then the correct aspect is sent.

I would have expected that upon setting "lit" the current aspect should be sent.

I have attached a worked example and annotated CBUS message log.

[litsignalmast.xml.txt](https://github.com/JMRI/JMRI/files/3494518/litsignalmast.xml.txt)
[litsigcbuslog.txt](https://github.com/JMRI/JMRI/files/3494519/litsigcbuslog.txt)

Running version 14.17.3

mstevetodd, 2019-08-13:
Note: I checked this on my Digitrax SE8C-based signalmasts, and enabling "Lit" does show the currently-selected aspect immediately, so perhaps the issue is in the CBUS implementation.

bobjacobsen, 2019-08-13:
Could you do a test for me please?

Open the Turnout Table next to the Signal Head Table so you can see changes in Turnouts as you change the signal.  Show some aspects and write down the Turnout Closed/Throw settings.  

When you untick ‘lit’, that should change the Closed/Thrown state of one or more turnouts. Does it?  Which ones?

And when you re-tick “lit”, do any of those change? Does it go back ot the original pattern?

Thanks.  This will help track down where the problem is.

Fire-Power, 2019-08-13:
Hi Bob
When I untick "lit" is sets all the turnout states to closed.
When I re-tick "lit" all the turnout states stay closed.
The original pattern is not restored.
Copies of tables at each stage attached.

[initial_signal_head.pdf](https://github.com/JMRI/JMRI/files/3496261/initial_signal_head.pdf)
[inital_lit_turnouts.pdf](https://github.com/JMRI/JMRI/files/3496262/inital_lit_turnouts.pdf)
[unlit_signal_head.pdf](https://github.com/JMRI/JMRI/files/3496263/unlit_signal_head.pdf)
[unlit_turnouts.pdf](https://github.com/JMRI/JMRI/files/3496266/unlit_turnouts.pdf)
[relit_signal_head.pdf](https://github.com/JMRI/JMRI/files/3496267/relit_signal_head.pdf)
[relit_turnouts.pdf](https://github.com/JMRI/JMRI/files/3496268/relit_turnouts.pdf)

icklesteve, 2019-08-13:
The Matrix SignalMast was not reactivating as the super lit flag was still false.
Setting the new status before the recalc. seems to resolve, see #7316

icklesteve, 2019-08-14:
Hi Nigel,
I forgot to say thanks for the xml file in initial post, definitely helped to focus issue.

The fix attempt is from package 3417 at http://builds.jmri.org/jenkins/job/Development/job/Packages/
so please let us know how you get on,
Steve

Fire-Power, 2019-08-14:
Many thanks Steve & Bob.
It is working as expected now.
Nigel

closed 2019-08-14 by Fire-Power