JMRI/JMRI #7313: Print not working from Linux

DaveTur, 2019-08-13:
JMRI DecoderPro not printing on Raspbian

I am using JMRI DecoderPro version 4.17.3+R12d2ded on a Raspberry Pi 3 running Raspbian GNU/Linux 9 (stretch) and Java 1.8.0_222 (en-AU).  This problem was initially found using JMRI 4.17.2 and Java 1.8.0_65.  

I have tried to print (a) all items from "Program" for a decoder and (b) Roster summary, to both a physical network printer and a PDF file without success, no entries being added to the CUPS log files.  Printing to these same destinations do work when using other applications (e.g. leafpad and Chrome).

Printing of these items works OK with the same version of JMRI on Windows 10 (64 bit) with Java 1.8.0_221.  I have not tried printing from other JMRI applications such as PanelPro.

rhwood, 2019-08-16:
Java relies on the CUPS printer service running in Raspbian, and will not show any printers unless CUPS has at least one printer installed and configured. On Raspbian Buster, CUPS is not installed by default, and the native programs are using some other software or capability to print.

To see if CUPS is installed and configured, run `systemctl status cups` (on Buster, I don't know if this is the same or different on Stretch). If installed and running, that command will return something like:
```
● cups.service - CUPS Scheduler
   Loaded: loaded (/lib/systemd/system/cups.service; enabled; vendor preset: ena
   Active: active (running) since Fri 2019-08-16 07:11:21 EDT; 23min ago
     Docs: man:cupsd(8)
 Main PID: 10059 (cupsd)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/cups.service
           └─10059 /usr/sbin/cupsd -l

Aug 16 07:11:21 raspberry systemd[1]: Started CUPS Scheduler.
```

You can also use `lpstat -p -d` to show the usable printers, with output like:
```
printer Brother-HL-3170CDW-series is idle.  enabled since Fri 16 Aug 2019 07:43:52 AM EDT
printer PDF is idle.  enabled since Fri 16 Aug 2019 07:10:48 AM EDT
system default destination: PDF
```

DaveTur, 2019-08-20:
Hi Randal,
Thanks for looking at my printing problem.  I confirm that CUPS is up and running on my RPi3, as follows:-

command:  pi@raspberrypi:~ $ systemctl status cups

response: cups.service - CUPS Scheduler
   Loaded: loaded (/lib/systemd/system/cups.service; enabled; vendor preset: ena
   Active: active (running) since Mon 2019-08-19 23:35:45 AEST; 20h ago
     Docs: man:cupsd(8)
 Main PID: 370 (cupsd)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/cups.service
           ├─370 /usr/sbin/cupsd -l
           └─602 /usr/lib/cups/notifier/dbus dbus://

Aug 19 23:35:45 raspberrypi systemd[1]: Started CUPS Scheduler.

CUPS sees my 2 physical and 1 file printer as :-

command: pi@raspberrypi:~ $ lpstat -p -d

response: printer BrotherDaveOffice is idle.  enabled since Mon 19 Aug 2019 23:35:56 AEST
printer Canon_MG8200_series is idle.  enabled since Sun 11 Aug 2019 23:12:30 AEST
printer PDF is idle.  enabled since Tue 13 Aug 2019 12:50:20 AEST
system default destination: PDF

I can browse to my CUPS admin panel by entering "localhost:631" into Chrome.

Looking at my CUPS configuration file /etc/cups/cups-files.conf shows the only un-commented lines as -
SystemGroup lpadmin
AccessLog /var/log/cups/access_log
ErrorLog /var/log/cups/error_log
PageLog /var/log/cups/page_log

When printing a file to PDF from the leafpad text editor -
    a)  The requested file is created as ~/PDF/Leafpad_job__1.pdf
    b)  in folder /var/log/cups -
         i)  The following 2 lines are appended to access_log
             localhost - root [20/Aug/2019:19:34:21 +1000] "POST /admin/ HTTP/1.1" 200 133 Resume-Printer successful-ok
             localhost - - [20/Aug/2019:20:11:41 +1000] "POST /printers/PDF HTTP/1.1" 200 14630 Print-Job successful-ok
          ii)  The following 2 lines are appended to cups-pdf_log
               Tue Aug 20 20:11:42 2019  [STATUS] ***Experimental Option: DecodeHexStrings
               Tue Aug 20 20:11:44 2019  [STATUS] PDF creation successfully finished (pi)

When attempting to print my DecoderPro roster (using File ->  Print Roster -> Summary, to 'printer' "PDF", 'print to file' not ticked) I see:-
    a) No new file created in folder ~/PDF
    b) No log files changed in folder /var/log/cups.

When attempting to print my DecoderPro roster with 'print to file' ticked, I can create ~/out.ps, that is opened by the default application qpdtview and does list a summary of my roster.  Printing this roster from qpdtview to does produce ~/PDF/out.pdf, that is again opened by the default application qpdfview.  I can also print out.ps to a physical network printer (a Brother HL-2170W) from qpdfview.

Only the 2 print operations from qpdtview are shown in the CUPS log files, neither of the 2 print attempts from DecoderPro are logged by CUPS.

I consider getting a print (both to PDF and to a physical printer) via the out.ps file using qpdtview as a work-around, rather than satisfactory 'normal' printing.

Please feel free to ask me for any other tests,
Regards
Dave

rhwood, 2019-08-22:
Unfortunately, since you just described a scenario that works for me, I don't have any other concrete suggestions.

That said, you may want to make sure you do not also have `lpr` or `lprng` also installed or running as services, since these packages conflict with `cups`.

DaveTur, 2019-08-26:
Hi Randall,

 

Thanks for the suggestion of checking for packages lpr and lprange.  

 

I do not have either of these packages installed, as seen by using the commands ‘sudo aptitude show lpr’ and ‘sudo aptitude show lprng’, both reporting State as ‘not installed’.

Also, these are not running, as shown by command ‘sudo ps –A | grep lpr’ returning nothing.

 

As I previously reported, I can print normally using CUPS from applications other than JMRI, having previously tried leafpad and Chrome, and since  successfully printed from LibreOffice Writer and CUPS Admin.

 

With what combination of JMRI, Java and Raspbian can you successfully print from DecoderPro ?

 

Best regards

Dave Turner

closed 2021-04-14 by stale[bot]

pjsg, 2023-03-05:
I had the same problem, but I solved it with

`sudo apt install lpr cups-bsd`

It turns out that JMRI printing ends up using `lpr` to print the output. An easy test is to see if you can do `lpr -PPDF somefile` and have it generate output in the `~/PDF` directory.