JMRI/JMRI #7318: NPE in passing tests

rhwood, 2019-08-13:
Passing tests (such as [Travis CI Job 571269179](https://api.travis-ci.org/v3/job/571269179/log.txt)) have ~16 NullPointerExceptions logged during those tests:

```
[INFO] Running jmri.jmrix.loconet.locomon.LocoMonPaneTest
WARN  - passing to xmit: unexpected exception:  [LnSensorUpdateThread] jmri.jmrix.loconet.LnPacketizer.sendLocoNetMessage()
java.lang.NullPointerException
	at jmri.jmrix.loconet.LnPacketizer.sendLocoNetMessage(LnPacketizer.java:118)
	at jmri.jmrix.loconet.LnSensorManager$LnSensorUpdateThread.run(LnSensorManager.java:285)
WARN  - passing to xmit: unexpected exception:  [LnSensorUpdateThread] jmri.jmrix.loconet.LnPacketizer.sendLocoNetMessage()
java.lang.NullPointerException
	at jmri.jmrix.loconet.LnPacketizer.sendLocoNetMessage(LnPacketizer.java:118)
	at jmri.jmrix.loconet.LnSensorManager$LnSensorUpdateThread.run(LnSensorManager.java:285)
WARN  - passing to xmit: unexpected exception:  [LnSensorUpdateThread] jmri.jmrix.loconet.LnPacketizer.sendLocoNetMessage()
java.lang.NullPointerException
	at jmri.jmrix.loconet.LnPacketizer.sendLocoNetMessage(LnPacketizer.java:118)
	at jmri.jmrix.loconet.LnSensorManager$LnSensorUpdateThread.run(LnSensorManager.java:285)
[INFO] Tests run: 17, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 3.383 s - in jmri.jmrix.loconet.locomon.LocoMonPaneTest
```
```
[INFO] Running jmri.jmrix.dccpp.DCCppThrottleTest
WARN  - notify: During dispatch to jmri.jmrix.sprog.update.SprogIIUpdateFrame[frame512,0,0,242x65,invalid,hidden,layout=java.awt.BorderLayout,title=SPROGII/SPROG 3/Pi-SPROG Firmware Update,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,0,242x65,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
Exception java.lang.NullPointerException [AWT-EventQueue-1] jmrix.sprog.update.SprogVersionQuery.notifyVersion()
[INFO] Tests run: 153, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.329 s - in jmri.jmrix.dccpp.DCCppThrottleTest
[INFO] Running jmri.jmrix.dccpp.DCCppTrafficControllerTest
WARN  - notify: During dispatch to jmri.jmrix.sprog.update.SprogIIUpdateFrame[frame512,0,0,242x65,invalid,hidden,layout=java.awt.BorderLayout,title=SPROGII/SPROG 3/Pi-SPROG Firmware Update,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,0,242x65,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
Exception java.lang.NullPointerException [AWT-EventQueue-1] jmrix.sprog.update.SprogVersionQuery.notifyVersion()
[INFO] Tests run: 6, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.022 s - in jmri.jmrix.dccpp.DCCppTrafficControllerTest
[INFO] Running jmri.jmrix.dccpp.DCCppPacketizerTest
WARN  - notify: During dispatch to jmri.jmrix.sprog.update.SprogIIUpdateFrame[frame512,0,0,242x65,invalid,hidden,layout=java.awt.BorderLayout,title=SPROGII/SPROG 3/Pi-SPROG Firmware Update,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,0,242x65,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
Exception java.lang.NullPointerException [AWT-EventQueue-1] jmrix.sprog.update.SprogVersionQuery.notifyVersion()
[INFO] Tests run: 8, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 1.084 s - in jmri.jmrix.dccpp.DCCppPacketizerTest
```
```
[INFO] Running jmri.jmrix.pi.RaspberryPiTurnoutTest
Exception in thread "Wait for Preferences to be loaded" java.lang.NullPointerException: Required nonnull default for jmri.jmrit.roster.RosterConfigManager does not exist.
	at jmri.InstanceManager.getDefault(InstanceManager.java:206)
	at jmri.jmrit.roster.Roster.getRoster(Roster.java:205)
	at jmri.jmrit.roster.Roster.getDefault(Roster.java:194)
	at jmri.jmrix.ecos.EcosLocoAddressManager.loadData(EcosLocoAddressManager.java:220)
	at jmri.jmrix.ecos.EcosLocoAddressManager.access$400(EcosLocoAddressManager.java:34)
	at jmri.jmrix.ecos.EcosLocoAddressManager$WaitPrefLoad.run(EcosLocoAddressManager.java:839)
	at java.lang.Thread.run(Thread.java:748)
Exception in thread "Wait for Preferences to be loaded" java.lang.NullPointerException: Required nonnull default for jmri.jmrit.roster.RosterConfigManager does not exist.
	at jmri.InstanceManager.getDefault(InstanceManager.java:206)
	at jmri.jmrit.roster.Roster.getRoster(Roster.java:205)
	at jmri.jmrit.roster.Roster.getDefault(Roster.java:194)
	at jmri.jmrix.ecos.EcosLocoAddressManager.loadData(EcosLocoAddressManager.java:220)
	at jmri.jmrix.ecos.EcosLocoAddressManager.access$400(EcosLocoAddressManager.java:34)
	at jmri.jmrix.ecos.EcosLocoAddressManager$WaitPrefLoad.run(EcosLocoAddressManager.java:839)
	at java.lang.Thread.run(Thread.java:748)
Exception in thread "Wait for Preferences to be loaded" [INFO] Tests run: 16, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.072 s - in jmri.jmrix.pi.RaspberryPiTurnoutTest
[INFO] Running jmri.jmrix.pi.RaspberryPiConnectionTypeListTest
java.lang.NullPointerException: Required nonnull default for jmri.jmrit.roster.RosterConfigManager does not exist.
	at jmri.InstanceManager.getDefault(InstanceManager.java:206)
	at jmri.jmrit.roster.Roster.getRoster(Roster.java:205)
	at jmri.jmrit.roster.Roster.getDefault(Roster.java:194)
	at jmri.jmrix.ecos.EcosLocoAddressManager.loadData(EcosLocoAddressManager.java:220)
	at jmri.jmrix.ecos.EcosLocoAddressManager.access$400(EcosLocoAddressManager.java:34)
	at jmri.jmrix.ecos.EcosLocoAddressManager$WaitPrefLoad.run(EcosLocoAddressManager.java:839)
	at java.lang.Thread.run(Thread.java:748)
Exception in thread "Wait for Preferences to be loaded" java.lang.NullPointerException: Required nonnull default for jmri.jmrit.roster.RosterConfigManager does not exist.
	at jmri.InstanceManager.getDefault(InstanceManager.java:206)
	at jmri.jmrit.roster.Roster.getRoster(Roster.java:205)
	at jmri.jmrit.roster.Roster.getDefault(Roster.java:194)
	at jmri.jmrix.ecos.EcosLocoAddressManager.loadData(EcosLocoAddressManager.java:220)
	at jmri.jmrix.ecos.EcosLocoAddressManager.access$400(EcosLocoAddressManager.java:34)
	at jmri.jmrix.ecos.EcosLocoAddressManager$WaitPrefLoad.run(EcosLocoAddressManager.java:839)
	at java.lang.Thread.run(Thread.java:748)
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.018 s - in jmri.jmrix.pi.RaspberryPiConnectionTypeListTest
[INFO] Running jmri.jmrix.pi.configurexml.RaspberryPiSensorManagerXmlTest
[... 35 lines cut ...]
[INFO] Running jmri.jmrix.pi.configurexml.RaspberryPiClassMigrationTest
Exception in thread "Wait for Preferences to be loaded" java.lang.NullPointerException: Required nonnull default for jmri.jmrit.roster.RosterConfigManager does not exist.
	at jmri.InstanceManager.getDefault(InstanceManager.java:206)
	at jmri.jmrit.roster.Roster.getRoster(Roster.java:205)
	at jmri.jmrit.roster.Roster.getDefault(Roster.java:194)
	at jmri.jmrix.ecos.EcosLocoAddressManager.loadData(EcosLocoAddressManager.java:220)
	at jmri.jmrix.ecos.EcosLocoAddressManager.access$400(EcosLocoAddressManager.java:34)
	at jmri.jmrix.ecos.EcosLocoAddressManager$WaitPrefLoad.run(EcosLocoAddressManager.java:839)
	at java.lang.Thread.run(Thread.java:748)
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.002 s - in jmri.jmrix.pi.configurexml.RaspberryPiClassMigrationTest
[INFO] Running jmri.jmrix.pi.configurexml.RaspberryPiTurnoutManagerXmlTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0 s - in jmri.jmrix.pi.configurexml.RaspberryPiTurnoutManagerXmlTest
[INFO] Running jmri.jmrix.pi.RaspberryPiAdapterTest
Exception in thread "Wait for Preferences to be loaded" [INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in jmri.jmrix.pi.RaspberryPiAdapterTest
[INFO] Running jmri.jmrix.pi.RaspberryPiSensorTest
java.lang.NullPointerException: Required nonnull default for jmri.jmrit.roster.RosterConfigManager does not exist.
	at jmri.InstanceManager.getDefault(InstanceManager.java:206)
	at jmri.jmrit.roster.Roster.getRoster(Roster.java:205)
	at jmri.jmrit.roster.Roster.getDefault(Roster.java:194)
	at jmri.jmrix.ecos.EcosLocoAddressManager.loadData(EcosLocoAddressManager.java:220)
	at jmri.jmrix.ecos.EcosLocoAddressManager.access$400(EcosLocoAddressManager.java:34)
	at jmri.jmrix.ecos.EcosLocoAddressManager$WaitPrefLoad.run(EcosLocoAddressManager.java:839)
	at java.lang.Thread.run(Thread.java:748)
[WARNING] Tests run: 14, Failures: 0, Errors: 0, Skipped: 1, Time elapsed: 0.089 s - in jmri.jmrix.pi.RaspberryPiSensorTest
```

devel-bobm, 2019-08-13:
The LocoNet-related NPEs are apparantly a result of a long-standing problem in the LocoNet code.  The problem could occur as part of the normal user experience, not just test execution.

P/R #7321 should address the LocoNet-related NPEs reported in this issue.

rhwood, 2019-08-14:
The RaspberryPi tests are showing two things:
- The `EcosLocoAddressManager` creates a thread that can run for up to 100 seconds, but also deliberately overrides the `dispose()` method with an empty "do nothing" implementation, so that when resetting the InstanceManager in tests, no hook is available to kill that thread.
- Some tests in `jmri.jmrix.ecos` need `JUnitUtil.initRosterConfigManager();` added to the `setUp` method for those classes. This will be easy to fix once the first item is addressed.

closed 2021-04-14 by stale[bot]