JMRI/JMRI #7323: Memory system names are arbitrary strings

bobjacobsen, 2019-08-14:
In JMRI 4.17.3, the memory system names `IMTRAINBO1WL` and `IMTRAINBO01WL` (note embedded zero in 2nd) are considered, at least in some places, to be the same.

This was first pointed out in a jmriusers thread via this exception:
```
2019-08-12 10:54:07,897 configurexml.ErrorHandler             ERROR - Unexpected error (Exception) while load(/C:/Users/Railroad/JMRI/My_JMRI_Railroad/CWLayout-StageRight.xml) in adaptor of type jmri.managers.configurexml.DefaultMemoryManagerXml Exception: java.lang.IllegalArgumentException: systemName is already registered. Current system name: IMTRAINBO01WL. New system name: IMTRAINBO1WL
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
java.lang.IllegalArgumentException: systemName is already registered. Current system name: IMTRAINBO01WL. New system name: IMTRAINBO1WL
at jmri.managers.AbstractManager.register(AbstractManager.java:228)
```
and discussed in a [jmridevelopers thread](https://jmri-developers.groups.io/g/jmri/topic/32849329#1861)

@danielb987 investigated this and found:

> AbstractMemory needs to override compareSystemNameSuffix() to do a String compare instead of using a AlphanumComparator as AbstractNamedBean.compareSystemNameSuffix() does.

This might also be true for other NamedBean types that aren't system-specific, i.e. SignalHeads, etc. Those should be checked.  The full list of [AbstractNamedBean subclasses is found in the Javadoc](http://jmri.org/JavaDoc/doc/jmri/implementation/AbstractNamedBean.html).

danielb987, 2019-08-14:
> @rhwood made a design comment (which I agree with):
> 
> > Thank you for finding that. Actually the opposite should occur. NamedBean types that need a specific comparison like that should override a generic comparison that does a simple String comparison.

Although I agree on @rhwood , I think there is a risk in doing that.  For now, the rules are more restrict than they need to be. If we change that and makes the string comparison the default, every bean type that is based on numbers needs to be changed to protect from cases like LT1 and LT01.

The current bug is confusing users but is not a big problem and easy to solve. But if we change the default to string comparison and then forget to change some bean type, we could get a severe bug there that bean manager doesn't protects against duplicates as it should have.

So for safety reasons, and since many (most?) of the bean types are number based, I think it's better to have the AlphanumComparator as the default and that beans that can use arbitrary strings overrides the compareSystemNameSuffix() method.

danielb987, 2019-08-14:
I also suggest adding a new exception `DuplicateSystemNameException`.

In the jmridevelopers thread discussion about this:

>>> So there are two bugs:
>>> - IllegalArgumentException is thrown instead of
>>> NamedBean.BadSystemNameException
>>
>> No. The problem is not that the system name itself is wrong. The problem is that the bean has a system name that is already registered. The managers doesn't throw a BadSystemNameException on duplicate system names. See AbstractManager.register(). This conforms to the javadoc for Manager.register().
>
>BadSystemNameException is a subclass of IllegalArgumentException; since we have API to be specific, we should be.

I think BadSystemNameException should be used only when the system name itself is wrong, for example a LocoNet turnout with the name `LTHello`. For duplicate system names, it would be better to have another exception, for example `DuplicateSystemNameException` which is thrown by `Manager.register()`, `XxxManager.newXxx()`, `ProvidingManager.provide()` and `XxxManager.provideXxx()` methods.

danielb987, 2020-08-19:
Fixed by #7379.

closed 2020-08-19 by danielb987