JMRI/JMRI #7354: RaspberryPiTurnoutTest concurrent modification EX in C/I

devel-bobm, 2019-08-23:
When C/I tests ran on P/R #7350, Travis Headless=true Static=false dumped the "concurrent modificaiton" exception shown below.  This message implies a threading issue of some sort in that test or in the code which it is testing.

```
[ERROR] testAddListener(jmri.jmrix.pi.RaspberryPiTurnoutTest)  Time elapsed: 0.026 s  <<< ERROR!

java.util.ConcurrentModificationException

	at java.util.HashMap$HashIterator.nextNode(HashMap.java:1445)

	at java.util.HashMap$KeyIterator.next(HashMap.java:1469)

	at java.util.AbstractCollection.addAll(AbstractCollection.java:343)

	at java.util.HashSet.<init>(HashSet.java:120)

	at jmri.InstanceManager.clearAll(InstanceManager.java:730)

	at jmri.util.JUnitUtil.resetInstanceManager(JUnitUtil.java:676)

	at jmri.jmrix.pi.RaspberryPiTurnoutTest.setUp(RaspberryPiTurnoutTest.java:33)

	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)

	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)

	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)

	at java.lang.reflect.Method.invoke(Method.java:498)

	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)

	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)

	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)

	at org.junit.internal.runners.statements.RunBefores.evaluate(RunBefores.java:24)

	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:27)

	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)

	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)

	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)

	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)

	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)

	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)

	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)

	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)

	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)

	at org.apache.maven.surefire.junit4.JUnit4Provider.execute(JUnit4Provider.java:365)

	at org.apache.maven.surefire.junit4.JUnit4Provider.executeWithRerun(JUnit4Provider.java:272)

	at org.apache.maven.surefire.junit4.JUnit4Provider.executeTestSet(JUnit4Provider.java:236)

	at org.apache.maven.surefire.junit4.JUnit4Provider.invoke(JUnit4Provider.java:159)

	at org.apache.maven.surefire.booter.ForkedBooter.invokeProviderInSameClassLoader(ForkedBooter.java:386)

	at org.apache.maven.surefire.booter.ForkedBooter.runSuitesInProcess(ForkedBooter.java:323)

	at org.apache.maven.surefire.booter.ForkedBooter.main(ForkedBooter.java:143)
```

rhwood, 2019-08-24:
How does opening an issue introduced by un-merged code in a PR and not commenting on the PR help?

rhwood, 2019-08-24:
Sorry, got the PR this was filed against wrong.

devel-bobm, 2019-08-24:
I should be more clear about this issue:

The C/I run was for a P/R with only some changes to LocoNet functionality, **but the exception was flagged for a Raspberry pi test**.  

A JMRI developer who is interested in the rPI code should take a look at this exception.

pabender, 2019-08-24:
The concurrent modification exception in this case is detected during a reset of the instance manager during test setup.

Typically, this means that some test prior to The test in question has not fully shut down ( it has left threads running ) and is accessing the resources  we are trying to reset.

In other words, this isn’t necessarily  a problem with the test file that the error was caught in, and that makes it much harder to debug.

devel-bobm, 2019-08-25:
Is it safe to assume that it is impossible to debug the failure without having that portion of the log which shows the test being executed immediately above the test which showed the concurrent modification problem?

pabender, 2019-08-25:
> Is it safe to assume that it is impossible to debug the failure without having that portion of the log which shows the test being executed immediately above the test which showed the concurrent modification problem?

Worse.  It’s really hard to debug this without knowing what threads were running at the time of the error.  The immediate preceding test may not be causing this.

devel-bobm, 2019-08-25:
Closed account insufficient debugging information available from original failure.

closed 2019-08-25 by devel-bobm