JMRI/JMRI #7380: Programmer in use exception after failure to read type

DaveTur, 2019-09-01:
Running DecoderPro 4.17.3 under Linux
A "programmer in use" exception is thrown when attempting to re-use the programmer after closing the closing the window due to an error.  I followed the following steps :-

1. New Loco
2. Read type for decoder.  Failed to read type, "Read MFG ID - CV 8" still displayed.
3. Closed "Create New Loco" window using "X"
4. Repeated above steps 3 and 4, got "jmri.ProgrammerException: programmer in use", followed by "Stopped due to error: Timeout talking to command station (306)"

devel-bobm, 2019-09-01:
Dave, 

There is a good chance that this is a system-specific problem.  What DCC system (or programming device) have you configured in your JMRI configuration profile?

DaveTur, 2019-09-04:
Hi Bob,

 

I am using the MERG system.  I was running JMRI on a Raspberry Pi 3 under Raspbian (Stretch).  The Rpi has the MERG CANPiCAP attached, talking to a MERG CANCMD.

 

Regards

Dave

dheap, 2019-09-05:
> @DaveTur 
>1. New Loco
> 2. Read type for decoder. Failed to read type, "Read MFG ID - CV 8" still displayed.
 
This indicates that JMRI still waiting for the command station to respond to the JMRI-initiated Read CV8 request. If you have the JMRI System Log open you will see JMRI repeatedly trying to reestablish communication with the command station.

Note that this is a failure of the command station to respond, not a failure of the locomotive to respond to the command station. This would generate a "No locomotive detected (301)" error and JMRI would terminate the identify request.

> 3. Closed "Create New Loco" window using "X"

This doesn't terminate the JMRI programmer instance that is still waiting for a response.

The log will still be showing attempts to reestablish communication with the command station.

This is usually beneficial because it allows a user whose command station has hung up to reboot the command station and JMRI will pick up roughly where it left off, although in this case the programmer terminates the read request after several retries and goes to the next CV (I use this feature occasionally when my command station hangs during Read All CVs on my NCE system.) 

> 4. Repeated above steps 3 and 4, got "jmri.ProgrammerException: programmer in use",

Because the JMRI programmer is still in use waiting for the Read CV8 request to respond.

We probably should change the IdentifyDecoder routine to abort completely with a "Timeout talking to command station (306)" after a few retries.

Dave in Australia

closed 2021-04-14 by stale[bot]