JMRI/JMRI #7389: Decoder definition - long address 2 CV - tag <splitVal> - 4.16

sidlo64, 2019-09-05:
In version 4.14 everything is OK.
In version 4.16 is problem.

How to reproduce:
Version 4.16 start DecoderPro
Connection Loconet Digitrax Loconet Simulator
Create new roster entry 
Public-domain and DIY - dccdoma.cz - UNI16ARD-8VYH
open comprehensive programator - folder Decoder
Read full sheet
reading finish with exception

2019-09-05 08:43:05,232 progdebugger.ProgDebugger             INFO  - read CV: 34 mode: Direct will read 123 [AWT-EventQueue-0]
2019-09-05 08:43:05,246 progdebugger.ProgDebugger             INFO  - read CV: 36 mode: Direct will read 123 [AWT-EventQueue-0]
2019-09-05 08:43:05,258 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.ClassCastException: java.lang.Long cannot be cast to java.lang.Integer
	at jmri.jmrit.symbolicprog.AbstractQualifier.availableStateFromEvent(AbstractQualifier.java:60)
	at jmri.jmrit.symbolicprog.AbstractQualifier.processValueChangeEvent(AbstractQualifier.java:48)
	at jmri.jmrit.symbolicprog.AbstractQualifier.propertyChange(AbstractQualifier.java:37)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at jmri.jmrit.symbolicprog.SplitVariableValue.actionPerformed(SplitVariableValue.java:392)
	at jmri.jmrit.symbolicprog.SplitVariableValue.setValue(SplitVariableValue.java:468)
	at jmri.jmrit.symbolicprog.SplitVariableValue.updateVariableValue(SplitVariableValue.java:307)
	at jmri.jmrit.symbolicprog.SplitVariableValue.propertyChange(SplitVariableValue.java:722)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at jmri.jmrit.symbolicprog.CvValue.notifyValueChange(CvValue.java:85)
	at jmri.jmrit.symbolicprog.CvValue.programmingOpReply(CvValue.java:457)
	at jmri.Programmer.notifyProgListenerEnd(Programmer.java:216)
	at jmri.progdebugger.ProgDebugger$3.run(ProgDebugger.java:233)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

dheap, 2019-09-05:
@sidlo64 I'll look into this ASAP.

dheap, 2019-09-05:
@sidlo64 I'm looking into this but it has turned out to be non-trivial.

devel-bobm, 2019-09-05:
@dheap,

FYI - perhaps userful, perhaps red herring, but it is curious to me that two different "cast from Long" exceptions popped up recently.  The other one I saw:

A recent post to the "jmriusers" group with subject "Lost my Sensor folder in the Table option" reported a comparable "java.lang.Long cannot be cast to java.langString" (there might be a transcription error in the quoted exception string in the original email!) occuring as part of the RowSorter functionality.  

David Sand determined that the proximal cause of that error message was that it occured whenever the table was dispaying a column containing integral values, such as when "Show debug debounce delay" was checked in the Sensors table.  Simply un-check the debounce delay checkbox and the problem disappeared.  I did not notice any github issues filed against that problem, tho.

dheap, 2019-09-05:
@devel-bobm Thanks. It may or may not be related.

The problem detected by @sidlo64 occurs because the new splitVal code stores the value internally as a Long (the previous code was limited to 2 CVs so an Integer was adequate for storage). Many of the panes in UNI16ARD-8VYH use qualifiers with the "Signal Tower Address" splitVal and the Qualifier code in JMRI currently expects property change values to be capable of being cast to the Integer type.

I'll have a look for the jmriusers post.

dheap, 2019-09-13:
@sidlo64 Sorry about the delay on this one. Ive got a code fix but it needs some associated tidying up. Family and other urgent priorities have kept me away from finishing it. I'll get back to it ASAP.

svatoun, 2019-09-26:
What is the correct approach: have a backwards-compatible (safe) behaviour, or overhaul all the code that potentially watches PropertyChange("Value") ? . There's no explicitly defined contract, but `SplitVariableValue`'s supertype defines `setIntValue(int)`, which (implicitly) assumes that relevant property change should report `Integer` - or `null`. That is, if the property event was correctly named, which it is not (violates Java Beans specification).

The previous `int`-firing code has been there since 2002, so I'd suggest a backward compatible behaviour for at least some transitional periods. While other `Vairable` subclasses (enum, const, ...) all use `int`-based contract, `SplitVariable` should define an **additional** event name to broadcast old+new Longs so clients that handle (generic) values are not triggered. Old clients can only receive `null` as old+new values from "Value" property change, if any.

BTW there are few places in `Variable` hierarchy where `setIntValue` override does not properly fire Value change events, while the counterpart `setValue`, or `selectValue` does.

dheap, 2019-09-27:
@svatoun
I'm actually coding up a backwards-compatible solution to this, but have been delayed by family responsibilities and (overdue but necessary) computer upgrade issues.

I'm away from my computer at present and once I get back to my new code, I'll take a look at the issues you raised, some of which I have already addressed, but others I hadn't detected.

Thanks for your input. Once I've got what I consider workable, I'll push a branch and create a Pull Request for further consideration.

Dave in Australia

svatoun, 2019-09-27:
@dheap - I have a solution ready; woring at least for @sidlo64 situation (I needed to work with his decoder configs, so there's a motivation :-D). I can attach the diff. I am sorry for effort duplication, need to proceed further (somehow) :-)

dheap, 2019-09-27:
@svatoun There's no way I can have anything ready for the next few days.

So if you want to run with this, I can't preempt that. I couldn't get a WIP PR up for at least 4-5 days. But I do have some concerns about some of your assertions/suggestions.

Please make sure you don't break other things in in other JMRI definitions, or other functionality.

I'd prefer if you could wait to see what direction I was taking so we could maybe combine ideas, but in the end I have to put family responsibilities ahead of JMRI coding...

dheap, 2019-09-27:
@svatoun I note that your main post has been considerably edited, with significant additions. Unfortunately those of us who monitor these discussions via email will not see these.

Please repost your initial comment in edited form as a new comment.

svatoun, 2019-09-27:
@dheap according to diff on the web, I made just typo/wording corrections to https://github.com/JMRI/JMRI/issues/7389#issuecomment-535797905.

But for the reference, there's original comment copied from GH's history:

> What is the correct approach: have a backwards-compatible (safe) behaviour, or overhaul all the code that potentially watches PropertyChange("Value") ? . There's no explicitly defined contract, but [...]

svatoun, 2019-09-27:
A possible fix is attached. Neds to review / iterate on, since durig the course I've discovered other subtle (most possible) bugs.

svatoun, 2019-09-27:
> Please make sure you don't break other things in in other JMRI definitions, or other functionality.

So far tests are passing, and the change is IMHO quite conservative. But tests were passing even on code although `PropertyChange` was fired while the property was still in its old value, was not fired at all after a change, or was reporting type-incompatible data :)

> I'd prefer if you could wait to see what direction I was taking so we could maybe combine ideas, but in the end I have to put family responsibilities ahead of JMRI coding...

That would be the best outcome, of course.

bobjacobsen, 2019-09-27:
Note: I'm no longer expert in this area of the code, and some of my understandings data back to when the original (bad) code was written.

@svatoun:
> A possible fix is attached. 

I don't see an attached file, sorry.  If you have a branch with this in it, could you push it to your GitHub repo and list either the branch name or a particular commit SHA?  Thanks.

svatoun, 2019-09-27:
Sorry - Github has referenced my branch automatically a few pages above (on a web), It's

https://github.com/svatoun/JMRI/commit/c5ac6e09d840426d495bb8b5934fdad57260e710

dheap, 2019-09-27:
@svatoun

The text I'm seeing at your repost is only a subset of the the text I'm seeing at your earlier comment (marked as Edited).

But I'm working under difficulties at the moment, with my only device and access being unsupported browsers on a very old iPad. So I'm finding it almost impossible to have any meaningful discussion and I certainly can't access your branch in a meaningful way, nor can I access my code repository with changes. It's also the middle of the night over here.

I won't have access to any/my computer for around 2-3 days so I'm asking that we withold  any further discussion until  I'm at my computer again and I am able to discuss meaningfully and create an accessible branch/view your branch.

As @bobjacobsen has commented, the original code was bad, my changes were overlaid on that and I'd plead that we hold off a couple of days until I'm in a position to engage meaningfully.

svatoun, 2019-09-27:
re. repost: no idea ... what you got by e-mail. I thought I've originally written what was reposted (actually copy/pasted from what Github shows as the original comment); at least I don't remember writing something substatially different.  

Anyway Github WebUI tracks (at least I assumed) all edits, so when your machine is back, you can see the original message, whatever it was.

I'd say there's no big time pressure; at the time I've decided to patch the bug, it was > 20 days old with no ETA and a comment indicating there are more important things to do ... and the bug  was simply in my way ;) so I simply attempted to patch to proceed further with my own work. No offense meant.

By no means I'd want to bypass a rather rigid review process. Waiting for your review/comments.

dheap, 2019-10-01:
@svatoun 
I've returned to my hardware now and have been looking into the issues you raised, checking whether I had addressed them in the code I had already written (I had) adding [new tests](https://github.com/JMRI/JMRI/compare/master...dheap:longval), waiting for CI tests to run and looking at your proposal. It's my bedtime now so I need to not engage in comments or discussion for around 10 hours so I get some sleep (one of the problems of this global village).

Till tomorrow...

dheap, 2019-10-02:
@svatoun I've spent quite some time today working on reviewing your comments and the two code proposals and composing a response. Unfortunately I've run out of time and personal circumstances mean I can't continue work on this for about the next 20 hours.

Till tomorrow...

svatoun, 2019-10-02:
> I can't continue work on this for about the next 20 hours

Hey, this is not a race against the time, no deadline show-like event or so :) Thank you for evaluating / integrating the code.

svatoun, 2019-10-08:
Ping ... it's about a week from the last entry, and issue #7407 says the test release is around the corner (Oct 12). This bug actually breaks functionality, so it would be probably good to have it in a release soon.

dheap, 2019-10-08:
@svatoun I'm aware of that and have been working on it between unavoidable personal commitments. Planning to allocate tomorrow to tidying up work on it.

dheap, 2019-10-09:
@svatoun Thanks for your input. I've looked at your code, comments and proposed changes:

- Some of the issues (such as the inability to represent full 64 bit values in `SplitVariableValue`) have been addressed (and in this case verified by Test code and in-decoder tests).
- I'm quite opposed to the use of marker values in numeric fields. We've already been bitten by that in IdentifyDecoder, where an int with a marker value of -1 was used to indicate "no productID".
- I share your concerns about the inconsistencies in the application of offset and factor in various setters and getters but we may risk breaking some existing decoder definitions set up using the current behaviour. I wouldn't make changes to that in this test release so users can isolate adverse effects to a particular release.
- For the same reasons, I'm suggesting that some of your corrections aren't merged in this particular release.
- I'm adding a getLongValue() to sit alongside getIntValue() in VariableValue (with a default compatibility behaviour), which involves no changes to other current subtypes and maintains full backward compatibility.
- I'm also refactoring  setValue(long) (an overloaded method) in SplitVariableValue and its subtypes to setLongValue(long) to sit alongside setIntValue().
- Would your proposal to fire a "generic null, null event" ensure that non-updated listeners would throw an exception (as at present) rather than silently using a truncated value?
- Given the time constraints to resolve the problem for @sidlo64 by the timeline in #7407 and the fact that I've made lots of other necessary changes to `SplitVariableValue`and the Qualifier processing code, I'm thinking to go with my object-based solution as an interim fix. After the release of 4.17.5 we can move over to your `fireValueChange` solution and revert/update _some_ of my changes to the Qualifier code.

I've created two alternative branches to choose one for a Pull request:
- Branch [longval](https://github.com/JMRI/JMRI/compare/master...dheap:longval) doesn't include all your other corrections, deferred until after 4.17.5.
- Branch [longval2](https://github.com/JMRI/JMRI/compare/master...dheap:longval2) does include all your other corrections.

It's past my bedtime now, so I'll have address this in the morning and create a Pull Request.

dheap, 2019-10-09:
See #7496 and #7497.

dheap, 2019-10-10:
We need to make a decision which PR to go with; the conservative #7496 (other changes added after V4.17.5) or the more inclusive #7497, but we need to get one or other into V4.17.5 (as per #7407 deadline).

@bobjacobsen Do you have any views either way?

@svatoun I'm guessing you'd favour #7497?

P.S. I don't want to make changes at this late stage so I will look at the (I consider minor) Code Climate [suggestions](https://codeclimate.com/github/JMRI/JMRI/pull/7497) (Method 27 lines and Method Cognitive Complexity 6) later.

svatoun, 2019-10-10:
Apologies; can properly respond only after 20CEST, I need to verify/re-read. Quick scan showed possible breakage of `SplitVariableTextField` in `longval` branch. General, there's a big design issue with the way how `SplitVariableValue` started to behave with respect to other types in the `VariableValue` type tree.

dheap, 2019-10-10:
@svatoun I've done (and just repeated) extensive testing of `SplitTextVariableValue`, reading from and writing to decoders and can't fault the behaviour. I did discover minor documentation errors, which I've just fixed.

In the past I've also done extensive testing of `SplitVariableValue` and its other subclasses as applied to ESU decoders and haven't found issues. But I have mainly relied on the unit tests for non-default values of factor and offset (although the ESU SwitchPilot series does use these).

The scope of this issue and consequent PR is to provide a fix for the issue discovered by @sidlo64. It's not about  major redesign of `SplitVariableValue` and its other subclasses. I'm just asking you to verify the changes I copied from your branch.

I've done real-world  testing of the both #7496 and #7497 and haven't discovered any issues. I intend to merge one or other of these ASAP so the fix for @sidlo64 makes it into V4.17.5. Please give me your preference.

svatoun, 2019-10-10:
> I've done real-world testing of the both #7496 and #7497 and haven't discovered any issues

@dheap - taking a work break in the office.
 Yes, my mistake - reading diff was inaccurate: the "breakage" I thought is there was from `extra4` charset attribute (for split text) entering the base class' `step1Actions`, but it cannot reach it it fact. Sorry - good.

> The scope [...] is to provide a fix for the issue discovered by @sidlo64.
OK.
I actually don't have a preference, #7497 mostly adds consistency bugfixes. If you do not feel confident in whether they do what they're documented to do, do not put them into the release. Can be merged later to have a full release cycle of testing.

If anything, I'd suggest to include the the change in `VariableValue.java` as broadcasting an event with old state still in place is a bad thing(tm).

dheap, 2019-10-10:
Merged #7496. [Extra changes](https://github.com/JMRI/JMRI/pull/7497/commits/9f5ceaa9bb4b9f7cedbdb653cc6ecf7191fcdf27) deferred until V4.17.6.

sidlo64, 2020-01-11:
Fixed by #7496

closed 2020-01-11 by sidlo64