JMRI/JMRI #7400: Move primary build tool to Maven?

pabender, 2019-09-08:
Is it time that we move our primary build tool to maven?  I would like to open up that question for a discussion.

We have been maintaining two different build chains for quite a while now (maven and ant).  The ant builds have always been our primary build tool, but maven has been used at least to a certain extent on our CI builds.
  
Maven is intended to use "convention over configuration", which reduces the size of the configuration files required.  (you can see that in our current build configurations where the ant build.xml file is approximately 2600 lines and the  maven pom.xml file is about 1300.  )

Here are a few advantages for maven I know of right off the top of my head:

1. It is possible to modularize our builds.
2. Maven is much better at library management than ant
3. The distribution process can be simplified.
4. Maven's surefire testing plugin can run tests without the PackageTest files (see the effort @rhwood is taking to remove the PackageTest files in #6680 )

The one that really has me thinking about this now, is how using maven would allow us to modularize our builds.  This process by itself gains us a couple of advantages 1) When we make a change to one module, but leave the others alone, maven won't recompile those other things, which reduces our development time 2) We can split the POM file up and target dependencies and other configuration to smaller pieces of the software (maven pom files are hierarchical, so they inherit at least some configuration from the parent). 3) Users of JMRI as a library can use the smaller pieces individually.

I envision us (eventually) having 5 modules (by content, not necessarily by current structure):

1. JMRI core
2. JMRI applications
3. JMRI tools (roughly the content of jmri.jmrit)
4. JMRI system extensions (roughly the content of jmri.jmrix)
5. JMRI network services (roughly the content of jmri.jmris, plus the web server and withrottle which are outside of jmri.jmris).

To do this, we have to make sure no two modules are co-dependent.  I've made an attempt to build two modules (apps and everything else) using the current directory structure, but there are more than a few references to apps from within jmri, so the two modules become co-dependent.  It is going to take some work to get us to where we can build multiple modules (and a restructuring of the code to emphasize the splits would be best ).

Just a word on distribution.  Theoretically, all it takes to build and distribute with maven is running 'mvn deploy'.  This will push any artifacts we generate up to maven central.  (There are 3 of us who currently have permission to push files up to maven central.  I've made sure the process works, but we haven't distributed anything that way yet).

The typical artifact is a jar file, but I learned recently that you can actually push any artifact you want up.  (pi4j is pushing their distributions up to maven central).

bobjacobsen, 2019-09-08:
I think a discussion of modularization would be a good thing.  I've been working on using [ArchUnit](https://www.archunit.org) to check our [conventions](http://jmri.org/help/en/html/doc/Technical/IntroStructure.shtml). See my [bobjacobsen/ArchUnit branch](https://github.com/bobjacobsen/JMRI/tree/ArchUnit) for that.  The number of violations is _huge_, but we can certainly get started if that's the consensus.  

But that seems separate from the question in this nicely-round-numbered Issue: "Move primary build tool to Maven?"

Could you say a few more words about what "_primary_ build tool" means?  

Right now, Ant is used for building releases, creating the release downloads, signing executables, creating the Javadoc outputs for the web, etc.  Is the idea to move all those to Maven instead?

I've never really understood the value of moving to Maven for things like that, so would appreciate some details.

pabender, 2019-09-09:
Sent from my iPhone

> On Sep 8, 2019, at 7:05 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> I think a discussion of modularization would be a good thing. I've been working on using ArchUnit to check our conventions. See my bobjacobsen/ArchUnit branch for that. The number of violations is huge, but we can certainly get started if that's the consensus.
> 
> But that seems separate from the question in this nicely-round-numbered Issue: "Move primary build tool to Maven?"
> 

Why I link the two is that maven is really designed around a module concept.  

Most maven modules produce a single jar file.
> Could you say a few more words about what "primary build tool" means?
> 
Currently, we can do most of what we do in any using maven.  In  order to do that, we are maintaining both an ant build chain and a maven build chain.

I would basically like to phase out the use of ant so we only have to maintain one build process.
> Right now, Ant is used for building releases, creating the release downloads, signing executables, creating the Javadoc outputs for the web, etc. Is the idea to move all those to Maven instead?
> 

Ultimately yes, but that doesn’t mean ant goes away all at once, just that we phase ant out as we figure out how to replace the processes with maven build steps.  

We already use maven to trigger any builds on appveyor, so maven calling ant isn’t an issue.

Right now, I think the only thing our maven build can’t do is the steps around creating the release packages, including signing builds, and deployment of artifacts to the website.

> I've never really understood the value of moving to Maven for things like that, so would appreciate some details.
> 
Maven is designed around a common build process ( see https://maven.apache.org/guides/introduction/introduction-to-the-lifecycle.html 
) and a common directory structure ( https://maven.apache.org/guides/introduction/introduction-to-the-standard-directory-layout.html ).  Because of the common process and structure, new developers familiar with maven have a smaller learning curve.

This is where the modular build plays into this.  In order to easily build the project as modules, maven wants each module to be in a directory that follows the structure.  So we physically need to separate the parts we want to build as modules.

bobjacobsen, 2019-09-09:
It would be an interesting test to build out the Maven support to do all the things that are now done with Ant.  I'll append that list below. Note that it's trivial to _get_ that list; AFAIK there's no equivalent way to query Maven for lifecycle, phase, etc.

That goes to an issue I've long had with Maven setups on other projects:  It's really hard to tell them to "just do this little thing".  We have Ant set up so that I can just recompile my main-code changes without compiling tests, testing and linting, etc.  I can ask Ant to just check for warnings with the specific tool I'm currently working against, or run a quick (no lint, no UML) Javadoc, etc. With Ant and @rhwood's nice `./runtest.csh` setup, I can run just one test class, or the tests in a small part of the class tree. It's easy to find the errors and warnings in the _short_  build output. Etc.

Maybe Maven can do all that., and we just haven't spent enough time improving the Maven set up for that kind of uses.  Or maybe Ant can still be used for all that, while Maven is used for other production-related things.  But I don't see any gain from losing those.

```
% ant -p
Buildfile: /Users/jake/Documents/Trains/JMRI/projects/JMRI/build.xml

    Provides build services for JMRI libraries and applications
    
Main targets:

 SampleMinimalProgram            build and run build and run SampleMinimalProgram class
 alltest                         build and run AllTest suite
 alltest-coveragecheck           Test Coverage check from logalltest
 alltest-coveragereport          Generate Test Coverage Report from logalltest
 appveyor                        CI tasks on Appveyor
 checkPropertiesFiles            check all JMRI properties files for duplicate properties in same file
 checklineends                   check text files for proper LF line ends (Linux, Mac only)
 checknonascii                   check for non-ASCII characters in properties files (Linux only)
 ci-checker-framework            Check java/src tree for null pointer dereferences; only compiles non-compiles classes, so do 'clean' first to do all
 ci-compile-deprecations         compile source, omitting tests, warning on deprecations - for Jenkins CI
 ci-test                         run AllTest CI tests
 ci-test-appveyor                run ci-test using Appveyor
 ci-test-appveyor-coveragecheck  Test Coverage check from ci-test-appveyor for Travis
 ci-test-travis                  run ci-test using Travis-CI
 ci-test-travis-coveragecheck    Test Coverage check from ci-test-travis for Appveyor
 ci-tests-deprecations           compile source and tests, warning on deprecations - for Jenkins CI
 clean                           remove compilation results to force rebuild
 compile                         compile source, omitting tests - identical to debug target
 coveragecheck                   Check test results in existing jacoco.exe file
 coveragereport                  Generate Test Coverage Report from existing jacoco.exe file
 cucumber                        run Cucumber tests
 debug                           compile source with debugging on, omitting tests - identical to compile target
 decodercheck                    run decoder XML checks
 decoderpro                      build and run DecoderPro (formerly DecoderPro 3) app
 decoderpro-old                  build and run the old DecoderPro app
 decoderpro-remote               build and run DecoderPro on remote target
 dist                            create a distribution jar file without tests
 dist-debug                      create a distribution jar file including tests
 fixlineends                     convert specific files to LF line ends
 gui3ide                         build and run build and run GUI3 IDE demo app
 gui3mdi                         build and run build and run GUI3 MDI demo app
 headless-coveragecheck          Generate Test Coverage Report via logheadlesstest
 headless-coveragereport         Generate Test Coverage Report via logheadlesstest
 headlesstest                    build and run HeadLessTest suite
 i18n-consistency-check          Verification of localized strings in source code
 installtest                     build and run build and run InstallTest app
 jar                             create working jar file with current contents
 jar-resources                   package all resources as a jar file
 javacc                          JavaCC related actions
 javadoc                         create Javadocs
 javadoc-lint                    create Javadocs
 javadoc-uml                     create Javadocs with UML
 jjdoc                           create JJDoc BNF documentation
 jjtree                          Run JJTree related actions
 jmrifaceless                    build and run build and run JmriFaceless app
 launchjmri.exe                  recreate the LaunchJMRI.exe used by the Windows release package
 locale                          check DecoderPro I8N
 logalltest                      run AllTest test suite with jacoco output
 logheadlesstest                 run headlesstest test suite with jacoco output
 make-test-release-branch        Make a new test release branch in Git from the head of the master branch
 package-linux                   create a Linux release package
 package-macosx                  create a Mac OS X install package
 package-windows                 create a Windows release package
 packages                        create all OS-specific release packages
[... cut ...]
 tests-warnings-check            compile test sources with ECJ warning generation on and fail if any warnings
 uploadjavadoc                   upload existing Javadocs to jmri.org without recreating
 warnings                        compile source with ECJ warning generation on, omitting tests
 warnings-check                  compile source with ECJ warning generation on and fail if any warnings, omitting tests
 zip                             build a .zip file of decoders
Default target: debug
```

bobjacobsen, 2019-09-09:
@rhwood wrote in #7404:
> I’m in favor of moving to maven with or without modularization - simply put every IDE can support maven because it has well defined build phases, but no IDE can support ant (in the same way that no IDE supports make; NetBeans’s ant support is entirely predicated on creating an ant build.xml that has specific targets that NetBeans has defined the purpose and meaning of).

Sorry, but that sentence doesn't seem logically consistent to me.  Nothing in it justifies "moving to maven" as opposed to "better supporting maven".  I can certainly agree with the second of those.

For example, having Maven and Eclipse set up so that Eclipse can work entirely with Maven would be great. NetBeans, which I use sometimes, would also be good.  Right now, both of those use Ant for some functions.

Perhaps I still don't understand what "primary" means in this context, but it seems to me that the path is to have Maven get better and better until it becomes clear it handles everything that Ant does.  Right now, it appears the converse is true.

pabender, 2019-09-09:
> That goes to an issue I've long had with Maven setups on other projects: It's really hard to tell them to "just do this little thing". We have Ant set up so that I can just recompile my main-code [...]

One of the differences between ant and Maven is Maven does not let you add build lifecycle phases.  You can have different profiles that do different things in those phases, but the phases are well defined.  You can also call individual actions of a Maven plugin to do specific things.

> Maybe Maven can do all that., and we just haven't spent enough time improving the Maven set up for that kind of uses. Or maybe Ant can still be used for all that, while Maven is used for other production-related things. But I don't see any gain from losing those.
> 

While that list of ant targets are impressive, we should look at your list of ant targets and remove the ones that are not being utilized currently.

Take, for example, the Travis targets.  The build/test actions on Travis are controlled by https://github.com/JMRI/JMRI/blob/master/scripts/travis.sh which is called through https://github.com/JMRI/JMRI/blob/master/.travis.yml

While two ant targets (tests-warnings-check and scanhelp  ) are called during the static tests, None of the *travis* ant targets are in use. 

Incidentally, it looks like those two ant target calls can be removed pretty easily.

Test-warnings-check uses EJC to check for warnings.  We can use the Maven compiler plugin to call the eclipse compiler ( see https://maven.apache.org/plugins/maven-compiler-plugin/non-javac-compilers.html ).  
Scanhelp just executes a shell script.  There is a Maven plugin that can do that: https://www.mojohaus.org/exec-maven-plugin/usage.html

We should probably create a Maven profile for these static tests.

While I am at it, let’s take a look at what Appveyor does. 

The compile on Appveyor is currently performed via ant, but that ant build is triggered via Maven
( https://github.com/JMRI/JMRI/blob/master/appveyor.yml calls Maven on line 32).  Of course looking at what happens in Maven:

https://github.com/JMRI/JMRI/blob/ec9d1f585e611320265aa2311bbfeb25fa56d19f/pom.xml#L93-L129

We see that it is calling the ant Appveyor target and then using Maven’s Surefire plug-in to check line endings.

The Appveyor target in ant has no actions of its own.  It just depends on the checkPropertiesFiles and ci-test-appveyor-coveragecheck ant targets.  

-test-appveyor-coveragecheck could be performed in exactly the same manner that the test execution on Travis occurs.  

The checkPropertiesFile task uses rscbundlecheck.jar to perform checks on the properties files.
There is a Maven plugin that replicates the process ( according to the documentation ) of integrating resourceBundleCheck into Maven, but it does not appear to be well supported ( last commit 2013... ).  If the plugin doesn’t work, we can directly call that ant target from Maven.

bobjacobsen, 2019-09-09:
Certainly agree that it's fine to remove things that are not in use.  

 - The travis and appveyor targets _were_ what was being used for Travis, which was convenient for people who wanted to run them locally. Not sure whether anybody is still doing that or not, but perhaps it would reduce the amount of WIP PRs needed.

Certainly agree it's fine to add things to Maven that Ant already does.

 - As to scanhelp and test-warnings-check et al: I have no doubt that Maven can do them.  But can it do them as easily as typing "ant scanhelp" and "ant warnings" _without_ doing lots of other stuff too? I used both of those these weekend. They were nice and efficient, with easily found errors.  It would be great if somebody could tell me a similar length Maven command that would be as quick and taciturn.

rhwood, 2019-09-09:
> On Sep 8, 2019, at 23:14, Bob Jacobsen <notifications@github.com> wrote:
> 
> @rhwood <https://github.com/rhwood> wrote in #7404 <https://github.com/JMRI/JMRI/issues/7404>:
> 
> I’m in favor of moving to maven with or without modularization - simply put every IDE can support maven because it has well defined build phases, but no IDE can support ant (in the same way that no IDE supports make; NetBeans’s ant support is entirely predicated on creating an ant build.xml that has specific targets that NetBeans has defined the purpose and meaning of).
> 
> Sorry, but that sentence doesn't seem logically consistent to me. Nothing in it justifies "moving to maven" as opposed to "better supporting maven". I can certainly agree with the second of those.
> 
> For example, having Maven and Eclipse set up so that Eclipse can work entirely with Maven would be great. NetBeans, which I use sometimes, would also be good. Right now, both of those use Ant for some functions.
> 
Eclipse is *already* setup to use JDT (JDT is Eclipse’s “Java Development Tools”, which is “yet another build system we maintain”, but is not removable from Eclipse.) + maven internally (there is one manual bootstrap step to run `mvn generate-sources` or `ant javacc` required), and prior to being configured to use maven internally was configured to only use JDT (still with that manual bootstrapping). We are still maintain a separate list of dependencies for Eclipse, but that is through inertia, not through an existing technical requirement.

NetBean’s integration with ant relies on a number of other files (under nbproject) that we built by hand as wrappers around the JMRI build.xml script; and separate manually maintained lists of dependencies (because NetBeans can’t read build.xml to figure out our dependencies).

So other than having assigned a custom task to a menu item or button in IDEs, ant isn’t really used by any IDE, so, no, I don’t think ant is supported by IDEs in the manner that maven is.

dheap, 2019-09-09:
> @bobjacobsen As to scanhelp and test-warnings-check et al: I have no doubt that Maven can do them.  But can it do them as easily as typing "ant scanhelp" and "ant warnings" _without_ doing lots of other stuff too? I used both of those these weekend. They were nice and efficient, with easily found errors.  It would be great if somebody could tell me a similar length Maven command that would be as quick and taciturn.

I use "ant scanhelp" and "ant test-warnings-check", albeit usually from within NetBeans rather than Terminal (which I do use a lot).

pabender, 2019-09-10:
> Certainly agree that it's fine to remove things that are not in use.
> 
> * The travis and appveyor targets _were_ what was being used for CI, which was convenient for people who wanted to run them locally. Not sure whether anybody is still doing that or not, but perhaps it would reduce the amount of WIP PRs needed.
> 

I take that as an argument that we have too many different build targets.  Newer developers, in particular, see the long list of targets and are somewhat overwhelmed.  Part of the simplicity of Maven is that the build lifecycle is fixed.  You can’t add arbitrary targets to it, which means new developers have an easier time acclimating to using it... especially if they were already building other projects with Maven.  

> Certainly agree it's fine to add things to Maven that Ant already does.
>
> * As to scanhelp and test-warnings-check et al: I have no doubt that Maven can do them.  But can it do them as easily as [...]

The command in Maven may not be as succinct, but the actual processing is typically similar, if not identical, to the ant process.

bobjacobsen, 2019-09-19:
Another point to consider:  I'm currently at a [scientific school](https://indico.cern.ch/event/769356/) that includes exercises for ~70 students. The students use their own laptops, but to reduce variability they run the exercises via a VirtualBox VM. The students pull the exercise content from GitHub, GitLab and CERN-based servers. The exercises are extensively tested the week before.  What could go wrong?

Well, one of my fellow instructors found out:  Using Maven to obtain dependencies is not 100% reliable.  A 4th level sub dependency (!) was withdrawn for security reasons, and suddenly 70 people couldn't get their work done.  

(Another example of this happened on a larger scale when `left-pad` was withdrawn from NPM a while ago, leaving Babel and React unbuildable)

For almost 20 years, JMRI has been distributing all the dependencies (except Java itself) needed to build and run.  If we do switch to Maven, we should figure out how to have it use the local-to-JMRI resources.  (And if that reduces the large surplus of Maven output w.r.t. Ant, that would be a good thing too)

rhwood, 2019-09-19:
> On Sep 19, 2019, at 03:48, Bob Jacobsen <notifications@github.com> wrote:
> 
> Another point to consider: I'm currently at a scientific school <https://indico.cern.ch/event/769356/> that includes exercises for ~70 students. The students use their own laptops, but to reduce variability they run the exercises via a VirtualBox VM. The students pull the exercise content from GitHub, GitLab and CERN-based servers. The exercises are extensively tested the week before. What could go wrong?
> 
> Well, one of my fellow instructors found out: Using Maven to obtain dependencies is not 100% reliable. A 4th level sub dependency (!) was withdrawn for security reasons, and suddenly 70 people couldn't get their work done.
> 
> (Another example of this happened on a larger scale when left-pad was withdrawn from NPM a while ago, leaving Babel and React unbuildable)
> 
> For almost 20 years, JMRI has been distributing all the dependencies (except Java itself) needed to build and run. If we do switch to Maven, we should figure out how to have it use the local-to-JMRI resources. (And if that reduces the large surplus of Maven output w.r.t. Ant, that would be a good thing too)
> 
We already use local-to-JMRI resources with maven, but if a dependency in JMRI got yanked for a security or legal concern, we *should fail* our builds because we *should not ship* with such a dependency, so in our case, having the builds break in that case is a benefit.

bobjacobsen, 2019-09-19:
> We already use local-to-JMRI resources with maven,

That is not my experience.

>  but if a dependency in JMRI got yanked for a security or legal concern, we *should fail* our builds because we *should not ship* with such a dependency, so in our case, having the builds break in that case is a benefit.

You and I differ on whether such a black-and-white view is appropriate. Luckily, it's not your call.

rhwood, 2019-09-19:
> > We already use local-to-JMRI resources with maven,
>
> That is not my experience.

It is your experience since we use resources in maven that *are exclusively* local-to-JMRI and tools like Travis-CI would not work if that were not the case. See  https://github.com/JMRI/JMRI/blob/29af90110c52a9b268bb9a2e7e9689140a97bcba/pom.xml#L1281-L1295 for where maven reads local-to-JMRI resources from. The resources we currently pull locally are:

- `lib/net/bobis/jinput/hidraw/jhidrawplugin/0.0/jhidrawplugin-0.0.pom` (not available anywhere else in source or other form)
- `lib/org/eclipse/paho/mqtt-client/0.4.0/mqtt-client-0.4.0.pom` (we use an ancient (2013?) version that Eclipse did not publish to Maven Central or their own software repo but only makes available on GitHub (we added this dependency to JMRI in 2018, when version 1.1.0 (published in 2016) was the most recent version on Maven Central and GitHub, but I have zero information about why we choose to use a much older dependency))
- `lib/org/netbeans/external/jemmy/22-00c9f753cd0a/jemmy-22-00c9f753cd0a.pom` (this is a build not available elsewhere)
- `lib/com/apple/AppleJavaExtensions/1.5/AppleJavaExtensions-1.5.pom` (Apple never published this to any software repo)
```
since none of those are available in [Maven Central](http://search.maven.org)

The reason other dependencies are not pulled from the local-to-JMRI location is that sources and Javadocs are also available on Maven Central (see [example listing from Jython](https://repo1.maven.org/maven2/org/python/jython-standalone/2.7.1/)) and Eclipse, IntelliJ, and NetBeans can all provide the JMRI developer nearly immediate access to the Javadocs and sources of those dependencies without requiring the developer search for those separately (i.e., by using the published site and not copying our own, we can directly inspect the code we are relying on). I rely on this capability extensively and would hate to loose it.

bobjacobsen, 2019-09-19:
@rhwood, please try this:

 - Disconnect from the internet
 - Drop your maven cache  (in my case, that’s `rm -rf ~/.m2`, others may vary)
 - Try to build JMRI with Maven.

I get:

```
[INFO] Scanning for projects...
[INFO] 
[INFO] ---------------------------< org.jmri:jmri >----------------------------
[INFO] Building JMRI 4.17.5-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
Downloading from central: https://repo.maven.apache.org/maven2/org/codehaus/mojo/javacc-maven-plugin/2.6/javacc-maven-plugin-2.6.pom
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  0.555 s
[INFO] Finished at: 2019-09-19T13:38:34+03:00
[INFO] ------------------------------------------------------------------------
[ERROR] Plugin org.codehaus.mojo:javacc-maven-plugin:2.6 or one of its dependencies could not be resolved: Failed to read artifact descriptor for org.codehaus.mojo:javacc-maven-plugin:jar:2.6: Could not transfer artifact org.codehaus.mojo:javacc-maven-plugin:pom:2.6 from/to central (https://repo.maven.apache.org/maven2): repo.maven.apache.org: unknown error: Unknown host repo.maven.apache.org: unknown error -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/PluginResolutionException
```

Does your setup do that build from local resources?  If so, how?

pabender, 2019-09-19:
@bobjacobsen

Sent from my iPhone

> On Sep 19, 2019, at 7:59 AM, Bob Jacobsen <notifications@github.com> wrote:
> 
> @rhwood, please try this:
> 
> Disconnect from the internet
> Drop your maven cache (in my case, that’s rm -rf ~/.m2, others may vary)
> Try to build JMRI with Maven.
> I get:
> 

Yes, this will fail, but that experiment cannot have any other outcome because you are doing something you really shouldn’t do unless you are connected to the internet.

You only need to be connected to the internet the first time a dependency changes, but you also need to be connected to know the dependency changes in the first place ( i.e. you can’t pull from github without being connected to the internet either... )
> [Does your setup do that build from local resources?  If so, how?
We only put files in the local cache that we cannot find elsewhere.

And this is part of why I want us to use maven, we don’t have to do as much work to manage our dependencies, which means updating dependencies is less of a hassle.

As an example of that,  what is currently required to update a jar:
Add the jar to the lib directory and remove the old version
Update build.xml for ant
Update pom.xml for maven
Update a file for netbeans
Update the .classpath file (for ?)

I usually remember the build.xml step and the pom.xml step, but the other steps can be problematic.

One area we have a real problem with in this process is that some of our jar files in lib have had the version information removed from them, which makes it easier to update them, but harder to tell if you have actually updated the library ( especially if you are working in a freshly created local git repo).

If we move to using maven exclusively, I just update the pom file.  Maven does the rest.  It is a much less error prone process.

bobjacobsen, 2019-09-19:
If that's the best reason we can find for "moving to maven _exclusively_", then we can close this Issue:  That's not a reason for taking away things that people depend on.

Again: If people want to work on making JMRI's Maven support better, please go ahead.  But that is _NOT_ the same as subtracting the advantages of the Ant support, so please don't do that until Maven can do what Ant does, including switching to a new branch and building while not connected to the internet.

(And note I'm still waiting for the Maven mavens to reduce the verbosity of the CI logs so I don't waste so much time with them on my cell phone; maybe I should switch those back to Ant if that can't be fixed, maybe not, but in any case this is an example of why the new build tools need to be _demonstrated_ before they become replacements)

pabender, 2019-09-19:
Sent from my iPhone

> On Sep 19, 2019, at 9:31 AM, Bob Jacobsen <notifications@github.com> wrote:
> (And note I'm still waiting for the Maven mavens to reduce the verbosity of the CI logs so I don't waste so much time with them on my cell phone; maybe I should switch those back to Ant 
> 

Verbosity on the CI builds is different.  We have those set so we can figure out what tests cause timeouts.  

We can turn the test output messages off, but we will need to turn them on if test builds start hanging again.

pabender, 2019-09-19:
@bobjacobsen also, the main reason I would like to build using maven is it makes building modules much  simpler.  With ant,  I have to repeat much of the configuration for each  module.  With maven, build configuration is inherited, whichmeans I don’t have to repeat much.

bobjacobsen, 2019-09-19:
AppVeyor has over 2000 lines of "Downloading from central" at the start of every log. It also has over 700 lines about "SKIPPED" tests that don't appear in Ant-based builds.  This is completely separate from having test outputs on, which we currently don't do in AppVeyor anyway.

As to building modules:  I'm amazed that smart people like you and Randall can't understand the difference between addition and subtraction.  If you want to build modules via Maven, please go ahead.  That's great.  If you want to add that capability in ways that will work for e.g. other developers or will be used for building test or production releases, either deal with the Maven issues I've raised instead of dismissing them, or put it in Ant.  You're welcome to add things.  You many _not_ subtract them without a much better argument than has been made so far.

pabender, 2019-09-19:
@bobjacobsen I believe the skipped messages are also a part of the  surefire build results. PR. #7433. Should turn those off.

Also, to your point on building modules, I created a branch that uses the current file structure to build modules, but with the current structure it is a significant  amount of work to configure.  Largely due to the fact that the code wasn’t really structured to be built as smaller pieces.  We have talked elsewhere about  the  structural issues we have with code reaching across boundaries. 

Restructuring the code into  module directories needs to be done.  This will make it more difficult  for people to Reach across boundaries,  but  in order to enforce that, we need to have   all our builds building modules.  I can probably get ant to do that, but it is not an insignificant amount of work.

The other argument for moving to maven is that I want to attract more participation.  Ant  is being used much less than it used to be in industry  maven is much more commonly used.

rhwood, 2019-09-19:
Appveyor refuses to cache anything for PRs; so we need to use maven (installed on AppVeyor) to download ant (not installed on AppVeyor) every time AppVeyor runs. So, yes, we download dependencies in maven to run ant.

Then we run the `appveyor` target in ant. That target in ant is the cause of the verbosity of skipped messages in AppVeyor, not maven. Ant was possibly made verbose for the same reasons Travis CI was made verbose—so we could find which tests were randomly timing out instead of just shrugging our shoulders and restarting the CI builds without knowing why they failed.

And that’s the trade off; you have easy to scroll through logs that provide no clue towards cause of failure when the test times out, or you can have that clue, but you cannot have that clue *and* short logs.

Randall

bobjacobsen, 2019-09-19:
@rhwood You have been misunderstanding my point since 2017.  I understand about turning on the list of every test name when that's needed.  That is currently _OFF_ in AppVeyor.  But the hundreds of test-skipped messages remain, despite the messages about succeeded tests being _OFF_.

So enough. Do what you want with maven.  Find somebody else to restart tests.  I'm not going to waste any more time on this. But mess with the existing build support at your peril.  I am _not_ kidding about that.

rhwood, 2019-09-19:
I thought https://github.com/JMRI/JMRI/blob/29af90110c52a9b268bb9a2e7e9689140a97bcba/build.xml#L1553 meant those skipped messages were on in ant. If not, what does this line indicate should be printed?

pabender, 2019-09-21:
This obviously isn’t the right time to consider the original question, so I am going to close this issue.

closed 2019-09-21 by pabender