JMRI/JMRI #7404: Update our modularization design and start to enforce it?

bobjacobsen, 2019-09-08:
I think another discussion of JMRI's dependency graph and modularization is probably a bit overdue. (This came from Issue #7400, but that's really about something else)

Our original [dependency conventions](http://jmri.org/help/en/html/doc/Technical/IntroStructure.shtml) still look pretty good to me as a starting point, but perhaps we should discuss those.  

There's certainly been a lot of code added since those were created that needs consideration.

Within the `jmri` package, there are several sub packages that aren't previously discussed. What should be the structure and packaging of them? What can then depend on, what depends on them? Should they be moved to another place in the hierarchy?
   * `jmri.jmris`
   * `jmri.plaf`
   * `jmri.profile`
   * `jmri.progdebugger`
   * `jmri.script`
   * `jmri.server`
   * `jmri.spi`
   * `jmri.web`
 
In addition, there are a lot of places that our existing [conventions](http://jmri.org/help/en/html/doc/Technical/IntroStructure.shtml) haven't been followed.
   * Lots of implementation code in `jmri` classes that refer to `jmri.jmrit` and other low-level classes
   * Similarly, references from `jmri.implementation` and `jmri.managers` to (mostly) `jmri.jmrit` and `jmri.jmrix`
   * `apps` references in the `jmri` tree

I've been working on using [ArchUnit](https://www.archunit.org) to check for this kind of thing. See my [bobjacobsen/ArchUnit branch](https://github.com/bobjacobsen/JMRI/tree/ArchUnit) for that.  The number of violations is quite large, but we can certainly get started if that's the consensus.  

Finally, we will eventually want to organize JMRI around Java modularization  Which packages will go in which modules for distribution? We're far from implementing anything now (see above), but we should make sure that any organization we standardize on is compatible with that.  What, if any, separate distribution (including the parts outside the jar files) are we working toward?

bobjacobsen, 2019-09-08:
The current ArchUnit output is attached.  A lot of the flags are due to `jmri.server` and `jmri.web` references, so perhaps we can start by clarifying the appropriate packaging for those.

[ArchUnitOutput.txt](https://github.com/JMRI/JMRI/files/3588175/ArchUnitOutput.txt)

pabender, 2019-09-08:
Sent from my iPhone

> On Sep 8, 2019, at 7:21 PM, Bob Jacobsen <notifications@github.com> wrote:
> There's certainly been a lot of code added since those were created that needs consideration.
> 
> Within the jmri package, there are several sub packages that aren't previously discussed. What should be the structure and packaging of them? What can then depend on, what depends on them? Should they be moved to another place in the hierarchy?
> 
> jmri.jmris
> jmri.plaf
> jmri.profile
> jmri.progdebugger
> jmri.script
> jmri.server
> jmri.spi
> jmri.web

I can’t tell you why this changed, but the original intention behind the jmri.jmris package was that it would contain all servers ( the s stands for server ) and would provide an alternative front end to the tools provided in the jmri.jmrit package.  

As with the jmri.jmrit package, the jmri.jmris package is supposed to be drpendent only on the interfaces in the jmri package.

rhwood, 2019-09-09:
I’m in favor of moving to maven with or without modularization - simply put every IDE can support maven because it has well defined build phases, but no IDE can support ant (in the same way that no IDE supports make; NetBeans’s ant support is entirely predicated on creating an ant build.xml that has specific targets that NetBeans has defined the purpose and meaning of).

When I started working on JMRI, jmri.jmris did not support every server, just the SRCP and “simple” servers; which are the only two servers that successfully abstract out of the classes in jmri.jmris package; other servers were scattered between the jmri.jmrit and jmri.jmrix packages. When I rearchitected the JSON server, the old jmri.jmris.json package was deprecated (not completely) and the jmri.server.json package was written as its replacement.

The decision predates me, but I never understood why the package names jmri.jmris, jmri.jmrit, and jmri.jmrix were considered better than jmri.tool, jmri.server, and jmri.external to begin with. When I have been writing new classes; I have tried to mirror the structure and meanings of the java.* packages in the JDK (so jmri.plaf has the same meaning as java.plaf and jmri.bean has the same meaning as java.bean, etc...).

SonarLint and SonarJava can flag packages that should not depend on each other in maven, CI (not using Travis CI), and IDE integrated processes, so I think we can leverage those to begin disentangling JMRI internal dependencies.

Randall

pabender, 2019-09-09:
> When I started working on JMRI, jmri.jmris did not support every server, just the SRCP and “simple” servers; which are the only two servers that successfully abstract out of the classes in jmri.jmris package; other servers were scattered between the jmri.jmrit and jmri.jmrix packages. 

There is more history to that which largely goes back to the KAM lawsuit.  That stalled many of our networking projects for a number of years and as a result several independently developed packages were put into place at nearly the same time.  We have never gone back to rationalize all of that.

pabender, 2019-09-09:
Just a quick check on the jmri.jmris package.

It references jmri.jmrit for operations and XmlFile.  I believe the XmlFile reference may be something we can remove.  Removing the operations reference is, obviously, more problematic.

The SRCP server references jmri.jmrix.SystemConnnectionMemo because the protocol requires us to enumerate the buses to which it connects.  We have an equivalence between bus in the SRCP nomenclature and system connection in the JMRI nomenclature.

bobjacobsen, 2019-09-09:
`XmlFile` should probably move to `jmri.util` perhaps with some implementation in `jmri.util.jdom` (although JDOM2 is used as implementation in a lot of classes)

The `jmri.jmrit` tree was meant to hold various column-organized tools that didn't interconnect very much.  Right now, many of them are cross-connected.  Moving `Roster` and `RosterEntry` to the `jmri` package as basic objects would help, but after that it starts to get complicated.

Is it worth it to spend time on cleaning that up?  Depends on what the desired modularity is, and what goal it's try to achieve.  It is sufficient to have all our existing code fit into four jar files (`apps`, `jmri` and its basically implementations & utilities, `jmrix` and `jmrit`)? Or three, combining `jmri` and `jmrit` et al?

As to the package names:  Those came from some conventions specification document via Alex Shepherd a long, long time ago. Java was still 1.0.2, tool chains were a _lot_ less powerful, and we hadn't yet managed to get the jmri.org domain name. I don't recall the exact motivation for it, but I do remember it seeming to be reasonable at the time.  That also came with the convention of prefixed class names (i.e. `LocoNetMessage`, `NceMessage`, etc) that's slowly drifted away, so we can drift away from this one too if there's an articulated reason for it.

bobjacobsen, 2019-09-09:
I've added separate callouts for the `jmri.server` and `jmri.web` referencing `jmri.jmrit` or `jmri.jmrix` to the ArchUnit check. See appended. Briefly, `jmri.web` doesn't reference either; `jmri.server` has operations-related references to `jmri.jmrit`, but that's about it.

[ArchUnitOutput.txt](https://github.com/JMRI/JMRI/files/3588608/ArchUnitOutput.txt)

rhwood, 2019-09-09:
`jmri.web` has about 534 references to `jmri.jmrix` and `jmri.jmrit`; the most used parts of the JMRI Web server exist solely to provided an alternate HTML GUI interface to windows defined in `jmri.jmrit`, so the `jmri.web` packages should be allowed to refer to other portions of JMRI (the same is true of the JSON server; it exposes other parts of JMRI, so should not be barred from accessing them).

closed 2021-04-14 by stale[bot]