JMRI/JMRI #7407: Create Test Release 4.17.5

bobjacobsen, 2019-09-10:
This is the next release in the 4.17 cycle. It's intended to be created around October 12 from the `HEAD` of the `master` branch.

bobjacobsen, 2019-09-27:
Normally, test release 4.17.5 would be created from Head of master sometime Saturday Oct 12 or Sunday 13th.  Unfortunately, I have other obligations that make that difficult.  

I would therefore like to do the 1st build of 4.17.5 sometime mid-day Thursday Oct 10, i.e. about two days earlier.

Let me know if that causes problems for anybody, please.  

The alternative would be Wednesday Oct 16 or later.  I prefer to bring test releases earlier, rather than later, during this part of the cycle to keep the changes-per-release down. But we can go to the later date if needed.

Future release timeline, roughly:
 - 2nd week of November:  Last in pre-4.18 cycle for major changes
 - end of November: Last from master in pre-4.18 cycle
 - December: only special-purpose releases as we move toward a production release 
 - Around December 15 to 20: 4.18 production release.

dheap, 2019-10-10:
Issue #7389 needs to be resolved in this release. PR #7496 _or_ #7497 will resolve.

bobjacobsen, 2019-10-10:
Parts of Northern California, particularly including where some of our release-signing infrastructure is located, are currently without power. It’s not likely to return until tomorrow or later. That’s puts me past my build window today, so I’d like to make the first 4.17.5 files next Wednesday. My apologies for the last minute change.

bobjacobsen, 2019-11-01:
[image]

Log output during run:
[4.17.5.log.txt](https://github.com/JMRI/JMRI/files/3796730/4.17.5.log.txt)

bobjacobsen, 2019-11-02:
JMRI 4.17.5 has been released. Files are available in the GitHub release section.

closed 2019-11-02 by bobjacobsen