JMRI/JMRI #7411: Error loading vsd file in JMRI 4.17.4

theosdavis, 2019-09-11:
While attempting to load the default (and only) VSD file, I was presented with the message "Error Loading VSD File / Sound-Element is missing an audio file: <notch-sound."

klk32003, 2019-09-13:
Thanks for reporting this Error.
I can confirm this error and will provide a solution.

The VSD file you mentioned contains 4 profiles, 3 Diesel loco sounds and one Steam loco sound. The affected profile is called ```GE 7FDL```. You can download a VSD file named ```GE_7FDL.vsd``` (which contains that profile) [on the web](https://github.com/JMRI/vsdecoder), were you'll find the other 3 profiles as well.

klk32003, 2019-09-17:
@theosdavis 
The VSD file example.vsd is now updated.
It will be available with the next JMRI Test Release 4.17.5 at the usual place.

If you like, you can download it now from https://github.com/JMRI/vsdecoder and move it to your local directory.