JMRI/JMRI #7418: Troubles running scripts with non-Latin1 chars

svatoun, 2019-09-13:
Actually two issues:

1/ JMRI executes scripts from file by using `FileReader` with unspecified encoding (defaults to JMRI JVM encoding). If the script file is UTF-8, the characters will be mangled in FileReader as it misinterprets the bytes.

2/ If the script **directly executed by JMRI** contains `# coding=UTF-8` directive, the parser will fail because it gets a character stream out of the reader, which should NOT declare encoding.

If a script needs to invoke other UTF-8 written script, then that child script must have `# coding=UTF-8` interpreted by Jython. But because of issue (2), such script cannot be directly executed by JMRI.
Because of (1), one cannot use `u"žluťoučký kůň"` but must write `"žluťoučký kůň".decode("utf-8")`

I propose to:
a/ change default encoding of scripts read to UTF8
b/ since the `# coding` directive must appear on 1st or 2nd line of the script, the `JmriScriptEngineManager` could inspect the two lines if it contains the directive; and if so, create a FileReader again, with proper encoding + discard the directive's line.

After that, scripts can use unicode literals freely, can be called directly, or indirectly.

Willing to supply a patch.

rhwood, 2019-09-13:
There is a different problem (so far) unique to Windows with Jython and UTF-8.

Some editors on Windows insert a Byte Order Marker (BOM), when writing "UTF-8" and Jython fails on that (Python has a unique to python encoding "utf-8-sig" that covers UTF-8 with a BOM). See @bobjacobsen's comments in #3790 for how to make some Windows editors produce clean UTF-8.

If that is the problem you have encountered, I would prefer not to attempt to incorporate manipulation of the script input to work around it. I'm certain there is reason the Python developers determined they should reject UTF-8 with a BOM, and I'd rather teach script writers to avoid that than "magically" make it work for the initial script (and fail on imports that are handled outside JMRI's control).

svatoun, 2019-09-13:
No, this was produced by Linux text editor (kate). Verified there's no BOM present, just UTF-8 encoded file.

rhwood, 2019-09-13:
so the script fails to load from file, but if cut and pasted into the script input window works?

Can you provide a stack-trace of the failure?

What OS are you running JMRI on?

rhwood, 2019-09-13:
A better change to just how JMRI handles scripts would be to force a default encoding of UTF-8 in JMRI (it would solve so many problems).

Do note that Jython places limits on the max script size it will interpret from a string that differs from what it places on file readers, so that will have to be considered.

Also note that JMRI already has a mechanism to allow users to use a FileInputStream instead of a FileReader to read scripts for Jython. Have you tried the existing mechanism (outlined in python.properties in the JMRI distribution on your computer or https://github.com/JMRI/JMRI/blob/master/python.properties).

svatoun, 2019-09-13:
> so the script fails to load from file, but if cut and pasted into the script input window works?
> Can you provide a stack-trace of the failure?
Just garbled output:

- original script:
https://www.dropbox.com/s/cofmd823zxvtied/script-original.jpg?dl=0
- output when run the script file:
https://www.dropbox.com/s/aykye03p8j2lo8c/run-external.jpg?dl=0
- script input window with the file loaded (also uses default encoding)
https://www.dropbox.com/s/ccn5qk0pcil1zb0/loaded-external.jpg?dl=0
- this is the **correct** output, when contents is copy-pasted into the script input window
https://www.dropbox.com/s/fsrhh08ypqcpw77/output-copypasted.jpg?dl=0

A failure occurs when the script itself contains coding declaration: 
```
# coding=UTF-8
# HlavniProgram.py
import jmri
print u'spuštěno HlavniProgram.py'
```
When run from JMRI, produces
```
javax.script.ScriptException: org.python.antlr.ParseException: org.python.antlr.ParseException: encoding declaration in Unicode string
        at org.python.jsr223.PyScriptEngine.scriptException(PyScriptEngine.java:207)
        at org.python.jsr223.PyScriptEngine.compileScript(PyScriptEngine.java:88)
        at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:47)
        at javax.script.AbstractScriptEngine.eval(AbstractScriptEngine.java:249)
        at jmri.script.JmriScriptEngineManager.eval(JmriScriptEngineManager.java:324)
        at jmri.jmrit.jython.RunJythonScript.invoke(RunJythonScript.java:111)
        at jmri.jmrit.jython.RunJythonScript.actionPerformed(RunJythonScript.java:84)
        at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
        at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
        at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
        at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
        at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
        at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:842)
        at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:886)
        at java.awt.Component.processMouseEvent(Component.java:6533)
        at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
        at java.awt.Component.processEvent(Component.java:6298)
        at java.awt.Container.processEvent(Container.java:2237)
        at java.awt.Component.dispatchEventImpl(Component.java:4889)
```

> 
> What OS are you running JMRI on?
Ubuntu Linux

Re defaults: while defaults can be fixed by adding `file.encoding=UTF-8` to JMRI startup properties, I sort of think it is a disruptive change - it won't affect XML and .properties, but may adversely affect other files. 

The `# coding` directive is needed to allow invocation using `execfile` -- Jython interprets the directive to use appropriate encoding to load the file.

svatoun, 2019-09-13:
> Also note that JMRI already has a mechanism to allow users to use a FileInputStream instead of a FileReader to read scripts for Jython. 

Apologies, which mechanism ?

rhwood, 2019-09-13:
The mechanism is described in the references provided. Edit python.properties in your profile or settings directory as indicated in comments in that file.

Before making any changes to work around a Jython behavior also verify that current version of Jython does not have the problem you are dealing with. I don’t want to write code into JMRI to deal with issues fixed upstream.

svatoun, 2019-09-13:
> The mechanism is described in the references provided. Edit python.properties in your profile or settings directory as indicated in comments in that file.
I've read the python.properties, but did not found a setting relevant to encoding or source parsing (in that file). 

I found `python.console.encoding` (see https://wiki.python.org/jython/Setting) but that one is already hardcoded in JMRI already; no effect even when put into `python.properties` in JMRI install root. Perhaps it affects just jython interactive console ?

> Before making any changes to work around a Jython behavior also verify that current version of Jython does not have the problem you are dealing with. I don’t want to write code into JMRI to deal with issues fixed upstream.

The current version of Jython (jython.org website) is `Jython is 2.7.1`; verified that the same exception as reported above is thrown by `Jython 2.7.1` when `# coding` appears in **character stream** (`ScriptEngine.eval()`) -- it makes even sense: since the API-provided input is already decoded into Unicode (java) String, it shouldn't contain any encoding information that could possibly clash.

rhwood, 2019-09-15:
JMRI includes a *Java* property, *not a Jython* property that would force JMRI to directly pass the raw byte stream to the Jython interpreter. Did you test that before writing #7423 (since you changed that behavior in that PR)?

The mechanism is setting [jython.exec=true](https://github.com/JMRI/JMRI/blob/3da6cc7fcac1624e3fc61e03a2a6e1b536564853/python.properties#L42) in the file `/path/to/your/settings/python.properties`. Did you test that JMRI without #7423 and that setting still exhibits your behavior?

svatoun, 2019-09-15:
Thanks for watching the PR, @rhwood. Did not originally, did after your comment. In **both cases** the console output. The direct Jython call correctly processes the encoding - verified by passing UTF8 file without appropriate `# coding`: the parser reported an exception with a note about PEP-263:

```
2019-09-15 14:50:41,279 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
SyntaxError: Non-ASCII character in file '<iostream>', but no encoding declared; see http://www.python.org/peps/pep-0263.html for details
```

I guess I should revert the changes in direct Jython calls and only change the processing through `ScriptEngine`.

What's your opinion on defaulting the reader to UTF-8 encoding for direct Jythons ?

The other alternative is just opposite: force ASCII input, which is the default, according to Python PEP-263, Implementation phase 2 ... which would require all non-ASCII scripts to contain appropriate encoding directive.

rhwood, 2019-09-15:
My suggestion is that we open a discussion with the Jython developers about this and implement their recommended method for handling this within a JSR 233 ScriptEngine (I want to do that to allow JMRI itself to inject core objects into the Jython context instead of relying on a pre-processed Jython script to do that.)

svatoun, 2019-09-15:
> My suggestion is that we open a discussion with the Jython developers about this and implement their recommended method for handling this within a JSR 233 ScriptEngine

I don't think it's possible implementation-wise: `ScriptEngine` API accepts only `Strings` or `Readers`, no raw bytes, if I read correctly. Jython cannot "switch" encoding in an existing Reader - there's no API for that, nor for extracting raw bytes (and re-decoding) once the declaration is read.

As long as JMRI continues to use standard `ScriptEngine`, it has to "somehow" ensure the Reader produces correct character stream.

bobjacobsen, 2019-09-15:
> As long as JMRI continues to use standard ScriptEngine, it has to "somehow" ensure the Reader produces correct character stream.

Originally Jython wasn't in JMRI as a standard `ScriptEngine`; JMRI didn't include any support for `ScriptEngine`s. Jython was just directly embedded in a somewhat hacky manner.

@rhwood added the `ScriptEngine` support, which is great because it allows use of JavaScript and perhaps eventually other languages. As part of that he changed Jython to being supported as a `ScriptEngine`.

But if there are advantages to including Jython _not_ as a `ScriptEngine`, i.e. via custom code, that can certainly be discussed. There are also disadvantages, i.e. increased code volume and complexity, which have to be weighed in the balance.

svatoun, 2019-09-15:
@bobjacobsen I wasn't proposing to increase custom code interfacing w/ Jython. In fact, the `jython.exec` property value `true` in `python.properties` (which enables the branch with Jython direct call) is not default+recommended (false is recommended in the docs).

When used through `ScriptEngine` interface, Jython cannot handle non-ASCII (non-default) encodings in the directly executed file - this is what I am trying to solve. One then cannot use Python-native support for Uncode (`u"unicodeString"`) but must use awkward `"unicodeString".decode("UTF-8"`)

svatoun, 2019-10-30:
Negative.

Original script:
```
print u'Žluťoučký kůň pěl ďábelské ódy'
```

Executing using PythonInterpreter, exception:
```
2019-10-30 23:13:14,049 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
SyntaxError: Non-ASCII character in file '<iostream>', but no encoding declared; see http://www.python.org/peps/pep-0263.html for details

	at org.python.core.Py.SyntaxError(Py.java:200)
	at org.python.core.ParserFacade.fixParseError(ParserFacade.java:105)
	at org.python.core.ParserFacade.parse(ParserFacade.java:190)
	at org.python.core.Py.compile_flags(Py.java:2185)
	at org.python.util.PythonInterpreter.execfile(PythonInterpreter.java:296)
	at org.python.util.PythonInterpreter.execfile(PythonInterpreter.java:291)
	at jmri.script.JmriScriptEngineManager.getEngineOrEval(JmriScriptEngineManager.java:419)
```

PythonInterpreter, with coding directive
```
Å½luÅ¥ouÄkÃ½ kÅ¯Å pÄl ÄÃ¡belskÃ© Ã³dy
```

Executing using ScriptEngine, output:
```
Žluťoučký kůň pěl ďábelské ódy
```

ScriptEngine, run with coding directive - exception:
```
javax.script.ScriptException: org.python.antlr.ParseException: org.python.antlr.ParseException: encoding declaration in Unicode string
	at org.python.jsr223.PyScriptEngine.scriptException(PyScriptEngine.java:226)
	at org.python.jsr223.PyScriptEngine.compileScript(PyScriptEngine.java:107)
	at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:64)
	at javax.script.AbstractScriptEngine.eval(AbstractScriptEngine.java:249)
	at jmri.script.JmriScriptEngineManager.eval(JmriScriptEngineManager.java:394)
	at jmri.script.JmriScriptEngineManager.eval(JmriScriptEngineManager.java:331)
```

Both engines, script invoked from "master|" script using execFile(), exception:
```
2019-10-30 23:09:51,518 jython.RunJythonScript                ERROR - Unable to execute script. [AWT-EventQueue-0]
javax.script.ScriptException: Traceback (most recent call last):
  File "<script>", line 5, in <module>
  File "<string>", line None
SyntaxError: Non-ASCII character in file '/space/src/jmri/fixes/JMRI/jython/test2.py', but no encoding declared; see http://www.python.org/peps/pep-0263.html for details

	at org.python.jsr223.PyScriptEngine.scriptException(PyScriptEngine.java:231)
	at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:59)
	at org.python.jsr223.PyScriptEngine.eval(PyScriptEngine.java:64)
	at javax.script.AbstractScriptEngine.eval(AbstractScriptEngine.java:249)
	at jmri.script.JmriScriptEngineManager.eval(JmriScriptEngineManager.java:394)
```
Tested on: 8c99e77554

rhwood, 2019-10-30:
Are the scripts you are attempting to use in UTF-8?

svatoun, 2019-10-31:
For the reference:
[test.zip](https://github.com/JMRI/JMRI/files/3795627/test.zip)
2 test files residing in `./jython`

closed 2021-04-14 by stale[bot]