JMRI/JMRI #7431: Possible issue with CAN sensors, simulation and NX panels

bobjacobsen, 2019-09-19:
First [jmriusers post](https://groups.io/g/jmriusers/message/164038):

> I have tried to set up some NX routing, but have found that JMRI crashes after I click the first NX sensor. I know NX routing works because I saw it demonstrated successfully at a cource run by Pete Brownlow a few months ago, I have put a simple example which should demonstrate reasonably succinctly what I am doing wrong in the files area under "NX Routing".

[Reply](https://groups.io/g/jmriusers/message/164043) by @dsand47 

> Are you doing your testing using CAN Simulation?
> 
> I was able to re-create the problem using CAN simulation.  When I activate a route, JMRI hangs waiting for a response from a CAN command.
> 
> If I move T1 and T2 to internal turnouts, the NX routing works as expected.
> 
> Sorry, but CAN issues are outside of my knowledge set.  If someone does work on this, I can provide a stack trace.

To me, this sounds like the CAN simulator is omitting a state-transition, or perhaps notification of that state transition.

dsand47, 2019-09-19:
The NX definition is setting two turnouts.  The current line in DestinationPoints is invoking **setCommandedState()** for one of the turnouts.  

[stack_trace.txt](https://github.com/JMRI/JMRI/files/3629311/stack_trace.txt)

icklesteve, 2019-09-19:
I'm not over-familiar with NX though it's something I've been meaning to play with.

Major thanks for the stack trace @dsand47 , points towards CbusEventTableDataModel
I've got it working ( AFAIK ) by locally disabling the CbusEventTable processing of the outgoing Frame.

The difference in the frames, top one is sent by normal Turnout Table, bottom 2 sent by NX

> 2019-09-19 14:50:48,092 eventtable.CbusEventTableDataModel    INFO  - processing message [5f8] 98 00 00 00 65 [AWT-EventQueue-0]
> 2019-09-19 14:51:43,671 eventtable.CbusEventTableDataModel    INFO  - processing message [5f8] 99 00 00 00 65 [Entry Exit Set Route]
> 2019-09-19 14:51:43,935 eventtable.CbusEventTableDataModel    INFO  - processing message [5f8] 98 00 00 00 64 [Entry Exit Set Route]

I'm working down the message chain to see how to resolve the thread issue but might not get a chance on this for a few days, some opportunities for tidying, especially around the setValueAt

Steve

icklesteve, 2019-09-28:
I've just had a chance to do a little more digging around this.

The main issue I'm having is that calls to fireTableCellUpdated and fireTableRowsInserted only work from the Event Dispatch Thread.
From the Entry Exit Set Route thread, only calls to fireTableDataChanged work, and must be without jmri.ThreadingUtil.runOnGUI

If I change ( current code causing freeze )

```
        ThreadingUtil.runOnGUI( ()->{
            fireTableCellUpdated(row, col);
        });
```

 to

```
        fireTableDataChanged();`
```

or

```
        if (javax.swing.SwingUtilities.isEventDispatchThread()) {
            ThreadingUtil.runOnGUI( ()->{
                fireTableCellUpdated(row, col);
            });
        } else {
            javax.swing.SwingUtilities.invokeLater(new Runnable() {
                public void run() {
                    fireTableCellUpdated(row, col);
                }
            });
        }
```

the NX thread works fine.

```
        ThreadingUtil.runOnGUI( ()->{
            fireTableDataChanged();`
        });
```

does NOT work.

Bear in mind that there are 25 columns with rows potentially in the 1000's so I'm keen to only update relevant cells.

Is there anything I've missed?  I'm inclined to go with the SwingUtilities check unless there's a better way of dealing with this?

rhwood, 2019-09-28:
Have you tried using `ThreadingUtil.runOnGUIEventually(() -> fireTableDataChanged());`?

icklesteve, 2019-10-21:
I have now thanks @rhwood , much appreciated.

icklesteve, 2021-03-12:
Closing issue pending further ocurrence.
Improved code for has been in place for a while now.

closed 2021-03-12 by icklesteve