JMRI/JMRI #7434: Programming ESU LokPilot

theosdavis, 2019-09-19:
I am unable to add Physical Outputs "Aux3" or "Aux4" to any row on the Function Output table in a programmer (including Service Mode Programmer).  I enter the values, write changes, and the write consistently fails, wherever I try them. Might this be a JMRI/ESU compatibility issue?
Using Digitrax PR3 and DCS50 with JMRI 4.17.4.

dheap, 2019-09-19:
@theosdavis Exactly which ESU LokPilot model? There are many ESU LokPilot models, in different families.

Did you pick the model manually or by Read Type from Decoder? With modern ESU decoders its essential to let JMRI do the choosing.

Please give the exact "Decoder Family" and "Decoder Model", as shown on the Roster Entry pane when you have a roster entry open.

There are no known ESU/JMRI compatibility issues.

theosdavis, 2019-09-20:
Programmer in JMRI chooses Family ESU LokPilot v4.0, Model LokPilot Micro 4.0 DCC.
I have LokPilot V4.0 Micro DCC w/Next 18.

dheap, 2019-09-20:
@theosdavis If you look at the technical specifications for this decoder version (ESU part 54686):
http://www.esu.eu/en/products/lokpilot/lokpilot-micro-v40-dcc/lokpilot-micro-v40-dcc-technical-data/

Plus the LokPilot V4.0 6thEdition  manual, in particular the Next18 version (pages 7 and 14) from:
http://www.esu.eu/en/downloads/instruction-manuals/digital-decoders/

Plus the LokProgrammer software, you will see that these decoders support 2 normal outputs (Front Light and Rear Light), possibly Aux1 and Aux2 as logic output to another board and possibly Aux3 and Aux4 by external I/O expansion board.

The CVs for outputs unsupported by your particular hardware configuration will always read back  as 0, as will any other unsupported CV in ESU decoders.

Both JMRI and LokProgrammer provide settings for all possible outputs as it is not possible/easy for JMRI/LokProgrammer to determine exactly how many outputs your hardware configuration supports.

If your JMRI Preferences->Roster->Roster pane has "When possible, Confirm CV writes by following them with a read", you will get Write errors for any of the CVs not supported by your particular hardware configuration. You need to turn that off when programming ESU decoders (or at least when trying to write all CVs).

theosdavis, 2019-09-20:
It appears, then, that I can find no way for the rear ditch lights of my Scaletrains N T4 to light.  This article, https://www.trovestar.com/generic/blog.php?Article=286, with its suggestion configuration of the Function Outputs in JMRI, suggests otherwise, but I can't seem to replicate it.

dheap, 2019-09-21:
@theosdavis Thanks for that link. @AlanUS is a well-known contributor to JMRI decoder definitions, translations etc.

@AlanUS has provide a downloadable [JMRI to ESU CV import file](https://www.trovestar.com/Assets/Collections/4/DCC/ScaleTrains_GE_Tier_4_GEVO_LokPilot_V4_ALM.txt) as well as an [ESU LokProgrammer file](https://www.trovestar.com/Assets/Collections/4/DCC/ScaleTrains%20GE%20Tier%204%20GEVO%20BNSF%20%28LokPilot%29-ALM.esux), both of which I downloaded.

I created a LokPilot Micro 4.0 DCC definition in JMRI, opened it in Program Track Mode and used the File->Import->Lokprogrammer CV List file option available when the entry is open to import Alain's [ScaleTrains_GE_Tier_4_GEVO_LokPilot_V4_ALM.txt](https://www.trovestar.com/Assets/Collections/4/DCC/ScaleTrains_GE_Tier_4_GEVO_LokPilot_V4_ALM.txt) file, then saved the Roster Entry.

I also opened the [ESU LokProgrammer file](https://www.trovestar.com/Assets/Collections/4/DCC/ScaleTrains%20GE%20Tier%204%20GEVO%20BNSF%20%28LokPilot%29-ALM.esux) in LokProgrammer, for comparison purposes only.

Examination and comparison of the JMRI & LokProgrammer Function Map and Function Outputs panes revealed full correspondence and reasonable settings in both (with a few peculiarities I'll describe later).

I don't have access to a Next18 LokPilot Micro 4.0 DCC decoder, but I do have a 21MTC LokPilot V4.0 in my test collection so I fitted that to my ESU decoder tester and did a Write Full Sheet on the CVs pane, followed by a Compare Full Sheet, (all matched).

Testing of the actual decoder indicated working ditch lights (with a few anomalies due the peculiarities of Alain's mapping):

**F6 Off:**
- Direction FWD: All ditch lights initially off. Front ditch lights come on and flash while F2 is held down but extinguish immediately the horn is released, with no Hold Time. 
- Direction REV: All ditch lights initially off. Front ditch lights come on and flash while F2 is held down but extinguish immediately the horn is released, with no Hold Time. Rear ditch lights are not active.

**F6 On:**
- Direction FWD: Front ditch lights initially on. Front ditch lights flash while F2 is held down and continue to flash when the horn is released, for the 5 second Hold Time specified in Function Settings.  Rear ditch lights are not active.
- Direction REV: Rear ditch lights initially on. Front ditch lights come on and flash while F2 is held down but extinguish immediately the horn is released, with no Hold Time. Rear ditch lights flash while F2 is held down and continue to flash when the horn is released, for the 5 second Hold Time specified in Function Settings.

I've created a modification of Alain's settings by removing the Aux 1[1] and Aux 2[1] mappings from the line activated by F2. This produces what I consider a better result. If you don't want to make the change manually, download and unzip this CSV file and use File->Import->Lokprogrammer CSV file:
[ScaleTrains_GE_Tier_4_GEVO_LokPilot_V4_ALM_Fix1.zip](https://github.com/JMRI/JMRI/files/3638374/ScaleTrains_GE_Tier_4_GEVO_LokPilot_V4_ALM_Fix1.zip)

Many other variations are possible. Tell us what you want.

Given that @AlanUS has created and (I assume) tested the original Lokprogrammer CV List file on a Next18 LokPilot Micro 4.0 DCC decoder, there must be some issue with the interaction between JMRI and your Digitrax PR3.

All my systems (several different NCE and SPROG IIv3) always do blind writes (no waiting for an ACK from the decoder) and I always have "When possible, Confirm CV writes by following them with a read" turned off.

You need to disable "When possible, Confirm CV writes by following them with a read" in your JMRI settings. If you still get write errors that break the bulk write process other than just flagging errors (I can't create write fails with any of my DCC systems), you'll need some advice from the Digitrax experts about doing "blind writes". I think that involves using Program on Main to write the Function Outputs pane.

I suspect your PR3 is creating errors when writing to Aux 5 and greater. Since examination of the Lokprogrammer software reveals that the LokPilot Micro 4.0 DCC doesn't support Aux 5 and beyond, I'll change the JMRI definitions to account for that.

dheap, 2019-09-21:
@theosdavis @AlanUS I've updated the definition to accomodate this and also updated my suggested fix file. See #7438 
[ScaleTrains_GE_Tier_4_GEVO_LokPilot_V4_ALM_Fix1.zip](https://github.com/JMRI/JMRI/files/3638613/ScaleTrains_GE_Tier_4_GEVO_LokPilot_V4_ALM_Fix1.zip)

Once merged, the fix will be available from  the [developer CI builds](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/).

dheap, 2019-09-21:
@theosdavis JMRI installers incorporating the fix are now available from the [developer CI builds](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/) link.

Although the changed definition will no longer show Aux 5 to Aux 10 on the Function Outputs page, an existing roster entry  will still have the not-used CVs stored and shown in the CVs pane, so its best to create a new roster entry using the fixed version.

dheap, 2019-09-23:
@theosdavis Have you had a chance to try out the new JMRI definition as per this https://github.com/JMRI/JMRI/issues/7434#issuecomment-533827567?

I'd like to know if it helps with your problem.

theosdavis, 2019-09-23:
Sadly, still the same results: no rear ditch lights.
It appears, at least with my setup (JMRI 4.17.5ish>PR3>DCS50>LokPilot 4.0 Next18), I am unable to write the correct values for mapping Aux3&4 (CV 16.2.394) and Output settings (CVs 16.0.291 and up).  Similarly, for CV 47 I am unable to get the correct setting of "8" to write (Keeps setting itself to 4).

dheap, 2019-09-23:
@theosdavis Could you please give us some further information?

What error message do you get when you try to write these CVs?

A copy of your session.log file (found in Help->Locations->Log Files Location) would be helpful.

Also please Read Full Sheet on the Information pane and get us a screen grab. I'm particularly interested in the Firmware version (4.x.x) and the Product ID.

I've seen some bad things with incorrect CV read/write (affecting LokProgrammer as well as JMRI) for some V4 decoders, including the LokPilot V4.0 I was using for testing this issue. A firmware update (needs LokProgrammer) cured the problem, although it corrupted my loaded CV settings in the process. I had to reset them using Alan's files. It gave no further trouble.

theosdavis, 2019-09-23:
Aha, that sounds like a possible explanation. 
After entering values, I would then 'compare' and receive the message "Confirm failed (305)"

[session.log](https://github.com/JMRI/JMRI/files/3643873/session.log)
[image]

theosdavis, 2019-10-02:
Perhaps I need to find someone nearby with a LokProgrammer?

dheap, 2019-10-03:
> @theosdavis Perhaps I need to find someone nearby with a LokProgrammer?

Yes, to install a firmware upgrade.

closed 2021-04-14 by stale[bot]