JMRI/JMRI #7435: Excessive output on Appveyor CI builds

pabender, 2019-09-20:
AppVeyor has over 2000 lines of "Downloading from central" at the start of every log. It also has over 700 lines about "SKIPPED" tests that don't appear in Ant-based builds.  This is completely separate from having test outputs on, which we currently don't do in AppVeyor anyway.

_Originally posted by @bobjacobsen in https://github.com/JMRI/JMRI/issues/7400#issuecomment-533143921_

pabender, 2019-09-20:
@bobjacobsen 

> AppVeyor has over 2000 lines of "Downloading from central" at the start of every log. It also has over 700 lines about "SKIPPED" tests that don't appear in Ant-based builds. This is completely separate from having test outputs on, which we currently don't do in AppVeyor anyway.
> 

I am looking into the output on appveyor.

The 2000 lines of download information are pretty quick to get rid of.  as part of #7433, I have now added the --quite flag to the maven command that launches ant.  That means that maven itself will only produce output if it has an error.

So now we need to diagnose where the excessive output is coming from (and it is excessive) within the ant builds.

Here are a couple of observations (from running locally, not on appveyor yet):

First, not all of the ant test tasks are the same.  If you just compare  `ant alltest` and `ant ci-test-appveyor` , `ant alltest` runs the application apps.tests.Alltest but 'ant ci-test-appveyor' is actually running the ant.tests.Alltest JUnit test suite using the  JUnit ant task.  There doesn't appear to be any explicit control over output for the `alltest` task, but the `ci-test-appveyor` task specifies an output formatter of "brief".  The brief formatter appears to interact with the printsummary and fork parameters to the junit task as well.
 
Second, It looks like most of the tests that are reported as "SKIPPED" are tests where an assumption failure occurs.

pabender, 2019-09-20:
The SKIPPED message are being generated by the brief formatted specified in the ‘ci-test-appveyor’ task:

https://github.com/apache/ant/blob/451ecde5c915d94fdb6c5c8e59b291868290bc14/src/main/org/apache/tools/ant/taskdefs/optional/junit/BriefJUnitResultFormatter.java#L279-L300

pabender, 2019-09-20:
@bobjacobsen 

#7433 has reduced the output on Appveyor from somewhere around 8000 lines to just under 2000 lines.  

This was accomplished by 

1. removing the brief output formatter from the ant ‘ci-test-appveyor’ task, which generated the Skipped lines for about 730 tests
2. Setting the —quiet flag on the Maven call in ‘appveyor.yml’ that starts the ant build, removing the logs of ants download activity.

Most of the remaining lines of output are from instances of #7184, which are triggered by our test infrastructure attempting to close windows after each test.  There is no immediately obvious fix for that issue.

rhwood, 2019-09-21:
It appears that the artifacts for AppVeyor are only uploaded if the unit tests pass, such that the ant logs are not available on test failures.

pabender, 2019-09-21:
> It appears that the artifacts for AppVeyor are only uploaded if the unit tests pass, such that the ant logs are not available on test failures.

The artifact uploading is handled by some magic in Appveyor.  All we do is tell it which artifacts to upload.  It looks like asking to upload the test results as an artifact won’t work

We have ( commented out ) the code required to upload test results to Appveyor.  This process seems to be triggered reliably even when a test fails, but either the upload fails midstream or the display is unreliable.  

I would like to have Appveyor upload only failed test results, but I haven’t been able to get the failure output formatter for the ant task to work, and that seems to be the way to do this.  Barring that approach, a custom output formatter will need to be created.

rhwood, 2019-09-21:
Found https://github.com/appveyor/ci/issues/401 which might help.

My understanding is that AppVeyor, and other CI/CD systems vendors, tends to think of artifacts as "items to deploy upon a successful build", not as "items to retain on a failed build", so using their default mechanisms would likely not work to retain logs of failures.

pabender, 2019-09-21:
Ok.  #7440 hopefully will push those results on failure.  I am going to restart that until we see a failure.

rhwood, 2019-09-21:
If you want to force a failure on AppVeyor, submit #7440 as a PR against [my scriptengine branch](https://github.com/rhwood/JMRI/tree/scriptengine) (the source for #7432), which I am expecting to fail on AppVeyor.

pabender, 2019-09-21:
@rhwood no need for that, it failed on the first attempt.  
https://ci.appveyor.com/project/JMRI/jmri/builds/27570053

It looks like the results were uploaded, but I have not looked at contents yet.  It will still be easier to figure out what failed if the test results are uploaded to the right place.

pabender, 2019-09-22:
I’ve got the logs ending with this on a failure now:

```
2106classname                                                       name                             InnerXml                                                                                                                            
2107---------                                                       ----                             --------                                                                                                                            
2108jmri.implementation.DefaultConditionalTest                      testTriggers                     <failure message="Unexpected ERROR or higher messages emitted: Cleaning up nameless invisible frame created by creating a dialog ...
2109jmri.jmrix.bachrus.SpeedoDialTest                               testShow                         <failure message="Unexpected ERROR or higher messages emitted: Cleaning up frame "SpeedoDial test frame" (a class jmri....
2110jmri.jmrix.bachrus.SpeedoConsoleActionTest                      testAction                       <failure message="Unexpected ERROR or higher messages emitted: Cleaning up frame "Speedo Console" (a class jmri.jmrix.b...
```

( this was produced on a branch where I turned on the errors for not closing windows in tests ).

The output was produced with a little bit of PowerShell scripting on the junit xml Test results.  I would like to pull the message out of the inner xml, but I have more than exhausted my knowledge of xml parsing in PowerShell now, so I am going to leave that for someone more knowledgeable on that subject than I am.

The classname and the name give you the fully qualified name of the test method that failed.  The InnerXml  is the XML contained inside the Testcase ( I.e. the children of the testcase node in the XmL ).  The result is selecting test cases with failure children, so all we should see is test cases that failed.  The failure node includes the message from the failed assertion.  

( it is possible to have two failures on one test, usually when a test emits an error message and then has an assertion failure... when tearDown is executed an assertion will fail due to the error message ).

pabender, 2019-12-07:
Closed via #7441

closed 2019-12-07 by pabender