JMRI/JMRI #7448: Issue with Withrottle when using Fkeys and JMRI 4.16, 4.17.4 or Jenkins 4.17.5

alain355, 2019-09-24:
Hello dear team

Today I found an issue with I think the Withrottle server: When I take control of one lok (equipped with a Zimo MX64) and use the WiThrottle Fkey, on some Fkeys already defined, it didn't correspond to what is programmed within the decoder (it works OK when Fkeys keyed in from my ECoS2 Command station).
I've reread the decoder Alternate Mapping which is as defined before (and respond correctly to Fkeys when pushed directly from my ECoS Command station).

Of course I have the same issue when using Engine Driver.

I have the same issue using 4.17.4 Beta, current Jenkins JMRI.4.17.5ish (20190924T0152Z+Rf4d06dc) or with 4.16 stable.

Attached are latest sessions
[4.17.5_session.log](https://github.com/JMRI/JMRI/files/3646249/4.17.5_session.log)
[4.16_session.log](https://github.com/JMRI/JMRI/files/3646250/4.16_session.log)

Regards from Paris

Alain

BHoffman351, 2019-09-24:
Hi Alain,

Just to clarify:
You press a function and a different function in the decoder responds, not the labels on the function buttons are incorrect after a remap?

Does the on-screen JMRI throttle also behave this way?

Regards,
Brett

> On Sep 24, 2019, at 5:01 AM, alain Carasso <notifications@github.com> wrote:
> 
> it didn't correspond to what is programmed within the decoder (it works OK when Fkeys keyed in from my ECoS2 Command station).
> I've reread the decoder Alternate Mapping which is as defined before (and respond correctly to Fkeys when pushed directly from my ECoS Command station).

alain355, 2019-09-24:
Hi Brett
Thx for your reply. It's not a label issue but a Fkey issue i think:
Let's say I push F5 on the JMRI Throttle or on Engine Driver: the responding result is not what I have when pushing the F5 from my command station. It's even either something not programmed within the decoder in a single Fkey, or nothing at all.
I have re-read again the Alternate F key function map from the decoder and it's still as defined as wanted (and as responding to my ECoS Command station).

Alain

alain355, 2019-09-24:
My mapping is defined as below:
F0 Fwd to Output1
F0 Rev to Output2
F1 to Output3
F2 to Output4
F3 shunt
F4 BEMF
F5 to output 1 and 2
F6 to output 3 and 4
F7 to output 1, 2, 3 and 4.

alain355, 2019-09-24:
I repeated the test using another lok with an ESU Lokpilot V4, no issue. Then again the lok with the Zimo MX64, and checking results on the ECoS F display: same kind of issue as before
F0, F1 F3 and F4 OK, but F2 activates F5, F7 activates F6 and F5 activates F2.
I think I'll delete the lok then recreate it again. It may be a decoder definition issue as there's no problem with the Lokpilot V4, only with this Zimo :( MX64

alain355, 2019-09-24:
I found the issue: it's linked to the saved decoder definition:
I created the lok duplicating an existing MX64 definition, as the decoder is an MX64, but:
>> All my MX64 (4 of them) have been updated to sw release V34. However there's no V34 definition for MX64. Thus the roster entry is based using a Zimo V22 to 30 definition. That's where I defined the mapping (and it works if using it from my command station).

If using JMRI to detect a new lok, of course it doesn't propose any MX64 as SW version is 34. So I used instead an MX640 definition (which will be partly OK only, a lot of CVs are missing). With that one the mapping is responding to what is ordered both from JMRI Throttle and from Engine driver.