JMRI/JMRI #7454: Layout Editor - odd tag <points />

sidlo64, 2019-09-26:
How to reproduce:

Start PanelPro, **create new panel** Layout Editor, Save panel as point.xml, exit PanelPro.

Start PanelPro, open panel point.xml, **add Shape** by one click point, Save panel, exit PanelPro.

Start PanelPro, open panel point.xml, select shape point and **Delete it**, Save panel, exit PanelPro.

Start PanelPro, **open panel** point.xml, an error message is displayed

```
2019-09-26 11:58:18,009 configurexml.ErrorHandler             ERROR - Parse error while parsing file /home/sidlo/.jmri/Moje_JMRI_železnice.jmri/point.xml Exception: org.jdom2.input.JDOMParseException: Error on line 59: cvc-complex-type.2.4.b: The content of element 'points' is not complete. One of '{point}' is expected.
See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
org.jdom2.input.JDOMParseException: Error on line 59: cvc-complex-type.2.4.b: The content of element 'points' is not complete. One of '{point}' is expected.

```

Odd tag element **«points /»** in xml file:

```
<layoutShape ident="S1" type="Open" level="3" linewidth="3" lineColor="#000000" fillColor="#404040" class="jmri.jmrit.display.layoutEditor.configurexml.LayoutShapeXml">
  <points />
</layoutShape>

```

bobjacobsen, 2019-10-30:
This has come up a couple times on the user list. It's causing people who get into this situation to have unloadable panel files.

We should either:
 - Change the XML Schema to allow a `<points>` element to have zero `<point>` elements, or
 - Change the writing code to never write an empty `<points>` element

geowar1, 2019-10-30:
I’m willing to do this work… but not until #7482 is merged.

geowar1, 2019-10-31:
See PR #7557

geowar1, 2019-11-01:
PR #7557 has been merged… ;-)

geowar1, 2019-11-04:
@sidlo64 If you can verify this fixed by PR #7557 this can be closed.

geowar1, 2019-12-31:
I'm going to assume this is fixed and close it. If you can reproduce it (post PR#7557 merge) please reopen and I'll investigate further. TIA.

closed 2019-12-31 by geowar1