JMRI/JMRI #7457: Add Turnout / Light Hardware Type select

icklesteve, 2019-09-26:
With NO panel loaded and MERG CBUS Sim connection in use, navigate to Turnout Table, select the table tab for MERG, click New Turnout and Internal is not available in the hardware type select.

Vice versa, select the Internal tab and the MERG connection is not available in the select.

[image]

Both Internal and MERG are available in the select when the table has the All tab selected.

bobjacobsen, 2019-09-26:
I'm confused.  What would you expect to happen instead?

pabender, 2019-09-26:
I agree with @bobjacobsen this is the right behavior for a context sensitive dialog.

icklesteve, 2019-09-26:
Previously all hardware types were available to select, see the current Light Table behaviour for this.

bobjacobsen, 2019-09-26:
But that's what "all" does.  If all the types where available on every tab, how do you expect the tabs to differ?

icklesteve, 2019-09-26:
The tab content is fine, no issue.
I wanted to view the CBUS Turnout tab while creating internal Turnouts so I could use a similar naming format.
Previously in the New Turnout hardware type select dropdown you could select a different system type than the selected tab.
This is no longer the case.

bobjacobsen, 2019-09-26:
Got it, thanks.  v4.14 behaves as you describe (you can select any type in "add" on each of the tabs)

bobjacobsen, 2019-09-26:
(Unintended close, buttons on phone very small)

bobjacobsen, 2019-09-26:
v4.16 also seems to show all types.  A [git bisect](https://www.jmri.org/help/en/html/doc/Technical/GitFAQ.shtml) might find where it changed pretty quickly.

rhwood, 2019-09-26:
It was likely me that made the change. I’ll find a way to restore the former behavior, but have the selected system match the tab that was selected when the user opened the add window.

Randall

icklesteve, 2019-09-26:
Thanks @rhwood , the Sensor Table still has the expected previous behaviour, it's the Turnout / Light / Reporter Tables which don't allow you to make the selection,

Steve.

PR #7467 (2019-10-05): fix: Allow connection selection from any add hardware dialog
Fixes #7457 by combining four similar blocks of code (in which three were not behaving correctly) into a single method and replacing the blocks with calls to the method.
commit 961abe2d (2019-10-05): fix: Allow connection selection from any add hardware dialog

Fixes #7457
closed 2019-10-05 by bobjacobsen