JMRI/JMRI #7463: Recommend that we raise CodeClimates thresholds

bobjacobsen, 2019-09-27:
Prompted by @rhwood's point in PR #7460, I looked into a specific case of the CodeClimate (henceCC) thresholds. 

The first CC flagged item in that PR points to this file:  [jmri/jmrix/rps/swing/CsvExportMeasurementAction.java](https://codeclimate.com/github/JMRI/JMRI/java/src/jmri/jmrix/rps/swing/CsvExportMeasurementAction.java/source?from_sha=70e9e900&to_sha=eb3ea91f)  

I recommend people take a look at that file.  (Not sure how long links into CC will remain good, unfortunately).   CC finds in it four instances of "similar blocks of code found in _2_ locations" (emphasis added) that are, at most, less than four lines long.  It also finds that a pretty straight-forward routine that handles one thing with two sub-cases (provide a default file chooser; only process the file if a file was actually chosen) plus thrown-exception handling 'has a Cognitive Complexity of 7 (exceeds 5 allowed)'.  

_None_ of those are a high-enough priority improvement that anybody should be asked to spend any time considering them.

Based on this, and knowing nothing of the granularity of CC's controls, I recommend: 

 - we raise the common code limit to at least three cases _and_ at least five lines

 - we raise the Cognitive Complexity flagging limit to 10

with the understanding that those increases are just a step to see what happens, and higher still might be better.

N.B:  "Cognitive Complexity" for source code seems to have [originated in work at SonarSource](https://www.sonarsource.com/docs/CognitiveComplexity.pdf) with no underlying work. (The [McCabe [1976] paper](http://juacompe.mrchoke.com/natty/thesis/FrameworkComparison/A%20complexity%20measure.pdf) that underlies the Cyclometric (not Cognitive) Complexity metric is the only reference cited.)  I've never seen the specific Cognitive Complexity metric _except_ in SonarSource and CodeClimate; I'd very much appreciate hearing about some independent work that demonstrates it has predictive validity.

pabender, 2019-09-27:
Sent from my iPhone

> On Sep 27, 2019, at 10:24 AM, Bob Jacobsen <notifications@github.com> wrote:
> Based on this, and knowing nothing of the granularity of CC's controls, I recommend:
> we raise the common code limit to at least three cases and at least five lines
> 
> we raise the Cognitive Complexity flagging limit to 10
> 
> with the understanding that those increases are just a step to see what happens, and higher still might be better.
> 

For what it’s worth, at work the cognitive complexity flag is set to 15.  5 is definitely too low.

silverailscolo, 2019-09-27:
Agreed that the current CC values are rather limiting. Suggested new threshold settings look good to me (after spending a couple of evenings refactoring CC flagged code). 

Thanks for the reference on Code Complexity, since understanding what is being measured is the first step in managing it.
CC seems to analyze a whole class, even if I only edit 1 line in 1 of the methods. That’s great to draw attention to possible improvements, but at the same time it distracts from the initial PR subject. And because of that, it might lead to all sorts of unrelated code changes just to get the scores down and make the red x turn green. My suggestion would be to weight new code flags higher than decade old stuff in decide to merge.

Egbert

rhwood, 2019-09-29:
While I agree that thresholds should be raised, I do want to add a note as to why "similar" matters:

While working on #7467, I found four similar blocks of code--three of which were not working correctly. By replacing the four blocks with a single method, I was able to:
1. reduce the amount of code in JMRI
2. create one method that can be tested
3. ensure identical behavior across four points in the JMRI UI

With any luck Code Climate will flag additional places where the newly created method can be used.

I guess this is just a demonstration the value of the similar code checks.

bobjacobsen, 2019-09-29:
> While I agree that thresholds should be raised, I do want to add a note as to why "similar" matters:
> 
> While working on #7467, I found four similar blocks of code--three of which were not working correctly. By replacing the four blocks with a single method, I was able to:
> 
> reduce the amount of code in JMRI
> create one method that can be tested
> ensure identical behavior across four points in the JMRI UI
> With any luck Code Climate will flag additional places where the newly created method can be used.
> 
> I guess this is just a demonstration the value of the similar code checks.

I'm curious whether the genesis of that was CodeClimate's detection of _identical_ code.  Seems to me the whole point was that at least one was non-identical.

rhwood, 2019-09-29:
None of these blocks had identical code — the variable names and classes in the identical sections of the blocks were all different.

Although that *could* mean that the duplication checks are aware of the need to parse without consideration for variable names.