JMRI/JMRI #7465: LayoutTrackEditorsTest.testEditTurnoutDone:250 » Jemmy Exception in doClick

danielb987, 2019-09-28:
After merge master into my branches, I get the following error on HEADLESS=false STATIC=false:

`LayoutTrackEditorsTest.testEditTurnoutDone:250 » Jemmy Exception in doClick`

I get this error even on my master branch. It seems that something has been broken in JMRI master.

I also gets a number of warnings:
```
WARN  - invalid character in number field of system name: ZSabcd:b [main] jmrix.roco.z21.Z21CanBusAddress.getBitFromSystemName()
WARN  - system name ZSabcd1 is in the wrong format.  Should be mm:pp. [main] jmrix.roco.z21.Z21CanBusAddress.getBitFromSystemName()
WARN  - invalid character in number field of system name: ZS0b:a [main] jmrix.roco.z21.Z21RMBusAddress.getBitFromSystemName()
WARN  - Z21 RM Bus hardware address out of range in system name ZS999 [main] jmrix.roco.z21.Z21RMBusAddress.getBitFromSystemName()
```

rhwood, 2019-09-28:
Have you done a maven or ant clean before attempting to build?

While it is not common for a change in master to cause unclean builds to break after a git pull, neither should we expect that builds not need to be cleaned after a git pull.

danielb987, 2019-09-28:
I get this on Travis.

danielb987, 2019-09-28:
My [master branch](https://github.com/danielb987/JMRI/) on Travis HEADLESS=false STATIC=false:
https://travis-ci.com/danielb987/JMRI/jobs/240014655

dsand47, 2019-09-29:
The Jemmy exception does not occur when running the LayoutEditor PackageTest or the specific test locally using ant clean, ant tests and the runtest.csh script.

All CI runs since the merge of #7425 have failed in Travis step 1 with the same error.

pabender, 2019-09-29:
I am seeing this locally also.

The fact that the error is occurring in doClick means that there is an exception thrown durring the processing of the button click event.  that at least implies the error is triggered somewhere in this code:

https://github.com/JMRI/JMRI/blob/5f8201b3d60228e24bd459cabddf22983f1416d9/java/src/jmri/jmrit/display/layoutEditor/LayoutTrackEditors.java#L1710-L1735

The particular class in question really should be refactored into a base class and several smaller classes.  I see markers in the code stating that segments of it are for particular types of turnout objects, including unique variables for each type.

dsand47, 2019-09-29:
The Jemmy error is in one of the **turnout** tests, not a **turntable** test.

What is not clear is if the error applies to all of the Done button clicks or just this one.  Since I don't know how to setup Travis locally, I cannot test commenting out this test to see if the error moves to another test.  Travis did not have any problems with the tests with my PR commits, only after the merge.

The LayoutTrackEditors class handles the edit dialogs for track segments, turnouts, slips and level crossings.  The turntable support appears to be a semi-related addition.

pabender, 2019-09-29:
> The Jemmy error is in one of the **turnout** tests, not a **turntable** test.

Sorry, I fixed my earlier link.

> What is not clear is if the error applies to all of the Done button clicks or just this one. Since I don't know how to setup Travis locally, I cannot test commenting out this test to see if the [...]

My Linux machine is doing this at home too.

Actually, I am doing some work on this as part of an effort to make sure windows are closed in tests ( so I can turn on the error messages ) and I was able to get this to move to another test ( testEditXingDone ) by fixing an issue with the test setup.

The tests ( as they are in master ) have one layout editor created in an ‘@BeforeClass’ method, but the instance manager is reset before each test AND the standard setup/tearDown occurs before/after each tests.  This means the contents of the managers are reset, but the LayoutEditor object is retaining references to items that should have been deleted.

For the Test I have failing now, I believe the tests were written to expect this retained information, instead of starting from a clean slate each time.

> The LayoutTrackEditors class handles the edit dialogs for track segments, turnouts, slips and level crossings. The turntable support appears to be a semi-related addition.

Right, but each one of those dialogs are different, yet they are all clumped together in one class.  That isn’t right.

dsand47, 2019-09-29:
That all makes sense.

I think the short term fix is to revert LayoutTrackEditorsTest so that Travis stops failing for everyone.   I would like to keep the changes in LayoutTrackEditors.   A number of people are upset over losing the ability to create new blocks from the track editor dialogs so ideally that change should be included in 4.17.5.  

As a separate activity, I can look at splitting LayoutTrackEditors into separate classes.  Is it better to keep them all in the jmri.jmrit.display.layoutEditor package or create something like a jmri.jmrit.display.layoutEditor.trackEditors package?

pabender, 2019-09-29:
@dsand47 I've just put up a pull request that (hopefully) shows where I am seeing an error now.

I actually do see an error in two tests testEditXingDone and testEditTurnoutDone.  See #7486 for the stack traces I am seeing locally.

pabender, 2019-09-29:
Strangely, the build on #7466 failed ( with this error and one other )

https://travis-ci.org/JMRI/JMRI/builds/591006324?utm_source=github_status&utm_medium=notification

But the builds of the same code on my branch succeeded:

https://travis-ci.org/pabender/JMRI/builds/591005010?utm_source=github_status&utm_medium=notification

Which means either the cause on Travis is different than what I see locally, or we somehow have an illegal argument exception that is intermittent. If I have time today, I will try to trace the illegal argument exception I see on the local build.

pabender, 2019-09-29:
Explicit initialization of the block manager appears to have helped with testEditTurnoutDone.

I am still getting an error locally with The crossing one, but #7466 passed its Travis tests

dsand47, 2019-09-29:
The stack trace for **Xing Blk BD** shows that editLevelXingBlockACPressed was called instead of the BD version.  Or the wrong text was in the field when the AC version was called.

pabender, 2019-09-29:
@dsand47  I don't know how that would happen, but the only changes I made to the tests in question are in #7466 (which has now been merged) and they impact only the common setup and tearDown of all the tests in the file.

in any event #7466 MAY take care of this particular issue on the CI servers

pabender, 2019-09-29:
@dsand47 one other thought about the test I still have failing... It shouldn't actually matter which callback is triggering the issue.  

The problem is that the BlockManager is saying the argument is illegal, i.e. the Block manager is saying it can't create a block for the name.  Reading the logic, if the block exists, it is supposed to return the existing one. If the block doesn't exist, and the name doesn't have the right prefix (which this one does not) it is to be treated as a user name for an Automatically named block.  The illegal argument exception is thrown after the block isn't found or created.

dsand47, 2019-09-29:
I suspect there might be an issue with register().  When I started working on this class test I had problems with the auto number not incrementing.  This occurs after register().  Maybe something is throwing an uncaught exception.

dsand47, 2019-09-30:
@pabender 
The turnout test failure has returned.  I have a copy of 4.16 source so I could revert LayoutTrackEditorsTest so that the other people are not affected.

After 4.17.5 is in I can figure out what is wrong with provideBlock and re-implement the tests.  

The current class structure was created at 4.9.5.  Previously, the edit methods were included in each of the track component classes.

pabender, 2019-09-30:
@dsand47,

The bottom line problem is NOT in the LayoutTrackEditorsTest or even the LayoutTrackEditors.  The bottom line cause of the problem is that BlockManager is throwing an illegal argument exception when provided with a block user name that should produce a valid automatic block name.

"Fixing" the tests in LayoutTrackEditorsTest by reverting them to a previous version only hides the problem.  Now that we have identified it, we need to fix it.

pabender, 2019-09-30:
@dsand47  also, which pull request, and was it updated after I made my changes?

pabender, 2019-09-30:
One general comment about the tests in LayoutTrackEditorsTest.

I see a lot of references such as:

https://github.com/JMRI/JMRI/blob/4df4423fb52461cd6cd5dce4a53c61dffdf89ad6/java/test/jmri/jmrit/display/layoutEditor/LayoutTrackEditorsTest.java#L84

That line works, but it is fragile.  The problem is if someone adds or removes an object of the type referenced, then the line above fails to get the right component.  I am not even positive that Jemmy will guarantee the same component will always be returned with the same index value.

It is better to reference objects whever possible by the labels associated with them:
https://github.com/JMRI/JMRI/blob/4df4423fb52461cd6cd5dce4a53c61dffdf89ad6/java/test/jmri/jmrit/speedometer/SpeedometerScaffold.java#L42

This requires that the object is actually labeled correctly however.  The code that makes the line from the SpeedometerScaffold work looks like this:
https://github.com/JMRI/JMRI/blob/4df4423fb52461cd6cd5dce4a53c61dffdf89ad6/java/src/jmri/jmrit/speedometer/SpeedometerFrame.java#L48
https://github.com/JMRI/JMRI/blob/4df4423fb52461cd6cd5dce4a53c61dffdf89ad6/java/src/jmri/jmrit/speedometer/SpeedometerFrame.java#L144-L147

Which only breaks if someone removes the label.

Personally, I only use the index value if it is the only option.

dsand47, 2019-09-30:
The one I looked at is #7464.  The latest failure was at 2019-09-29T23:03:23Z.

pabender, 2019-09-30:
Oh darn that is subtle.

The manager creates the bean, and fires a property change during register, which causes the combo box to update its list, but it finds a bean that isn’t registered yet and asks for it, but the manager already knows about the bean, but hasn’t fully created it, and so it tries ( and fails ), throwing the illegal argument exception.

I haven’t looked at more than the stack trace, but it could be as simple as a misplaced call to fire the property change

danielb987, 2019-09-30:
I hope PR #7468 will fix this.

danielb987, 2019-09-30:
> I haven’t looked at more than the stack trace, but it could be as simple as a misplaced call to fire the property change

Maybe it's better to fix that, than my PR. Could this problem be with other managers too?

dsand47, 2019-09-30:
I checked Logix, Memory, Route, and Section.  They all increment the auto number after register so they are also susceptible to the same problem as Block. There may be others.

I suspect that the other managers are less likely to have this issue.  The turnout test creates 4 blocks for a double cross-over and the level crossing test creates two blocks, both with no pauses.  The errors always occur on a subsequent block, not the first block.  

What is strange is that the error only appears to occur when running the test within Travis.  It does not occur when using the GUI (which has a natural delay when the user changes fields) or when using the local test script.

bobjacobsen, 2019-09-30:
> What is strange is that the error only appears to occur when running the test within Travis. It does not occur when using the GUI (which has a natural delay when the user changes fields) or when using the local test script.

Might be a threading issue in the test.  

Swing actions are atomic:  One event will run to a conclusion before any additional ones are invoked.  That guarantee is lost if the contents of Swing objects (i.e. like the content of a field) are manipulated from _off_ the GUI (Swing/AWT) thread.

danielb987, 2019-09-30:
Full stack trace.

```
java.lang.IllegalArgumentException: Could not create block "DX New B"
	at jmri.BlockManager.provideBlock(BlockManager.java:190)
	at jmri.BlockManager.provide(BlockManager.java:374)
	at jmri.BlockManager.provide(BlockManager.java:37)
	at jmri.swing.NamedBeanComboBox.getSelectedItem(NamedBeanComboBox.java:214)
	at jmri.swing.NamedBeanComboBox.getSelectedItem(NamedBeanComboBox.java:78)
	at javax.swing.plaf.basic.BasicComboBoxUI$Handler.propertyChange(BasicComboBoxUI.java:1702)
	at javax.swing.plaf.basic.BasicComboBoxUI$PropertyChangeHandler.propertyChange(BasicComboBoxUI.java:635)
	at javax.swing.plaf.metal.MetalComboBoxUI$MetalPropertyChangeListener.propertyChange(MetalComboBoxUI.java:207)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at java.awt.Component.firePropertyChange(Component.java:8434)
	at javax.swing.JComboBox.setModel(JComboBox.java:318)
	at jmri.swing.NamedBeanComboBox.sort(NamedBeanComboBox.java:327)
	at jmri.swing.NamedBeanComboBox.lambda$new$0(NamedBeanComboBox.java:86)
	at jmri.util.ThreadingPropertyChangeListener.lambda$propertyChange$0(ThreadingPropertyChangeListener.java:27)
	at jmri.util.ThreadingUtil.runOnGUI(ThreadingUtil.java:111)
	at jmri.util.ThreadingPropertyChangeListener.propertyChange(ThreadingPropertyChangeListener.java:26)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:328)
	at java.beans.PropertyChangeSupport.fireIndexedPropertyChange(PropertyChangeSupport.java:358)
	at jmri.managers.AbstractManager.fireIndexedPropertyChange(AbstractManager.java:488)
	at jmri.managers.AbstractManager.register(AbstractManager.java:254)
	at jmri.BlockManager.createNewBlock(BlockManager.java:106)
	at jmri.BlockManager.createNewBlock(BlockManager.java:152)
	at jmri.BlockManager.provideBlock(BlockManager.java:187)
	at jmri.BlockManager.provide(BlockManager.java:374)
	at jmri.BlockManager.provide(BlockManager.java:37)
	at jmri.swing.NamedBeanComboBox.getSelectedItem(NamedBeanComboBox.java:214)
	at jmri.swing.NamedBeanComboBox.getSelectedItem(NamedBeanComboBox.java:78)
	at javax.swing.plaf.basic.BasicComboBoxUI$Handler.contentsChanged(BasicComboBoxUI.java:1865)
	at javax.swing.AbstractListModel.fireContentsChanged(AbstractListModel.java:118)
	at javax.swing.DefaultComboBoxModel.setSelectedItem(DefaultComboBoxModel.java:93)
	at javax.swing.JComboBox.actionPerformed(JComboBox.java:1314)
	at javax.swing.plaf.basic.BasicComboBoxUI$Handler.focusLost(BasicComboBoxUI.java:1838)
	at java.awt.AWTEventMulticaster.focusLost(AWTEventMulticaster.java:230)
	at java.awt.AWTEventMulticaster.focusLost(AWTEventMulticaster.java:229)
	at java.awt.Component.processFocusEvent(Component.java:6431)
	at java.awt.Component.processEvent(Component.java:6295)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
	at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:1023)
	at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:689)
	at java.awt.Component.dispatchEventImpl(Component.java:4760)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at org.netbeans.jemmy.QueueTool$JemmyQueue.dispatchEvent(QueueTool.java:656)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

danielb987, 2019-10-03:
I still have this problem.
`LayoutTrackEditorsTest.testEditTurnoutDone:245 » Jemmy Exception in doClick`

I have merged Master into my branch [LogixNG_new_3](https://github.com/danielb987/JMRI/tree/LogixNG_new_3) and [Travis](https://travis-ci.com/danielb987/JMRI/jobs/241831197) still has the error despite PR  #7469 is merged into this branch.

I think this Issue should be reopened but I cannot do that.

rhwood, 2019-10-03:
If you can get a stack trace is it showing the same recursion?

dsand47, 2019-10-04:
> If you can get a stack trace is it showing the same recursion?

@rhwood 
See if this shows what you are looking for.

I wrapped the block manager register call in a try/catch.  A Travis run against my repository branch finishes successfully with several caught exceptions in the tests.  The problem appears that under certain conditions, the same system name can be assigned to different user names.

```
WARN  - register exception: sys = IB:AUTO:0005, user = DX New C [AWT-EventQueue-1] jmri.BlockManager.createNewBlock()
722java.lang.IllegalArgumentException: Could not create block "DX New B"
723	at jmri.BlockManager.provideBlock(BlockManager.java:168)
724	at jmri.BlockManager.provide(BlockManager.java:342)
725	at jmri.BlockManager.provide(BlockManager.java:37)
726	at jmri.swing.NamedBeanComboBox.getSelectedItem(NamedBeanComboBox.java:224)
727	at jmri.swing.NamedBeanComboBox.getSelectedItem(NamedBeanComboBox.java:79)
728	at javax.swing.plaf.basic.BasicComboBoxUI$Handler.propertyChange(BasicComboBoxUI.java:1702)
729	at javax.swing.plaf.basic.BasicComboBoxUI$PropertyChangeHandler.propertyChange(BasicComboBoxUI.java:635)
730	at javax.swing.plaf.metal.MetalComboBoxUI$MetalPropertyChangeListener.propertyChange(MetalComboBoxUI.java:207)
731	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
732	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
733	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
734	at java.awt.Component.firePropertyChange(Component.java:8434)
735	at javax.swing.JComboBox.setModel(JComboBox.java:318)
736	at jmri.swing.NamedBeanComboBox.sort(NamedBeanComboBox.java:351)
737	at jmri.swing.NamedBeanComboBox.lambda$new$0(NamedBeanComboBox.java:88)
738	at jmri.util.ThreadingPropertyChangeListener.lambda$propertyChange$0(ThreadingPropertyChangeListener.java:27)
739	at jmri.util.ThreadingUtil.runOnGUI(ThreadingUtil.java:111)
740	at jmri.util.ThreadingPropertyChangeListener.propertyChange(ThreadingPropertyChangeListener.java:26)
741	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
742	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:328)
743	at java.beans.PropertyChangeSupport.fireIndexedPropertyChange(PropertyChangeSupport.java:358)
744	at jmri.managers.AbstractManager.fireIndexedPropertyChange(AbstractManager.java:488)
745	at jmri.managers.AbstractManager.register(AbstractManager.java:254)
746	at jmri.BlockManager.createNewBlock(BlockManager.java:101)
747	at jmri.BlockManager.createNewBlock(BlockManager.java:140)
748	at jmri.BlockManager.provideBlock(BlockManager.java:165)
749	at jmri.BlockManager.provide(BlockManager.java:342)
750	at jmri.BlockManager.provide(BlockManager.java:37)
751	at jmri.swing.NamedBeanComboBox.getSelectedItem(NamedBeanComboBox.java:224)
752	at jmri.swing.NamedBeanComboBox.getSelectedItem(NamedBeanComboBox.java:79)
753	at javax.swing.plaf.basic.BasicComboBoxUI$Handler.contentsChanged(BasicComboBoxUI.java:1865)
754	at javax.swing.AbstractListModel.fireContentsChanged(AbstractListModel.java:118)
755	at javax.swing.DefaultComboBoxModel.setSelectedItem(DefaultComboBoxModel.java:93)
756	at javax.swing.JComboBox.actionPerformed(JComboBox.java:1314)
757	at javax.swing.plaf.basic.BasicComboBoxUI$Handler.focusLost(BasicComboBoxUI.java:1838)
758	at java.awt.AWTEventMulticaster.focusLost(AWTEventMulticaster.java:230)
759	at java.awt.AWTEventMulticaster.focusLost(AWTEventMulticaster.java:229)
760	at java.awt.Component.processFocusEvent(Component.java:6431)
761	at java.awt.Component.processEvent(Component.java:6295)
762	at java.awt.Container.processEvent(Container.java:2239)
763	at java.awt.Component.dispatchEventImpl(Component.java:4889)
764	at java.awt.Container.dispatchEventImpl(Container.java:2297)
765	at java.awt.Component.dispatchEvent(Component.java:4711)
766	at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
767	at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:1023)
768	at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:689)
769	at java.awt.Component.dispatchEventImpl(Component.java:4760)
770	at java.awt.Container.dispatchEventImpl(Container.java:2297)
771	at java.awt.Component.dispatchEvent(Component.java:4711)
772	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
773	at java.awt.EventQueue.access$500(EventQueue.java:97)
774	at java.awt.EventQueue$3.run(EventQueue.java:709)
775	at java.awt.EventQueue$3.run(EventQueue.java:703)
776	at java.security.AccessController.doPrivileged(Native Method)
777	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
778	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
779	at java.awt.EventQueue$4.run(EventQueue.java:733)
780	at java.awt.EventQueue$4.run(EventQueue.java:731)
781	at java.security.AccessController.doPrivileged(Native Method)
782	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
783	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
784	at org.netbeans.jemmy.QueueTool$JemmyQueue.dispatchEvent(QueueTool.java:656)
785	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
786	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
787	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
788	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
789	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
790	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
791
```

rhwood, 2019-10-04:
So I thought the call the call to `getSelectedItem()` in `sort()` was the problem, so I removed that call, but see the recursion later in `sort()` when the model gets swapped out.

I think I need to distill the failing test into its own test case for NamedBeanComboBox to get to the bottom of this (two things are needed: a way to ensure getSelectedItem only calls super getSelectedItem when the model changes, and a way to ensure an external change while a user is editing does not break the editing).

Randall

pabender, 2019-10-06:
@rhwood do you have any other thoughts on this?  

@dsand47 opened PR #7473, which he has marked as WIP, which catches an exception during the register... but I don’t know if that is the right solution, but it passed tests.

@danielb987 created PR #7468 which just moves where the register occurs.  It’s Travis test failed, but not due to an Instance of this problem.

rhwood, 2019-10-06:
I’m rebuilding my principal computer today (it’s been mostly unusable for a week over some disk formatting error from years ago that caused a cascading failure that eventually became untenable), so I can’t go into details or provide a rapid fix.

There remains an issue with NamedBeanComboBox that may only be dumb luck we didn’t see it before where a change in the underlying model can cause this exception that is not addressed by either PR.

That said, if #7468 prevents the error in testing, *and* works in real life, it’s acceptable.

I don’t like #7473 because it catches *all* exceptions, if it caught just the one class of exceptions it might be acceptable.

I’m worried either one might lead to a stack overflow through excessive recursion however (but if the timing/threading of real world use is such that doesn’t happen, it’s a test only problem).

pabender, 2019-10-06:
@rhwood thanks.

I also spotted a race condition in BlockManager that needs to be addressed.  See my note on #7468. It isn’t causing the recursion, but it needs to be addressed.

danielb987, 2019-10-08:
#7468 is merged. Is this issue now fixed and can be closed?

PR #7469 (2019-10-03): fix: Prevent recursion when creating a new bean
Prevents a NamedBeanComboBox from recursively requesting a new NamedBean when a NamedBean is created based on user input into the NamedBeanComboBox.

Fixes #7465
closed 2019-11-01 by danielb987