JMRI/JMRI #7470: Inputline validation icons blurred

svatoun, 2019-10-01:
I am aiming to extend input validation to more JMRI inputs + potentially use in my application, however the visual appearance of the indicator is defective.

Environment: Ubuntu 19.04; KDE5. Display: laptop 1920x1080, Acer 1600x1200.
System default font size: 11px
JMRI default font size: 12px, later changed to 14px
Antialiasing: on; tried off, with the same results.

The validation icons (tried: warning, error, ok) are blurred, triangle lines are not sharp (when seen in the dialog in its actual size). The shape can be guessed, but is definitely not well painted. The same for error / OK. 

Screenshots are at: 
- https://www.dropbox.com/s/n51wtfo3g2azqps/error_nimbus.png?dl=0
- https://www.dropbox.com/s/qryg5k2ubjr8dr1/ok_nimbus.png?dl=0
- https://www.dropbox.com/s/sx83vijry1lvmpr/warning_metal.png?dl=0

I could guess it's because font resizing because the icons are drawn as font shapes; anyway the dialog font and the input line font appears cleanly (see the screenshots). The same results in GTK+, Nimbus, Metal. Didn't try Motif L&F.
While drawing graphics as font can potentially work better with large font sizes, or maybe hires displays ,raster icons offer at least defined appearance on a display.

In addition, I suggest to more differentiate OK (circle + checkmark) and ERROR (circle with something blurred inside) - please use red cross for ERROR. The shape is better differentiated from a triangle (WARNING) or circle (or a checkmark alone). Circles can be confused, especially when the shape inside foggy and small.

I have copied a standard PNG icon into the screenshot for comparison:
https://www.dropbox.com/s/027pc6a7dd3twy1/icon_warning_nimbus.png?dl=0

Because of the non-standard icon resource load, I am unable to rebrand it to clear-looking icons in my application; the blurry appearance may pose an issue for visually-impaired people.

rhwood, 2019-10-01:
Are those icons clear when shown at https://www.patternfly.org/v4/design-guidelines/usage-and-behavior/alerts-and-notifications ?

How did you set/unset antialiasing?

Randall

rhwood, 2019-10-02:
Actually, the page in the prior message has PNGs of the icons, so that’s not a good example (it is a good example of PNGs not scaling well, however). Are the icons clear on https://www.patternfly.org/v4/design-guidelines/styles/icons <https://www.patternfly.org/v4/design-guidelines/styles/icons> ?

rhwood, 2019-10-02:
And what version of Java are you using?

svatoun, 2019-10-02:
> And what version of Java are you using?
`DecoderPro version 4.17.5ish+sdedic+20191002T0629Z+R764bd32c8a starts under Java 1.8.0_161 `
also tried with `4.17.4` release, the same JDK.

`How did you set/unset antialiasing?`
Good catch; settings in KDE control panel. GIMP UI was affected by the setting (native GTK+) and Kate as well (native KDE), but perhaps Java font rendering was not. Restarted the JMRI process, of course.

> Actually, the page in the prior message has PNGs of the icons
Sorry, I don't understand that message; the first 3 links are screenshots from JMRI dialog, display them in 1:1 scale to see the actual size of the screenshot. Dropbox magnifies by default.

Additional info:
I am now in different environment - same laptop, no X restart, just different monitors (DELL 2560x1600, DELL 1920x1080); OK and ERROR look little sharper, warning is still 'shaky'.

svatoun, 2019-10-02:
> Are the icons clear on https://www.patternfly.org/v4/design-guidelines/styles/icons

The icons are clear, in their 50px default rendered size. We need clear apperance in approx 16px size.

svatoun, 2019-10-02:
Noticed that my browser uses 16pt font; changed to 12pt (the setting from JMRI). Icons on the website are also clear in their 40px apperance on screen.

rhwood, 2019-10-02:
Am waiting on maven central to process a fix for the jagged edges (when rendering the font into a bitmap for the border, antialiasing was off); will have a PR that addresses that tomorrow.

The problem with using PNGs is that they do not scale (PNGs are bitmaps, not vectors), which is why, when using a PNG for an icon in Ubuntu, 16x16, 22x22, 24x24, 32x32, 36x36, 48x48, 64x64, 72x72, etc… pixel size PNGs are all used, and Ubuntu has to pick the correct size (within those sizes there are different DPI icons as well, so that a HIDPI icon (name-size@2.png or name-size@3.png) has a different pixel density than the “normal” version.

It is because of the difficulty of matching PNG size to font size that I choose to use a font directly in the validators.

svatoun, 2019-10-02:
> The problem with using PNGs is that they do not scale (PNGs are bitmaps, not vectors), which is why, when using a PNG for an icon in Ubuntu, 16x16, 22x22, 24x24, 32x32, 36x36, 48x48, 64x64, 72x72, etc… pixel size PNGs are all used, and Ubuntu has to pick the correct size (within those sizes 

I beg your pardon; I am developing A11Yed Swing apps let's almost two decades - although definitely not UX engineer. The requirement to have so totally scalable graphics is rather unusual (comparing to  my professional experience). From that position, this seems rather an over-compensation, but the code is already in place and will hopefully get visually better. Your call.

closed 2020-04-15 by svatoun