JMRI/JMRI #7479: Yet another Maven configuration problem?

bobjacobsen, 2019-10-05:
Travis and AppVeyor supposed to be using Maven to run, among other things, the equivalent of `ant warnings-check` and `ant test-warnings-check`.  Those are the more stringent versions of the more accepting (i.e. warns on some items without failing) `ant warnings-check` and `ant test-warnings-check` that we run in Jenkins.  That [Jenkins job](http://builds.jmri.org/jenkins/job/Development/job/Builds/) has been showing warnings have been getting past the Travis and Appveyor CI checks without causing those builds to fail:

[image]

```
    [javac] Compiling 4405 source files to /Users/jake/Documents/Trains/JMRI/projects/JMRI/target/classes
    [javac] ----------
    [javac] 1. WARNING in /Users/jake/Documents/Trains/JMRI/projects/JMRI/java/src/jmri/jmrit/display/layoutEditor/LayoutTrackEditors.java (at line 891)
    [javac] 	private JButton editLayoutSlipDoneButton;
    [javac] 	                ^^^^^^^^^^^^^^^^^^^^^^^^
    [javac] The value of the field LayoutTrackEditors.editLayoutSlipDoneButton is not used
    [javac] ----------
    [javac] 2. WARNING in /Users/jake/Documents/Trains/JMRI/projects/JMRI/java/src/jmri/jmrit/display/layoutEditor/LayoutTrackEditors.java (at line 892)
    [javac] 	private JButton editLayoutSlipCancelButton;
    [javac] 	                ^^^^^^^^^^^^^^^^^^^^^^^^^^
    [javac] The value of the field LayoutTrackEditors.editLayoutSlipCancelButton is not used
    [javac] ----------
    [javac] 3. WARNING in /Users/jake/Documents/Trains/JMRI/projects/JMRI/java/src/jmri/jmrit/display/layoutEditor/LayoutTrackEditors.java (at line 1315)
    [javac] 	private JButton editLevelXingDoneButton;
    [javac] 	                ^^^^^^^^^^^^^^^^^^^^^^^
    [javac] The value of the field LayoutTrackEditors.editLevelXingDoneButton is not used
    [javac] ----------
    [javac] 4. WARNING in /Users/jake/Documents/Trains/JMRI/projects/JMRI/java/src/jmri/jmrit/display/layoutEditor/LayoutTrackEditors.java (at line 1316)
    [javac] 	private JButton editLevelXingCancelButton;
    [javac] 	                ^^^^^^^^^^^^^^^^^^^^^^^^^
    [javac] The value of the field LayoutTrackEditors.editLevelXingCancelButton is not used
    [javac] ----------
    [javac] 5. WARNING in /Users/jake/Documents/Trains/JMRI/projects/JMRI/java/src/jmri/jmrit/display/layoutEditor/LayoutTrackEditors.java (at line 1526)
    [javac] 	private JButton editLayoutTurntableDoneButton;
    [javac] 	                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    [javac] The value of the field LayoutTrackEditors.editLayoutTurntableDoneButton is not used
    [javac] ----------
    [javac] 6. WARNING in /Users/jake/Documents/Trains/JMRI/projects/JMRI/java/src/jmri/jmrit/display/layoutEditor/LayoutTrackEditors.java (at line 1527)
    [javac] 	private JButton editLayoutTurntableCancelButton;
    [javac] 	                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    [javac] The value of the field LayoutTrackEditors.editLayoutTurntableCancelButton is not used
    [javac] ----------
    [javac] ----------
    [javac] 7. WARNING in /Users/jake/Documents/Trains/JMRI/projects/JMRI/java/src/jmri/jmrix/dccpp/DCCppConstants.java (at line 25)
    [javac] 	};
    [javac] 	 ^
    [javac] Unnecessary semicolon
    [javac] ----------
    [javac] 7 problems (7 warnings)
```

After 20 minutes of work, I haven't been able to figure out how the "clearer and simpler" Maven configuration decides what to run and what not to run.  Hence this Issue in hopes somebody else will (and also document that Maven setup somewhere, specifically what's done in what order)

rhwood, 2019-10-05:
I’ll take a look at the Travis-CI configuration. 

On AppVeyor, maven does three things:
- ensure ant is installed (it’s historically been more reliable to use maven to bootstrap ant than to use a power shell script to download and install ant every time in AppVeyor)
- run `ant appveyor` (If using maven to bootstrap, the ant CLI is not available)
- run the test `jmri.util.FileLineEndingsTest` outside ant (that test appeared to have less errors running directly from maven, but since the test itself has been rewritten since it was separated, so it merely be historical)
The AppVeyor maven config does not appear to have changed in 3 years, so any recent changes in how AppVeyor runs tests a result of changes to `build.xml`

pabender, 2019-10-05:
@rhwood, I am taking a look at this.  There were some errors in the configuration I have partly fixed.

When I swapped the warnings over to using maven, there were no errors produced, so I didn’t catch it wasn’t working correctly

pabender, 2019-10-06:
#7481 fixes the ecj checks on the travis static job.  At least for now, I restored using the ant target for this.  

I fixed several issues in the maven configuration, but it still isn't triggering any warnings.

I also went ahead and fixed the errors that had crept in.

pabender, 2019-10-06:
incidentally, at least over the last 3 years, I didn't find any evidence that Appveyor ever ran any of the ant warnings check targets, and travis has only been running the tests-warnings-check ant target.

bobjacobsen, 2019-10-06:
At some point, `ant tests-warnings-check` started including `ant warnings-check`, so the 2nd part of that is OK.

pabender, 2019-10-06:
> At some point, `ant tests-warnings-check` started including `ant warnings-check`, so the 2nd part of that is OK.

Ok, then #7481 restores the behavior we had prior to configuring Travis to use maven for the ecj checks.  I can’t merge it because it is failing on the doClick error on Travis headed.

There has not been any change in what appveyor does with respect to the ecj checks in quite a while.

PR #7481 (2019-10-07): restore ecj error check on travis
This fixes #7479
closed 2019-10-07 by bobjacobsen