JMRI/JMRI #7493: jmri.server.json.idtag.JsonIdTagHttpServiceTest leaves state that causes it to fail

bobjacobsen, 2019-10-08:
When running `jmri.server.json.idtag.JsonIdTagHttpServiceTest`, some files are left behind:
```
% find . -newer time.stamp 
./temp/idtags
./temp/idtags/JMRIIdTags.xml
./temp/idtags/JMRIIdTags.xml.bak
```
Is it OK to explicitly remove them?  Is there some call that should be used to reset them in tests?

Leaving these in place causes the test to fail the next time:
```
There were 2 failures:
1) testDoPut(jmri.server.json.idtag.JsonIdTagHttpServiceTest)
java.lang.AssertionError: expected null, but was:<ID1>
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.failNotNull(Assert.java:755)
	at org.junit.Assert.assertNull(Assert.java:737)
	at org.junit.Assert.assertNull(Assert.java:747)
	at jmri.server.json.idtag.JsonIdTagHttpServiceTest.testDoPut(JsonIdTagHttpServiceTest.java:138)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)

2) testDoGetList(jmri.server.json.idtag.JsonIdTagHttpServiceTest)
java.lang.AssertionError: expected:<1> but was:<2>
	at org.junit.Assert.fail(Assert.java:88)
	at org.junit.Assert.failNotEquals(Assert.java:834)
	at org.junit.Assert.assertEquals(Assert.java:645)
	at org.junit.Assert.assertEquals(Assert.java:631)
	at jmri.server.json.idtag.JsonIdTagHttpServiceTest.testDoGetList(JsonIdTagHttpServiceTest.java:206)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
```

rhwood, 2019-10-08:
I hadn’t noticed. I’ll take a look. I don’t think the test cares if those are removed (I think the test only cares that the IdTag manager has those during the test, but is indifferent to the manager writing those files).

bobjacobsen, 2019-10-11:
I think it's a bit more complicated than I described, unfortunately.  It seems that some _other_ files also need to be present to trigger the fault.  Perhaps in a test profile?  Trying to get a sequence of operations that reliably recreates this.

bobjacobsen, 2019-10-30:
@rhwood If I add an `rm -rf temp` to `scripts/run_tests_separately`, it runs cleanly.  I'm not sure that's a good idea, though, as it's meant to check that the tests are independent even when running together as a package, without separate cleanup.

bobjacobsen, 2019-11-03:
This seems to be a minimal example:

```
ant realclean tests
./runtest.csh java/test/jmri/server/json/operations/JsonOperationsHttpServiceTest.java
./runtest.csh jmri.server.json.idtag.JsonIdTagHttpServiceTest
```

Without the 2nd line, the 3rd line passes; with the 2nd line, it fails. `rm -rf temp/` restores the 3rd line to normal operation.

The question is whether the problem is `JsonIdTagHttpServiceTest` not properly ignoring external state, i.e. something in `temp/`, or whether the problem is that `JsonOperationsHttpServiceTest.java` (and perhaps others) are not cleaning up after themselves.

rhwood, 2019-11-03:
> The question is whether the problem is JsonIdTagHttpServiceTest not properly ignoring external state, i.e. something in temp/, or whether the problem is that JsonOperationsHttpServiceTest.java (and perhaps others) are not cleaning up after themselves.

Probably both. The Operations tests add tags to roster entries, and then the IdTag tests are looking for a specific state. In either case, its a side effect of using a real IdTagManager instead of a mock one, since neither test is intending to test that an IdTagManager works, but both need one. I'll see if I can gin up a fix for both tomorrow.

bobjacobsen, 2019-11-03:
Removing the `temp/idtags/JMRIIdTags.xml` file is enough for `JsonIdTagHttpServiceTest` to succeed.  I think that means that the `JUnitUtil.initIdTagManager()` call in `JsonIdTagHttpServiceTest`s `setUp` routine isn't really enough to give a clear manager.

bobjacobsen, 2019-11-03:
@rhwood Thanks!

danielb987, 2021-03-10:
@bobjacobsen Is this issue resolved? It seems so, and if so, this should be closed.

bobjacobsen, 2021-03-10:
I think so. I haven't heard of recurrences.

closed 2021-03-10 by bobjacobsen