JMRI/JMRI #7499: setUp and tearDown of tests

pabender, 2019-10-11:
(Maybe move this discussion somewhere better suited?)

Piece-meal teardown is fine for unit tests, which in theory should know what needs to be set up and torn down.   

For running entire apps, I think we should move more toward a sandbox approach. Not quite sure how to do that:  Fork to another JVM for those tests? Run them as separate steps in the infrastructure?  Write some "clean up the JVM regardless" code?  Not clear to me.  But finding every possible loose end and cleaning them up in detail sounds a lot like whack-a-mole to me.

_Originally posted by @bobjacobsen in https://github.com/JMRI/JMRI/pull/7498_

pabender, 2019-10-11:
I think we may want to re-think our entire approach to configuring unit tests.

We have most, if not all, of our tests calling JUnitUtil.setUp and JUnitUtil.tearDown.  One of the functions of that method is to reset the instance manager, but, I think resetting the instance manager is the wrong approach. It's error prone and it  allows tests to infect other tests that follow (and not necessarily just the immediately following tests) because of how the InstanceManger is created.

Our InstanceManager is an example of what is known as the ServiceLocator design pattern.  The instance of the InstanceManager is accessed with a static instance method.

The problem with our current test configuration is that we re-use the same InstanceManager for every single test.  Objects created during one test and not torn down are still accessing the same instance manager the currently active tests are using, which is where state leakage occurs.  (yes, the objects need to be cleaned up as well, but if each test is using a different instance manager, it matters less).

In tests, What we really need to do is move our access of the InstanceManager towards using a Mock InstanceManger we can recreate for each test, which will make our tests more reliable.

The Application start tests (which are currently run via Cucumber) are different than the unit tests.

Incidentally, another thing that will help with test isolation is breaking the code into multiple modules.  When each module can be compiled and tested independently, you can completely shutdown the jvm between tests of each module.

bobjacobsen, 2019-10-12:
I don't understand two of the points made above:

- re InstanceManager:

This test passes:
```
    @Test
    public void testInstanceManagerReset() {
        JUnitUtil.resetInstanceManager();
        ShutDownManager sdm1 = InstanceManager.getDefault(ShutDownManager.class);
        JUnitUtil.resetInstanceManager();
        ShutDownManager sdm2 = InstanceManager.getDefault(ShutDownManager.class);
        
        Assert.assertTrue(sdm1 != sdm2);
    }
```

From this I conclude that we are successfully resetting the InstanceManager itself.  We might be not clearing up the individual objects it managed (see the `Disposable` interface and  tests in InstanceManagerTest), but the InstanceManager itself isn't part of the problem any more due to some nice work by @danielb987 and @rhwood.

 - re modules:  Do you mean modules in the Java 9+ sense?  If so, I must be missing something.  We're quite capable of testing whatever part of the code we want independently, from individual test classes[(http://builds.jmri.org/jenkins/job/Development/job/Separate_Tests/) to large chunks of the tree:
```
./runtest.csh jmri.jmrix.PackageList
```
and we can define whatever suitable chunks of running we'd prefer.  What do Java 9+ modules help with here?  (There may be other benefits, see Issue #7404; here I'm just asking about unit testing)

pabender, 2019-10-12:
> I don't understand two of the points made above:
>
> * re InstanceManager:
>
> This test passes:
>
> ```
>     @Test
>     public void testInstanceManagerReset() { [...]

No, the instance manager IS still a problem.  The problem is this method:

https://github.com/JMRI/JMRI/blob/c40367180386a238d7ed361733ebda3426211244/java/src/jmri/InstanceManager.java#L803

In your test example, if an object is created before the second resetInstanceManager, but starts a thread that eventually calls InstanceManager.getDefault ( or any other static method in the class ) it is going to get the last instance of the instance manager, which may not be the one you want.

Even if objects are torn down correctly, timing issues sometimes prevent them from being fully shutdown until after the next test starts and resets the InstanceManager.  This is part of why we sometimes have errors in tests with automatic initialization of managers during tests.

Also, as it stands right now, we can can’t take advantage of parallel test runners ( in a single JVM ) because of the static calls to the InstanceManager.

> * re modules:  Do you mean modules in the Java 9+ sense?  If so, I must be missing something.  We're quite capable of testing whatever part of the code we want independently, from individual test [...]

I was more thinking in terms of Maven modules, but the concept is not dependent on Maven.  Java 9 modules provide the same concept.

The idea is that you compile,test, and package the modules separately and in dependency order.  If two modules are not dependent on one another, you can actually compile/test/package them in parallel.  Also, if you change a leaf module without touching the core, you can compile/build/test that leaf without touching any other modules. ( here a leaf is a module which is not a dependency for any other module in the tree ).

Right now we may be able to test chucks of the tree, but it is hard to compile smaller chunks of the tree.  If we make it easier to do that, we make it simpler to containerize groups of tests.

It is also hard to compile/build/test the middle of our tree independently.  I.e. I can’t compile/build/test the jmri package without including the jmri.jmrix and jmri.jmrit subtrees. ( when using the PackageTests to run tests, this is a structural issue ).

danielb987, 2019-10-12:
@bobjacobsen 
> From this I conclude that we are successfully resetting the InstanceManager itself. We might be not clearing up the individual objects it managed (see the `Disposable` interface and tests in [...]

My work on the InstanceManager was never merged so the problem I was working on still persist.

pabender, 2019-10-12:
@bobjacobsen to my point on the Instance Manager being a problem in testing, almost every article I have read that mentions testing with a service locator mentions that testing can be a problem.

Wikipedia has this to say ( from https://en.m.wikipedia.org/wiki/Service_locator_pattern#Disadvantages )

> The registry makes code harder to test, since all tests need to interact with the same global service locator class to set the fake dependencies of a class under test. However, this is easily overcome by injecting application classes with a single service locator interface.

[baeldung](https://www.baeldung.com/java-service-locator-pattern) has this to say:

> Finally, let's consider a few reasons to avoid using the Service Locator pattern.
>
> One argument against it is that it makes unit testing difficult. With dependency injection, we can pass mock objects of the dependent class to the tested instance. On the other hand, this is a bottleneck with the Service Locator pattern.

But there are other similar comments elsewhere.

We are NOT using any form of dependency injection, when it comes to the Instance Manager, so we must be using the single instance.

closed 2021-04-14 by stale[bot]