JMRI/JMRI #7504: Problem saving Logix conditionals

devel-bobm, 2019-10-13:
From "Mitch" via jmriusers.org post # 164699:

```
Hey guys,

Just noticed a bug.

Was runnign traisn alll day today on a jenkins build from 3 days ago, editing and saving changes, multiple re-laucnhes to update changes etc.

Decided to download todays Jenkins build, and found a bug in saving files.

Tables > File > Store configuration & panel to file..

On next launch, logix are missing conditionals, some conditionals are there but have nothing in them.

I use Dropbox for file sharing, so i restored my file back to before I updated, and it opened just fine. But saving with the lastets built corrupts the file..

Please note that I have not saved any changes between the file launching correctly and updating to the latest build

Backdating to the build I had 3 days ago, saving works fine.

I don't mind, I'll just stay on this build for a while until I hear news, but I figured I should flag this.

-- 
Thanks
Mitch 
```

silverailscolo, 2019-10-15:
Hoping to eliminate one possible cause: this issue might be side effect of my commit working on iterators for manager persistance, specifically ``DefaultLogixManagerXml.java`` in https://github.com/JMRI/JMRI/pull/7375/files#diff-900fde11be3cb8a0ce437b5f91f5a9ff but I don't see it when reading it again, sorry.

dsand47, 2019-10-15:
At 4.7.4, the Logix/Conditional relationship was formalized as parent/child.  Conditionals are still treated as named beans but they are not registered and therefore are not in the named bean set.

The getSystemNameList method in DefaultConditionalManager builds a list of conditional names by looking at the conditionals for each Logix.  

There are two possible solutions:
- Fix DefaultConditionalManagerXml to use the getSystemNameList/getBySystemName process.
- Implement an overriding getNamedBeanSet method in DefaultConditionalManager.

bobjacobsen, 2019-10-16:
We have lots of tests that successfully read and rewrite panel files containing conditionals.  So something about normal operation must be different from that test case, but I have no idea what it would be.

bobjacobsen, 2019-10-16:
On second look, that's wrong.  It looks like `LoadAndStoreTestBase` is for some reason _not_ detecting that those files are being written without conditional information.  This is because the loadref files have been updated to match the write-without-conditional content, i.e. `java/test/jmri/configurexml/loadref/PanelFileSchemaTest382.xml` doesn't have conditional elements.  Perhaps that'll let somebody track down the change that caused this.

dsand47, 2019-10-16:
@bobjacobsen   The missing conditionals are caused by PR #7375.  DefaultConditionalManagerXml was changed to use getNamedBeanSet instead of getSystemNameList.  Since Conditionals are dependent on a parent Logix, they are not registered.  This results in an empty named bean set.

This affected Logix, LRoutes and Sensor Groups.

dsand47, 2019-10-16:
PR #7515 implements a custom getNamedBeanSet to satisfy the requirement for DefaultConditionalMangaerXml.  It also replaces the Conditionals blocks in 11 loadref XML files.

rhwood, 2019-10-16:
#7515 and #7516 have conflicting resolutions to this issue.

dsand47, 2019-10-16:
> #7515 and #7516 have conflicting resolutions to this issue.

We now have 3 solutions for this issue.

1. Change DefaultConditionalManagerXml to use the old style getSystemNameList to create the XML output.
2. Implement a custom getNamedBeanSet to return the Conditional objects that are referenced by every Logix.  They are not registered since they are dependent on a parent Logix.  This is the approach in #7515.
3. Change DefaultConditionalManager to register Conditional beans.  This is the approach in #7516.

In any of these solutions, the missing Conditional entries have to be added back to the 11 affected loadref test xml files.  #7515 has the xml fixes.

rhwood, 2019-10-17:
The load and store tests, and the focus on them, reveal that the DefaultConditionalManager class *is not* and *was not* tested. This class is the broken class (specifically `getNamedBeanSet()` did not return the set of Conditionals), and it’s not appropriate that the only tests for this are in XML loading tests, given that the broken class does not handle XML.

Put another way DefaultConditionalManagerXml is working 100% correctly, using current, non-deprecated functionality that the DefaultConditionalManager claimed to support. Any solution to this issue must independently test the methods of DefaultConditionalManager before we can verify that any of these solutions actually fix the DefaultConditionalManager.

pabender, 2019-10-17:
> The load and store tests, and the focus on them, reveal that the DefaultConditionalManager class _is not_ and _was not_ tested. This class is the broken class (specifically `getNamedBeanSet()` did [...]

I’ll second this sentiment.  Unit tests SHOULD be testing a unit of the code,  or several units integrated together.  The fix needs to be in the default conditional manager.

dsand47, 2019-10-17:
> I’ll second this sentiment. Unit tests SHOULD be testing a unit of the code, or several units integrated together. The fix needs to be in the default conditional manager.

This brings up an interesting question.  Do class tests also need to test inherited methods?

Since **getNamedBeanSet** is inherited from AbstractManager, testing the methods in **DefaultConditionalManager** would not have included a test for **getNamedBeanSet**.  Since Conditionals are subordinate to a parent Logix, they are not registered since there is no reason to obtain them directly except during panel xml creation.  Prior to the change in question, **DefaultConditonalManagerXml** used **getSystemNameList** to obtain the sorted Conditional set.

Since DefaultConditionalManager **does** inherit from AbstractManager, it **should** support getNamedBeanSet.  PR #7515 adds a custom **getNamedBeanSet** while #7516 enables **register** which will make the inherited getNameBeanSet work properly.

Since #7515 already has the loadef changes to fix the load and store tests, I can go with either the custom getNamedBeanSet or the standard register solution.

danielb987, 2019-10-17:
One way to solve that is to have a base test class that all the test classes inherit. The base class test the method getNamedBeanSet().

Example: (look at the *TestBase classes in JMRI for more examples)

```
public abstract class AbstractManagerTestBase<E extends NamedBean> {

    public Manager<E> m;

    // The implementing class needs to implement this method to provide
    // a set of beans for the test base class.
    public abstract Set<E> getBeans();

    @Test
    public void testGetNamedBeanSet() {
        Assert.assertEquals("sets are equal", m.getNamedBeanSet(), getBeans());
    }
}

public ConditionalManagerTest extends AbstractManagerTestBase<Conditional> {

    public Set<E> getBeans() {
        ....
    }

    public void setup() {
        ...
        m = InstanceManager.getDefault(ConditionalManager.class);
    }
}
```

bobjacobsen, 2019-10-18:
Can we get a consensus on which PR to merge please?

dsand47, 2019-10-19:
7515 re-adds the Conditionals for the 11 loadref xml files. 7516 does not. A custom getNamedBeanSet vs. using “register” is a philosophical question. 

Dave Sand

dsand47, 2019-10-19:
#7515 is a better solution since the custom getNamedBeanSet is guaranteed to have Conditional data integrity since the Conditional list is derived from the parent Logix definitions.

danielb987, 2019-10-19:
> #7515 is a better solution since the custom getNamedBeanSet is guaranteed to have Conditional data integrity since the Conditional list is derived from the parent Logix definitions.

Yes. #7516 introduces a new bug due to this. When a Conditional is removed, it's only removed from the Logix, not from the ConditionalManager. With #7516, it's not possible for the user to remove the Conditionals from the ConditionalManager. So the user will end up with Conditionals stored in the ConditionalManager that are not in use by a Logix.

dsand47, 2019-10-19:
> Yes. #7516 introduces a new bug due to this. When a Conditional is removed, it's only removed from the Logix, not from the ConditionalManager. With #7516, it's not possible for the user to remove [...]

When I restructured the Logix/Conditional relationship in 4.7.4, the Conditional bean was demoted to an object dependent on its parent Logix.  The register and deregister calls were commented rather than removed.  Probably as a reminder.

rhwood, 2019-10-19:
If conditionals are attributes of Logic why is there a ConditionalManager at all. It seems to me that all conditionals should be managed solely by their controlling logix.

Randall

dsand47, 2019-10-19:
> If conditionals are attributes of Logic why is there a ConditionalManager at all. It seems to me that all conditionals should be managed solely by their controlling logix.

If I were to do that project now, that is probably how it would be done.  In 2017, I was just becoming comfortable with Java and the JMRI environment.   

It will be interesting to see what happens with the LogixNG project that I have seen mentioned a few times.

bobjacobsen, 2019-10-20:
PR #7515 has been merged, #7516 marked as WIP pending testing.

dsand47, 2019-10-20:
I have tested the latest Development build along with local builds and the missing Conditionals in the panel xml file problem is resolved.

dsand47, 2019-11-04:
I have tested the fix with the 4.17.5 test release and the problem no longer exists.

devel-bobm, 2019-11-08:
Closing the issue based solely on @dsand47's comment that the problem is resolved in 4.17.5.

closed 2019-11-08 by devel-bobm