JMRI/JMRI #7508: JMRI on macOS Catalina

dsand47, 2019-10-14:
JMRI can be installed and with a little security tweaking it will run.

The current major issue occurs when trying to display tables in PanelPro.

[image]

In Catalina, **/System** is in a read-only APFS volume.  The other data, including user data, is in a separate volume.  In a standard system, they are **Macintosh HD** and **Macintosh HD - Data**.  

A second issue is that under certain circumstances the app is relocated:
```
PWD: '/private/var/folders/bl/1j_ppzws64z9s5kxl0k1wp3r0000gp/T/AppTranslocation/71CFFFB0-20D2-4F29-9725-09C7F4712F60/d'
CLASSPATH: '.:classes:target/classes:jmri.jar:'
Java CMD: '/Library/Java/JavaVirtualMachines/adoptopenjdk-11.jdk/Contents/Home/bin/java'
APPNAME: 'PanelPro'
OPTIONS: ' -noverify -Djava.security.policy=lib/security.policy -Djava.rmi.server.codebase=file:target/classes/ -Djava.library.path=.::lib -Dapple.laf.useScreenMenuBar=true -Dcom.apple.macos.useScreenMenuBar=true -Dfile.encoding=UTF-8 -Xms96m -Xmx2048m'
DOCK_OPTIONS: '-Xdock:name=PanelPro -Xdock:icon=/private/var/folders/bl/1j_ppzws64z9s5kxl0k1wp3r0000gp/T/AppTranslocation/71CFFFB0-20D2-4F29-9725-09C7F4712F60/d/PanelPro.app/Contents/Resources/PanelPro.icns'
Exit Status: 1
```
I am still working on testing install location scenarios.

rhwood, 2019-10-14:
Relocation (at least used to) only occur if the app is not signed. Are you working with a published release of JMRI?

Randall

rhwood, 2019-10-15:
My testing shows that the first error is a symptom of a different issue:
[image]

There is no framework `/System/Library/Frameworks/gluegen-rt.Framework` present on my systems (either upgraded to or newly installed with Catalina), so I think the stack trace is merely showing a fallback location after the signing error.

JMRI 4.16 does run from its correct location (`/Applications/JMRI 4.16`) on my system (I have `/Applications/JMRI 4.12`, `/Applications/JMRI 4.14`, and `/Applications/JMRI 4.16` as my install locations).

rhwood, 2019-10-15:
Updated prior comment with location of gluegen-rt.jnilib

dsand47, 2019-10-15:
Sorry, I forgot to include the first error in my original post.

I removed all of the residual testing files including ~/Preferences/JMRI.  Then I copied a fresh download of 4.16 to /Applications.

When I launch PanelPro, I get the typical prompt about running a downloaded application.

[image]

After setting up a simulated connection and restarting, I tried to display the sensor table.

[image]

Selecting cancel causes the error shown in my first posting.

If I run `xattr -r -d com.apple.quarantine /Applications/JMRI\ 4.16/` then the gluegen error does not occur:  `Initialised JOAL using OpenAL`

This is not a long term solution but it will allow me to use JMRI on Catalina.

bobjacobsen, 2019-10-15:
The jnilib seems to be from JOGL.  The most recent version (a development build) is at https://jogamp.org/deployment/autobuilds/master/gluegen-b916-916/   I don't know whether (1) it would work to just update that or (2) whether both gluegen and JOGL needed to be updated together and (3) if even that would fix this.  There's info on how to get various parts here: https://jogamp.org/wiki/index.php/Downloading_and_installing_JOGL#Using_the_7z_jogamp-all-platforms_archive

Unfortunately  I don't have a way to work on this until at least a week from now,.

mattharris, 2019-10-15:
I'll take a closer look at this.

rhwood, 2019-10-15:
We (JMRI) must sign the *.jnilib files as we ship them as part of JMRI.

https://stackoverflow.com/a/53856274

This *may* also require that we update those libraries, and strip out other libraries like x86 (32 bit) and PowerPC mac libraries, or x86_64 libraries built on a version of macOS older than 10.9.

Randall

bobjacobsen, 2019-10-19:
@rhwood seems to be pointing to having to sign all three jnilibs we distribute, but only libgluegen is being flagged in the discussion.  Is it possible that we're just seeing the 1st error?  (I'm still a couple days away from having a Catalina machine to test with)

mattharris, 2019-10-20:
I believe that to be the case. Given that the OpenAL rely on libgluegen, libjoal and libopenal to function, it's likely that libgluegen is the first one hit in the initialisation phase and that, once that one's signed, we'll get hit by the others.

bobjacobsen, 2019-10-20:
PR #7529 addresses signing of some specific libraries.  See the discussion in that PR as to whether that'll be enough. 

A special dmg with those libraries signed is available from a [dedicated Jenkins job](http://jmri.tagadab.com/jenkins/job/TestReleases/job/Catalina/).  I'd appreciate it if somebody with Catalina could [try it](http://jmri.tagadab.com/jenkins/job/TestReleases/job/Catalina/).  Note that you'll have to copy the "JMRI" folder on the disk image to your main drive; you can't run from the disk image any more.

dsand47, 2019-10-20:
Here are my results from using the special dmg.  The JMRI folder was copied to /Applications.

Using the PanelPro icon in Launchpad:

[image]

After using System Preferences >> Security and Privacy to set Run Anyway, nothing happens on subsequent starts.  The JMRI launcher has been relocated and can't find anything.

```
dmba:log das$ cat launcher.log 
PWD: '/private/var/folders/bl/1j_ppzws64z9s5kxl0k1wp3r0000gp/T/AppTranslocation/5D35CDD7-0A01-43ED-8D84-132BFF1B793C/d'
CLASSPATH: '.:classes:target/classes:jmri.jar:'
Java CMD: '/Library/Java/JavaVirtualMachines/jdk1.8.0_172.jdk/Contents/Home/bin/java'
APPNAME: 'PanelPro'
OPTIONS: ' -noverify -Djava.security.policy=lib/security.policy -Djava.rmi.server.codebase=file:target/classes/ -Djava.library.path=.::lib -Dapple.laf.useScreenMenuBar=true -Dcom.apple.macos.useScreenMenuBar=true -Dfile.encoding=UTF-8 -Xms96m -Xmx2048m'
DOCK_OPTIONS: '-Xdock:name=PanelPro -Xdock:icon=/private/var/folders/bl/1j_ppzws64z9s5kxl0k1wp3r0000gp/T/AppTranslocation/5D35CDD7-0A01-43ED-8D84-132BFF1B793C/d/PanelPro.app/Contents/Resources/PanelPro.icns'
Exit Status: 1
```

If I start PanelPro using StartJMRI directly, the relocation does not happen but the gluegen error still occurs.

I have been running JMRI just fine by running `xattr -rd com.apple.quarantine /Applications/JMRI`. 

**Output requested in the PR:**

```
dmba:JMRI das$ codesign -d -v -v lib/macosx/libjoal.jnilib
Executable=/Applications/JMRI/lib/macosx/libjoal.jnilib
Identifier=libjoal
Format=Mach-O universal (i386 x86_64)
CodeDirectory v=20200 size=679 flags=0x0(none) hashes=17+2 location=embedded
Signature size=9011
Authority=Developer ID Application: Robert Jacobsen (V2UXGA8SJW)
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=Oct 19, 2019 at 9:23:30 PM
Info.plist=not bound
TeamIdentifier=V2UXGA8SJW
Sealed Resources=none
Internal requirements count=1 size=168

dmba:JMRI das$ codesign -vv --deep --strict lib/macosx/libjoal.jnilib
lib/macosx/libjoal.jnilib: resource fork, Finder information, or similar detritus not allowed
dmba:JMRI das$ 
```

If I run `xattr -d com.apple.FinderInfo libjoal.jnilib` then the second code sign returns:
```
dmba:macosx das$ codesign -vv --deep --strict libjoal.jnilib
libjoal.jnilib: valid on disk
libjoal.jnilib: satisfies its Designated Requirement
```

bobjacobsen, 2019-10-20:
@dsand47 Thank you!  That was very helpful.

When you have a chance, can you try another one?  I think that the automated process that Jenkins uses isn't up to date.  So I made the following file manually on my Mac.  If that works OK, I can figure out how to update signer.jmri.org

https://drive.google.com/file/d/1RLaCcF2eOLVlWucUoZU89dtZRahFoaVg/view?usp=sharing

Thanks in advance.

dsand47, 2019-10-20:
> @dsand47 Thank you! That was very helpful.
>
> When you have a chance, can you try another one? I think that the automated process that Jenkins uses isn't up to date. So I made the following file [...]

This version works much better.

The initial start comes up with the expected dialog about running a downloaded program.
[image]
However, the contents of PanelPro.app have been relocated and StartJMRI can't find anything.
```
PWD: '/private/var/folders/bl/1j_ppzws64z9s5kxl0k1wp3r0000gp/T/AppTranslocation/BC77C2C5-ECCB-4ACD-A980-B944FFCA7E64/d'
CLASSPATH: '.:classes:target/classes:jmri.jar:'
Java CMD: '/Library/Java/JavaVirtualMachines/jdk1.8.0_172.jdk/Contents/Home/bin/java'
APPNAME: 'PanelPro'
OPTIONS: ' -noverify -Djava.security.policy=lib/security.policy -Djava.rmi.server.codebase=file:target/classes/ -Djava.library.path=.::lib -Dapple.laf.useScreenMenuBar=true -Dcom.apple.macos.useScreenMenuBar=true -Dfile.encoding=UTF-8 -Xms96m -Xmx2048m'
DOCK_OPTIONS: '-Xdock:name=PanelPro -Xdock:icon=/private/var/folders/bl/1j_ppzws64z9s5kxl0k1wp3r0000gp/T/AppTranslocation/BC77C2C5-ECCB-4ACD-A980-B944FFCA7E64/d/PanelPro.app/Contents/Resources/PanelPro.icns'
Exit Status: 1
```
Subsequent attempts to start whether using the PP Icon or **right click open** on PanelPro result in relocation.

If I drill down and open StartJMRI directly, relocation does not occur and PanelPro starts.  Selecting Tools >> Tables >> Sensor Table **works**.  The gluegen error DOES NOT occur.

If I remove the quarantine from the JMRI package, relocation does not occur and gluegen does not fail.

As far as I can tell, signing is now working.  The next challenge is dealing with relocation.

If I modify StartJMRI to force JMRI_HOME to `/Applications/JMRI`, then starting using the PanelPro icon works.

rhwood, 2019-10-20:
@bobjacobsen With your personal build, I am seeing the same results @dsand47 saw before except I am not having to run `xattr -d com.apple.FinderInfo libjoal.jnilib`:

```
rhwood@Randalls-MacBook-Pro JMRI % codesign -d -v -v /Applications/JMRI/lib/macosx/libjoal.jnilib
Executable=/Applications/JMRI/lib/macosx/libjoal.jnilib
Identifier=libjoal
Format=Mach-O universal (i386 x86_64)
CodeDirectory v=20200 size=679 flags=0x0(none) hashes=17+2 location=embedded
Signature size=9011
Authority=Developer ID Application: Robert Jacobsen (V2UXGA8SJW)
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=Oct 19, 2019 at 10:23:30 PM
Info.plist=not bound
TeamIdentifier=V2UXGA8SJW
Sealed Resources=none
Internal requirements count=1 size=168
rhwood@Randalls-MacBook-Pro JMRI % codesign -vv --deep —strict /Applications/JMRI/lib/macosx/libjoal.jnilib
/Applications/JMRI/lib/macosx/libjoal.jnilib: valid on disk
/Applications/JMRI/lib/macosx/libjoal.jnilib: satisfies its Designated Requirement
```
I have an Apple Developer’s code signing certificate, so if there is a process you want me to test, I can.

However, forcing JMRI_HOME to anything other than the directory JMRI is being run from should be avoided; I don't think I am unique in running multiple released versions of JMRI and expecting those to work without interfering with each other.

rhwood, 2019-10-20:
@bobjacobsen Can you publish `selfsign.php` someplace so at least the signing process can be followed by someone other whoever controls the machine that runs on?

rhwood, 2019-10-20:
Found this is in my computer's logs (not sure which specific log) in Console.app:
```
default	19:44:54.329099-0400	lsd	SecTranslocateCreateSecureDirectoryForURL: created /private/var/folders/y3/8y6xh50x3dvf93rz2xqqnl3m0000gn/T/AppTranslocation/684286BD-68B1-4445-B674-F3BAF2A80DA6/d/DecoderPro.app
default	19:44:54.361793-0400	Dock	LAUNCHING:0x0-0x4e94e9 DecoderPro foreground=1 bringForward=1 seed=1177 userActivityCount=0
default	19:44:54.382483-0400	loginwindow	-[PersistentAppsSupport applicationReady:] | App: DecoderPro, ready, updating active tracking timer
default	19:44:54.382570-0400	loginwindow	-[ApplicationManager checkInAppContext:eventData:] | ApplicationManager: Checked in app : DecoderPro
default	19:44:55.622432-0400	loginwindow	-[PersistentAppsSupport applicationQuit:] | for app:DecoderPro, _appTrackingState = 2
default	19:44:55.622466-0400	loginwindow	-[PersistentAppsSupport applicationQuit:] | App: DecoderPro, quit, updating active tracking timer
```

bobjacobsen, 2019-10-21:
@rhwood The script is in the #7529 branch. I added some details there.

bobjacobsen, 2019-10-21:
@dsand47 Thanks for the detailed info.  Relocation is controlled by the quarantine attribute, so we have to figure out why that's being set.  Could you check it at each step of the process?
 - On the downloaded .dmg
 - On the JMRI folder and .app bundles when the .dmg is mounted
 - On the JMRI folder and .app bundles after they've been copied to /Applications
 - After you first try to run one

My understanding, which is probably wrong, is that they should be present after download and open, but then go away when copied to /Applications.

dsand47, 2019-10-21:
The dmg is quarantined but the contents are not.

When the contents are copied to the filesystem, the quarantine is applied to each file.  There is a flag that is set to x0083.

When the app is run, the confirmation to run a downloaded app occurs.  If run anyway is selected, the quarantine flag on the app tree components is set to x00c3.  The relocation still occurs.

 I did the same check using a bootable Mojave backup and the quarantine process works the same except the flags are x0183 and x01c3.   And the relocation does not occur.

Apple has a new notarization process.  I have no idea what it means, but maybe that has some impact on app relocation.

rhwood, 2019-10-21:
@bobjacobsen Are you sure you sent the signed DMG?

On a re-downloaded copy (mounted and copied to `/Applications/JMRI` using Finder), I get:
```
rhwood@Randalls-MacBook-Pro JMRI % codesign -d -v -v ~/Downloads/JMRI.4.17.5ish+jake+20191020T1949Z+R8a7c223080.dmg
/Users/rhwood/Downloads/JMRI.4.17.5ish+jake+20191020T1949Z+R8a7c223080.dmg: code object is not signed at all

rhwood@Randalls-MacBook-Pro JMRI % codesign -d -v -v /Volumes/JMRI\ 4.17.5ish+jake+20191020T1949Z+R8a7c223080/JMRI/DecoderPro.app 
/Volumes/JMRI 4.17.5ish+jake+20191020T1949Z+R8a7c223080/JMRI/DecoderPro.app: code object is not signed at all

rhwood@Randalls-MacBook-Pro JMRI % codesign -d -v -v /Volumes/JMRI 4.17.5ish+jake+20191020T1949Z+R8a7c223080/JMRI/DecoderPro.app/Contents/MacOS/StartJMRI: code object is not signed at all

rhwood@Randalls-MacBook-Pro JMRI % codesign -d -v -v /Applications/JMRI/DecoderPro.app
/Applications/JMRI/DecoderPro.app: code object is not signed at all

rhwood@Randalls-MacBook-Pro JMRI % codesign -d -v -v /Volumes/JMRI\ 4.17.5ish+jake+20191020T1949Z+R8a7c223080/JMRI/lib/macosx/libgluegen-rt.jnilib 
Executable=/Volumes/JMRI 4.17.5ish+jake+20191020T1949Z+R8a7c223080/JMRI/lib/macosx/libgluegen-rt.jnilib
Identifier=libgluegen-rt
Format=Mach-O universal (i386 x86_64)
CodeDirectory v=20200 size=269 flags=0x0(none) hashes=4+2 location=embedded
Signature size=9011
Authority=Developer ID Application: Robert Jacobsen (V2UXGA8SJW)
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=Oct 19, 2019 at 10:23:51 PM
Info.plist=not bound
TeamIdentifier=V2UXGA8SJW
Sealed Resources=none
Internal requirements count=1 size=176
rhwood@Randalls-MacBook-Pro JMRI % codesign -v -v -d --deep /Volumes/JMRI\ 4.17.5ish+jake+20191020T1949Z+R8a7c223080/JMRI/DecoderPro.app 
/Volumes/JMRI 4.17.5ish+jake+20191020T1949Z+R8a7c223080/JMRI/DecoderPro.app: code object is not signed at all

rhwood@Randalls-MacBook-Pro JMRI % codesign -v -v -d /Applications/JMRI/DecoderPro.app
/Applications/JMRI/DecoderPro.app: code object is not signed at all 
```

For reference, this is what the Eclipse app looks like:
```
rhwood@Randalls-MacBook-Pro JMRI % codesign -v -v -d /Applications/Eclipse\ Java.app 
Executable=/Applications/Eclipse Java.app/Contents/MacOS/eclipse
Identifier=org.eclipse.platform.ide
Format=app bundle with Mach-O thin (x86_64)
CodeDirectory v=20500 size=516 flags=0x10000(runtime) hashes=7+5 location=embedded
Signature size=9079
Authority=Developer ID Application: Eclipse Foundation, Inc. (JCDTMS22B4)
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=Sep 17, 2019 at 2:00:43 AM
Info.plist entries=14
TeamIdentifier=JCDTMS22B4
Runtime Version=10.10.0
Sealed Resources version=2 rules=13 files=1344
Internal requirements count=1 size=184
```

So the problem may be simply that we aren't signing the app bundles correctly.

rhwood, 2019-10-21:
I have tried signing this myself with these results off my signed copy:

```
rhwood@Calculus JMRI % codesign -v -v -d dist/release/JMRI.4.17.5ish+rhwood+20191021T0941Z+Rf65392df92+*
Executable=/Users/rhwood/Developer/JMRI/JMRI/bobjacobsen/JMRI/dist/release/JMRI.4.17.5ish+rhwood+20191021T0941Z+Rf65392df92+signed.dmg
Identifier=JMRI.4.17.5ish+rhwood+20191021T0941Z+Rf65392df92+signed
Format=disk image
CodeDirectory v=20200 size=343 flags=0x0(none) hashes=1+6 location=embedded
Signature size=4736
Authority=Apple Development: Randall Wood (DMNE3U7DE6)
Authority=Apple Worldwide Developer Relations Certification Authority
Authority=Apple Root CA
Signed Time=Oct 21, 2019 at 6:35:46 AM
Info.plist=not bound
TeamIdentifier=K2672N7845
Sealed Resources=none
Internal requirements count=1 size=216
dist/release/JMRI.4.17.5ish+rhwood+20191021T0941Z+Rf65392df92+unsigned.dmg: code object is not signed at all

rhwood@Calculus JMRI % codesign -v -v -d /Volumes/JMRI\ 4.17.5/JMRI/DecoderPro.app
Executable=/Volumes/JMRI 4.17.5/JMRI/DecoderPro.app/Contents/MacOS/StartJMRI
Identifier=jmri.DecoderPro
Format=app bundle with generic
CodeDirectory v=20200 size=207 flags=0x0(none) hashes=1+3 location=embedded
Signature size=4817
Authority=Apple Development: Randall Wood (DMNE3U7DE6)
Authority=Apple Worldwide Developer Relations Certification Authority
Authority=Apple Root CA
Signed Time=Oct 21, 2019 at 6:35:35 AM
Info.plist entries=11
TeamIdentifier=K2672N7845
Sealed Resources version=2 rules=13 files=2
Internal requirements count=1 size=176
```

I've uploaded what I signed to 
https://www.dropbox.com/s/hzcm0s7copoxjg6/JMRI.4.17.5ish%2Brhwood%2B20191021T0941Z%2BRf65392df92%2Bsigned.dmg?dl=0

dsand47, 2019-10-21:
Randall,

I downloaded your dmg. I cannot mount it.

dmba:Desktop das$ codesign -d -v -v JMRI.4.17.5ish+rhwood+20191021T0941Z+Rf65392df92+signed.dmg
Executable=/Users/das/Desktop/JMRI.4.17.5ish+rhwood+20191021T0941Z+Rf65392df92+signed.dmg
Identifier=JMRI.4.17.5ish+rhwood+20191021T0941Z+Rf65392df92+signed
Format=disk image
CodeDirectory v=20200 size=343 flags=0x0(none) hashes=1+6 location=embedded
Signature size=4736
Authority=Apple Development: Randall Wood (DMNE3U7DE6)
Authority=Apple Worldwide Developer Relations Certification Authority
Authority=Apple Root CA
Signed Time=Oct 21, 2019 at 5:35:46 AM
Info.plist=not bound
TeamIdentifier=K2672N7845
Sealed Resources=none
Internal requirements count=1 size=216

Dave Sand

https://www.dropbox.com/s/hzcm0s7copoxjg6/JMRI.4.17.5ish%2Brhwood%2B20191021T0941Z%2BRf65392df92%2Bsigned.dmg?dl=0

bobjacobsen, 2019-10-21:
The machine that does the signing for Jenkins is an older Mac that's using version 10.12 of the signing tools (run `xcrun  --show-sdk-version` to show on your own machine; not this is _not_ the same as either the Xcode nor macOS version). It can't properly notarize the .dmg.  

My laptop is running version 10.15 of the tools, which should be able to do this. 

Not sure what I can do about the signing machine, will look around to see what I can do.  But I'm physically away until Friday, so it'll be at least that long.  In the meantime, I think we should continue to experiment with manual experiments.

rhwood, 2019-10-21:
Hmm. I’ll try again tomorrow morning (I may have used the wrong certificate).

Randall

bobjacobsen, 2019-10-22:
Here's a current file, named "JMRI.4.17.5ish+jake+Catalina+Re3c05ac8b1.dmg":
https://drive.google.com/file/d/1ZHihJI1YMQA-3ZbwdXB57oLuftM1h89m/view?usp=sharing

It's got the internal libraries, the three app bundles and the .dmg signed, and then the .dmg notarized.  As far as I know, this should be correct to validate at each step of the process.

dsand47, 2019-10-22:
> Here's a current file, named "JMRI.4.17.5ish+jake+Catalina+Re3c05ac8b1.dmg":
> https://drive.google.com/file/d/1ZHihJI1YMQA-3ZbwdXB57oLuftM1h89m/view?usp=sharing
>
> It's got the internal libraries, the [...]

Initial testing looks good.

The quarantine flags are x0183/x01c3 instead of x0083/x00c3.  Apparently the "1" removes the requirement to relocate at application startup.  The "1" is probably the result of notarization.

If the JMRI package is moved to ~/Documents or ~/Desktop, StartJMRI does not find any of the jar files in **lib**.  It appears that the `ls` command returns an empty result for scripts in these locations.  I suspect this is related to System Preferences >> Security and Privacy >> Privacy Tab >> Files and Folders which covers Downloads, Documents and Desktop.

I have all of my JMRI stuff outside of these locations.  The only JMRI related items on the Desktop are Aliases to PanelPro in the various JMRI versions.

bobjacobsen, 2019-10-22:
Thank you!  Does it work properly when dragged to /Applications?

dsand47, 2019-10-22:
Yes for **/Applications** and apparently anywhere else outside of Documents, Downloads and Desktop.

rhwood, 2019-10-22:
@bobjacobsen Your latest works in `/Applications/JMRI`, `/Applications/JMRI Test`, and `/Volumes/JMRI 4.17.5/JMRI` (the DMG mounted by double-clicking it in `~/Downloads`). As @dsand47 noted, it does not work from protected locations.

This is the events logged by the OS when launching from `~/Downloads/JMRI`:

```
default	06:42:43.335525-0400	lsd	Non-fatal error enumerating at <private>, continuing: Error Domain=NSCocoaErrorDomain Code=260 "The file “PlugIns” couldn’t be opened because there is no such file." UserInfo={NSURL=PlugIns/ -- file:///Users/rhwood/Downloads/JMRI/DecoderPro.app/Contents/, NSFilePath=/Users/rhwood/Downloads/JMRI/DecoderPro.app/Contents/PlugIns, NSUnderlyingError=0x7f93cd725c20 {Error Domain=NSPOSIXErrorDomain Code=2 "No such file or directory"}}
default	06:42:43.335635-0400	lsd	- 45683955: Checking whether application is managed at file:///Users/rhwood/Downloads/JMRI/DecoderPro.app/jmri.DecoderPro
default	06:42:43.373583-0400	Finder	LAUNCHING:0x0-0xd00d0 DecoderPro foreground=1 bringForward=1 seed=203 userActivityCount=0
default	06:42:43.382838-0400	dmd	Received xpc stream event (distributed notification matching) with name: com.apple.LaunchServices.applicationRegistered user info: {
    bundleIDs =     (
        "jmri.DecoderPro"
    );
    isPlaceholder = 0;
}
default	06:42:43.392026-0400	loginwindow	-[PersistentAppsSupport applicationReady:] | App: DecoderPro, ready, updating active tracking timer
default	06:42:43.392119-0400	loginwindow	-[ApplicationManager checkInAppContext:eventData:] | ApplicationManager: Checked in app : DecoderPro
default	06:42:43.402925-0400	distnoted	register name: com.apple.sharedfilelist.change object: com.apple.LSSharedFileList.ApplicationRecentDocuments/jmri.decoderpro token: f432c pid: 1146
default	06:42:46.742152-0400	lsd	Non-fatal error enumerating at <private>, continuing: Error Domain=NSCocoaErrorDomain Code=260 "The file “PlugIns” couldn’t be opened because there is no such file." UserInfo={NSURL=PlugIns/ -- file:///Users/rhwood/Downloads/JMRI/DecoderPro.app/Contents/, NSFilePath=/Users/rhwood/Downloads/JMRI/DecoderPro.app/Contents/PlugIns, NSUnderlyingError=0x7f93cd5ee750 {Error Domain=NSPOSIXErrorDomain Code=2 "No such file or directory"}}
default	06:42:46.742244-0400	lsd	- 45683955: Checking whether application is managed at file:///Users/rhwood/Downloads/JMRI/DecoderPro.app//jmri.DecoderPro
default	06:42:46.772219-0400	dmd	Received xpc stream event (distributed notification matching) with name: com.apple.LaunchServices.applicationRegistered user info: {
    bundleIDs =     (
        "jmri.DecoderPro"
    );
    isPlaceholder = 0;
}
default	06:42:47.774380-0400	loginwindow	-[PersistentAppsSupport applicationQuit:] | for app:DecoderPro, _appTrackingState = 2
default	06:42:47.774411-0400	loginwindow	-[PersistentAppsSupport applicationQuit:] | App: DecoderPro, quit, updating active tracking timer
```

I can't tell if this indicates we've missed something that needs to be signed, or if we should ship everything inside the .app containers for macOS (this would allow that copy to and launch from temporary location to "just work").

bobjacobsen, 2019-10-23:
I think I have the signing pipeline working now. When people get a chance, please try the installer at

http://builds.jmri.org/jenkins/job/TestReleases/job/Catalina/24/

dsand47, 2019-10-23:
> I think I have the signing pipeline working now. When people get a chance, please try the installer at
> 
> http://builds.jmri.org/jenkins/job/TestReleases/job/Catalina/24/

The quarantine flags went back to x0083/x00c3.  

It appears the notarization did not work:
```
dmba:Downloads das$ spctl --assess --type open --context context:primary-signature --verbose JMRI.4.17.5+Rf65392d.dmg 
JMRI.4.17.5+Rf65392d.dmg: rejected
source=Unnotarized Developer ID
```

rhwood, 2019-10-26:
Java 8 from [https://java.com](https://java.com) cannot be installed on macOS Catalina without disabling Gatekeeper because Oracle isn't notarizing it (they seem to claim that because some security feature of Catalina prevents them from automatically testing and building it, they won't). It also appears that Apple is going begin not allowing Gatekeeper to be disabled in January 2020. This leads to a need to update our installation instructions and process, but:

- Do we just want to point users to another (signed and notarized) JDK from another vendor (like [AdoptOpenJDK](https://adoptopenjdk.net))
- Do we want to embed the JRE we want to run JMRI on? Oracle is *not* providing users versions of Java more recent than 8, with the recommendation that developers who need a newer version embed the JRE in their apps, so its likely we will need to eventually do this anyway.

rhwood, 2019-10-26:
An update to the last comment: AdoptOpenJDK is providing access to newer JREs, it's just not immediately obvious. You have to select the Java version at https://adoptopenjdk.net/releases.html and click on the OS to get the option for the JRE.

bobjacobsen, 2019-10-28:
Catching up on a few parts of this:

 - I think the Jenkins build pipeline is now properly creating signed, notarized and stapled downloads.  I'd appreciate it if people could check [Packages 3485](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/3485/) or [later](http://jmri.tagadab.com/jenkins/job/Development/job/Packages/). 

 - There's some current Catalina status on the JMRI macOS install page [here](https://www.jmri.org/install/MacOSX.shtml#catalina).

 - I'm not sure whether we should point new Catalina users to a non-Oracle JVM, and if so which one.  We're already seeing lots of people running JMRI on later-than-8 JVMs, and we do some checking on that in Jenkins.  The [current advice](https://www.jmri.org/install/MacOSX.shtml#catalina) is to use the Oracle 11 installer which (at least for me) went in without trouble, but we could change that.

 - I don't think we can embed the JRE with JMRI apps unless we do it in each of the DecoderPro, PanelPro, SoundPro apps.  At least, I don't know how to do that.  There are some good reasons to put everything inside those app bundles and distribute them just like normal macOS apps, but there are likely to be problems too.  So perhaps we should create a separate "Package JMRI apps as separate, closed bundles on macOS" as a separate Issue.

dsand47, 2019-10-28:
Package 3485 works fine when copied to /Applications or within the home directory outside of Documents, Downloads or Desktop.

For the 4th item we might consider a unified JMRI.app (ducking my head).  Since it is already possible to transition between DP and PP on the fly, the first run requests the preferred appearance when using the current profile.

rhwood, 2019-10-29:
Package 3485 runs as expected from the DMG or from /Applications. I *did* find a way we could automatically get the location of the installer for AdoptOpenJDK:

```
curl -L 'https://api.adoptopenjdk.net/v2/binary/releases/openjdk11?type=jre&os=mac&openjdk_impl=hotspot&heap_size=normal&release=latest' --output ~/Downloads/foo.tar.gz
tar -xzf ~/Downloads/foo.tar.gz -C ~/Downloads
rm ~/Downloads/foo.tar.gz
```
downloads the OpenJDK 11 JRE for macOS and leaves a working JRE in `~/Downloads`.

(I didn't find a way to get the most recent installer for Java 11 directly although `curl -L 'https://api.adoptopenjdk.net/v2/info/releases/openjdk11?type=jre&os=mac&openjdk_impl=hotspot&heap_size=normal&release=latest'` gets a JSON object that includes the download link for the installer.

rhwood, 2019-10-29:
(Realized I sent that prior comment without completing it).

I think this means we *could* download and install a private JRE in something like (on macOS) `~/Library/Application Support/JMRI/Java` if a system-wide Java is not found.

rhwood, 2019-10-31:
I think the user can specify `-Dlog4j.configuration=/path/to/custom/configuration` in the `default_options` string in `~/Library/Preferences/JMRI/jmri.conf` to change the logging by JMRI on macOS without changing anything in /Applications.

rhwood, 2019-10-31:
> Copying default.lcf to settings: works fine.

We should update that help doc, since it states it can only be copied to the preferences directory, which is not the same as the settings directory.

Randall

dheap, 2019-10-31:
As @dsand47 has said, copying default.lcf to **settings:** has always worked fine for me.

rhwood, 2020-05-12:
I think this can be considered resolved.

closed 2020-05-12 by rhwood