JMRI/JMRI #7522: CMRI "enable polling at startup" not working

garnold335, 2019-10-18:
The CMRI "enable polling at startup" is not working for me in JMRI 4.17.4.
I set up a CMRI connection and added two SMini nodes which I set to "enable polling on startup".
I saved the configuration in the preferences pane.
However when I start Decoder Pro or Panel Pro, the node are showing up with polling not enabled.
As a result I have to manually enable polling on the nodes using CMRInet Manager every time I restart JMRI. 
Is this a known issue? Is there a fix?

devel-bobm, 2019-11-25:
JMRI's default action for C/MRI is to poll only those nodes with defined Sensor objects.  Did you have any sensors defined?

For example, if you have three nodes defined and no sensors defined, JMRI will not poll the nodes.  If you then define sensor CS2003, JMRI will poll node 2 (only).  If you then define CS5, JMRI will only poll nodes 0 and 2.  If you then define CS1011, JMRI will then poll nodes 0, 1, and 2 (not necessarily in that order).

garnold335, 2019-11-30:
Yes, I have two SMINI nodes configured and both have sensors defined.

jerryg2003, 2019-12-18:
Very strange:  I can't get "enable polling at startup" to stay unchecked.  No matter whether I uncheck it when I create a node, or edit after I create a node (and Save and Restart JMRI each time), it still comes back as checked (see also #6433 when I first reported this and posted by Bob J.).  Not a significant problem but interesting that someone sees the opposite issue...

jerryg2003, 2021-03-15:
More information (using 4.23.2):

Still can't get "enable polling at startup" to stay unchecked.  Note the following "unusual" behavior:

1. Uncheck the box on the node configuration pane.  Click "Update node" and "Done".  Get pop-up saying "Remember to save your node info to disk.  Use the Save button in the Pref window."  Click OK, then Done.
2. Click the "Save" button on the Pref window.  After a few seconds the window closes.
3. Click "Quit" on the main PanelPro pane and get pop-up sayving "Preferences has unsaved changes. Do you want to save them?"  Hit "Yes."

When I reenter PanelPro, "Enable polling at startup" is still checked.

closed 2021-07-22 by stale[bot]