JMRI/JMRI #7528: Portals are being renamed when loaded

bobjacobsen, 2019-10-19:
The current head of master will take a portal definition element that looks like this:
```
      <portal systemName="IP3" portalName="2-3">
        <fromBlock blockName="OB102">
          <path pathName="1P3" blockName="OB102" />
        </fromBlock>
        <fromSignal signalName="IF$vsm:DB-HV-1969:block_distant($0003)" signalDelay="0.0" />
        <toBlock blockName="OB103">
          <path pathName="2P4" blockName="OB103" />
        </toBlock>
        <toSignal signalName="IF$vsm:DB-HV-1969:block_distant($0009)" signalDelay="0.0" />
      </portal>
```

and transform the name to an auto-generated name:
```
162c161
<       <portal systemName="IP3" portalName="2-3">
---
>       <portal systemName="IP:AUTO:0003" portalName="2-3">
```

According to `git bisect`, this started with commit 
```
commit 041588496b638fd82f1d40ccce1965beaa2f5afe (HEAD)
Author: Daniel <db123@bergqvist.se>
Date:   Wed Oct 9 13:11:11 2019 +0200
    Let AbstractManager handle auto system names
```

which was merged as part of PR #7495 "Refactor and synchronize lastAuto*Ref in managers".  

Unfortunately, we didn't have any load/store tests for portal elements that would have raised this during the exam process for that PR. We do test validation of XML files with them from `java/test/jmri/jmrit/display/verify`, but those files have various other problems passing the load/store tests, which is probably why they're not used for that.

The first question for this Issue: Is this renaming a problem?  It seems likely to break various use cases to me, but I don't really understand the use of the portal element.  C.f. discussion in PR #7513 (not yet merged due to this)

Second question:  Are there other managers that are now doing this?

Third (which generates the label):  Is this going to be a problem that users encounter, in which case we should hold 4.17.5?

petecressman, 2019-10-19:
Portals were in use for several years before they became NamedBeans.  Until that time and continuing to the present, the user name is the only significant identifier. They were converted to NamedBeans to take advantage of the newly introduced NamedBeanHandle's ability to change user names dynamically. 

During all this time users have been unable define a Portal using a SystemName. No matter how a SystemName for a Portal might be put into the panel xml file, upon loading, OBlockManagerXml's call to PortalManager to create or provide a Portal has always used a null SystemName. Portals are intended to have their SystemNames created upon loading and users are never allowed to dictate a SystemName .  

My regret is not enforcing this rule in PortalManager as well when the conversion to NamedBeans was done. BTW, it is far to late to offer my opinion that this would be desirable for all internal NamedBeans.

danielb987, 2019-10-20:
There are three small changes to revert this for PortalManager, if desired:

Add the field:
`private int _nextSName = 1;`

Change:
`sName = getAutoSystemName();`
to
`sName = generateSystemName();`

Add this method 
```
    public String generateSystemName() {
        String name;
        do {
            name = "IP" + Integer.toString(_nextSName++);
        } while (getBySystemName(name) != null);
        if (log.isDebugEnabled()) log.debug("generateSystemName \"{}\"", name);
       return name;
    }
```

petecressman, 2019-10-20:
Thanks Daniel.  Not really necessary, either way works. Actually, I'd like to delete 
        if (sName == null) {
        etc....
and directly code
       sName = getSystemNamePrefix() + getAutoSystemName();

i.e. continue to not allow system names be determined by users.

I believe if the group insists users must be allowed to do this, then a PortalManagerXml class will need to be written to run before OBlockManagerXml, which will have to be modified extensively.  I am not volunteering to do this. I also oppose any requirement that Portal system names by used for the various links of Portals in OBlocks, OPaths, Warrants, etc.

danielb987, 2019-10-20:
Note that the method `getAutoSystemName()` returns a valid system name so you don't need to do `getSystemNamePrefix() + getAutoSystemName()`.

Does the method `createNewPortal(String sName, String userName)` need the system name as a parameter? If it always creates the system name, I think you could remove the `sName` parameter. That would make the intention clearer, I think.

bobjacobsen, 2019-10-22:
Could we get a resolution of this please?  Either a PR to resolve it, or a determination that there's no user impact.  It's marked as a "Add to Next Test Release" item, i.e. a regression, that's holding up the release.

petecressman, 2019-10-22:
There is no user impact having Portal system names generated at load time.  However some storeAndLoad tests need to be modified to pass. i.e the Portal system names in the xml file are meaningless. I will submit a PR to clean this up, but there is no need to hold up the release for it.

bobjacobsen, 2019-10-30:
Thanks to everybody who worked on this.

closed 2019-10-30 by bobjacobsen