JMRI/JMRI #7541: MacOS and Java

bobjacobsen, 2019-10-24:
(Originally JMRI/website#333 by @Tenderschieber, moved here as it's a development item)

When starting Decoder Pro 4.16+R6f9aced on MacOS it uses
```
JAVA_HOME="/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home"
```
But this isn't supported for years so you face some trouble if you have a newer installation.

The proper line is
```
JAVA_HOME=$( /System/Library/Frameworks/JavaVM.framework/Versions/Current/Commands/java_home )
```
which also causes trouble since the start script uses these lines:
```
function heap_size() {
# get Java heap size
heap=$( "${JAVACMD}" -XX:+PrintFlagsFinal -version 2>/dev/null | grep MaxHeapSize | awk '{print $4}' )
heap=$( expr ${heap} / 1048576 ) # bytes to MB
```
But this causes an error because of a bug in Bash or somewhere else which delivers two lines. This error appears only in non-interactive mode but not interactively.

So the solution (or workaround) is:
```
function heap_size() {
# get Java heap size
heap=$( "${JAVACMD}" -XX:+PrintFlagsFinal -version 2>/dev/null | grep MaxHeapSize | awk '{print $4}' )
heap=$( echo ${heap} | awk '{print $1}' )
heap=$( expr ${heap} / 1048576 ) # bytes to MB
```
It removes the unnecessary value and the shell script starts without any error.

Tenderschieber, 2019-10-24:
I had a look only to this program but I'm sure the other ones have the same issue.

rhwood, 2019-10-25:
@bobjacobsen I’ll look into this over the weekend.

rhwood, 2019-10-26:
> When starting Decoder Pro 4.16+R6f9aced on MacOS it uses
>
> `JAVA_HOME="/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home"`
> But this isn't supported for years so you face some trouble [...]

As documented in the launcher immediately above the line you claim "isn't supported for years", we document that the launcher explicitly looks for Oracle's Java 8, and if that is not present, we use `/usr/libexec/java_home` which is symlinked to `/System/Library/Frameworks/JavaVM.framework/Versions/Current/Commands/java_home`

So basically, it works as expected, and is using the process you recommend if and only if Oracle's Java 8 is not present in its default location. (For me `java_home` as we call it in the script returns:

```
% /usr/libexec/java_home --version 1.8+ --failfast 2>/dev/null
/Library/Java/JavaVirtualMachines/adoptopenjdk-11.jdk/Contents/Home
```

which works as expected.

rhwood, 2019-10-26:
I am also failing to get the `heap_size()` function fail as described.

@Tenderschieber can you provide:
- your macOS version
- output of `$( /usr/libexec/java_home )`
- Java version (we need the complete output of `$( /usr/libexec/java_home )/bin/java -version`)
- output of running (assuming you are running JMRI from `/Applications/JMRI`) `/Applications/JMRI/DecoderPro.app/Contents/MacOS/StartJMRI` in the Terminal.app
- copy of `~/Library/Preferences/JMRI/log/launcher.log` after running DecoderPro with a fixed version of StartJMRI
- copy of `~/Library/Preferences/JMRI/log/launcher.log` after running DecoderPro with StartJMRI as we ship it

Tenderschieber, 2019-10-26:
10.14.6 (18G103)

> /usr/libexec/java_home
/Library/Java/JavaVirtualMachines/jdk-13.0.1.jdk/Contents/Home

> $( /usr/libexec/java_home )/bin/java -version
java version "13.0.1" 2019-10-15
Java(TM) SE Runtime Environment (build 13.0.1+9)
Java HotSpot(TM) 64-Bit Server VM (build 13.0.1+9, mixed mode, sharing)

> ps -ef | grep java
  502 20969 20919   0  9:43pm ttys003    0:12.37 /Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/bin/java -Xdock:name=DecoderPro …

> /Library/Internet\ Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/bin/java -version
java version "1.8.0_202"
Java(TM) SE Runtime Environment (build 1.8.0_202-b08)
Java HotSpot(TM) 64-Bit Server VM (build 25.202-b08, mixed mode)

There's no problem with JDK 8.

I've inserted these lines:
function heap_size() {
    # get Java heap size                                                                                                                                        
    heap=$( "${JAVACMD}" -XX:+PrintFlagsFinal -version 2>/dev/null | grep MaxHeapSize | awk '{print $4}' )
echo "XXX${heap}XXX" >&2
#    heap=$( echo ${heap} | awk '{print $1}' )                                                                                                                  
echo "ZZZ${heap}ZZZ" >&2
    heap=$( expr ${heap} / 1048576 ) # bytes to MB                                                                                                              

When I start it with these lines:
        # Default JRE location                                                                                                                                  
#        JAVA_HOME="/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home"                                                                           
#        JAVA_HOME="/Library/Java/JavaVirtualMachines/jdk-13.0.1.jdk/Contents/Home"                                                                             
#        JAVA_HOME="/Library/Java/JavaVirtualMachines/jdk1.8.0_202.jdk/Contents/Home"                                                                           
        JAVA_HOME="/Library/Java/JavaVirtualMachines/adoptopenjdk-12.jdk/Contents/Home"
#       JAVA_HOME=$( /System/Library/Frameworks/JavaVM.framework/Versions/Current/Commands/java_home )                                                          

I get these from a bash trace:
+++ /Library/Java/JavaVirtualMachines/jdk-13.0.1.jdk/Contents/Home/bin/java -XX:+PrintFlagsFinal -version
+++ grep MaxHeapSize
+++ awk '{print $4}'
++ heap='8589934592
8589934592'
++ echo 'XXX8589934592
8589934592XXX'
XXX8589934592
8589934592XXX
++ echo 'ZZZ8589934592
8589934592ZZZ'
ZZZ8589934592
8589934592ZZZ
+++ expr 8589934592 8589934592 / 1048576
expr: syntax error
The "XXX" and "ZZZ" mark the borders of the variable's content.

Later it exits with 1.

Have also a look to the fallback JDK and the used one.

All together I would say: Whatever your script shall do it doesn't do it.

rhwood, 2019-10-27:
@Tenderschieber please provide the complete outputs requested, not severely truncated and commented version of that.

Both logs show JMRI behaving as expected using the Java version JMRI supports, so please tell us, what are you trying to do that led you to discover this issue to begin with?

bobjacobsen, 2019-10-27:
#7544 caused this to auto-close, but it seems there's more here. Reopened.

Tenderschieber, 2019-10-27:
> Both logs show JMRI behaving as expected using the Java version JMRI supports, so please tell us, what are you trying to do that led you to discover this issue to begin with?

Which Java version do you support?
[launcher-output.log](https://github.com/JMRI/JMRI/files/3775568/launcher-output.log)

rhwood, 2019-10-27:
> Which Java version do you support?

JMRI supports Java version 8, but should run on any LTS version of Java newer that that, preferring an Oracle supplied _JRE_ (not a _JDK_). JMRI instructs the user to download Java from https://java.com; the Java from that site is the Java we support and recommend.

Prior to JMRI version 4.17.5 (our next release), JMRI required a HotSpot JVM that did not have a SoftMaxHeapSize setting to support a specific garbage collection technology; with the next release any JVM should work (this requirement was lifted in #7542 and #7544 which I think means we can close this issue).

The launcher script works in the following way, and is documented to do so in that script:

If (and only if) JAVA_HOME is not defined prior to launching, either on the command line, or in `~/Library/Preferences/JMRI/jmri.conf`:
- Attempt to use an Oracle JRE version for Java 1.8 or 9.
- If an Oracle JRE is not found, use the JRE/JDK returned by `/usr/libexec/java_home --version 1.8+ --failfast`.
- If a JRE/JDK is still not found, prompt the user to install one.

Despite not having received the debugging information requested, the JMRI sources now work with multiple JVMs and multiple garbage collectors on macOS, while still following the above algorithm.

Tenderschieber, 2019-10-27:
> JMRI supports Java version 8

It's not documented on the main page, in the release notes or installation instructions.

> but should run on any LTS version of Java newer that that, preferring an Oracle supplied JRE (not a JDK)

It's not documented on the main page, in the release notes or installation instructions.
: /usr/libexec/java_home --version 1.8+ --failfast
/Library/Java/JavaVirtualMachines/adoptopenjdk-12.jdk/Contents/Home

It also looks for other Java versions and tries to use them.

> If a JRE/JDK is still not found, prompt the user to install one.

It uses a non-supported version.

Let's have a look to the installation instructions:
> Depending on how your Mac is set up, you might need to install Java:
Let's try the easy way first. Open the JMRI folder you just put in Applications, and double click on DecoderPro. If this starts up OK, you have Java installed. Move to the next step. If it opens a dialog box that offers to install Java, or takes you to the Java install web site, just follow those instructions.

> If that doesn't work, and you have a current macOS version, you can get the latest Java JRE installer as a free download at: java.com/download.

Let's see … There's a way it start you guess. It's different from this which is documented in the installation instructions. And both are different from the way the script tries to start.

> Despite not having received the debugging information requested

Which one do you miss?

rhwood, 2019-10-27:
> On Oct 27, 2019, at 8:19 AM, Tenderschieber <notifications@github.com> wrote:
> 
> JMRI supports Java version 8
> 
> It's not documented on the main page, in the release notes or installation instructions.
> 
It is the *only* Java version linked to in 53 locations on the JMRI site, and *is* called out on https://jmri <https://jmri/>.org/
> but should run on any LTS version of Java newer that that, preferring an Oracle supplied JRE (not a JDK)
>
> It's not documented on the main page, in the release notes or installation instructions.
> : [...]
I am totally shocked that this tries to use an existing Java installation if the supported one is not available. As stated repeatedly, both in this conversation, and in the files you have been modifying this is an expected behavior.
> If a JRE/JDK is still not found, prompt the user to install one.
> 
> It uses a non-supported version.
> 
Very well, if that is a problem, submit a PR that *prevents* JMRI from using any version of Java other than 1.8. I won’t. I’ve fixed the issues you have described in this PR, have had them described as not fixed, and do not have any need for this anymore. If you can’t get the current JMRI code at Github.com <http://github.com/> working for you, then submit a fix against that code yourself.
> Let's have a look to the installation instructions:
>
> Depending on how your Mac is set up, you might need to install Java:
> Let's try the easy way first. Open the JMRI folder you just put in [...]
java.com/download <http://java.com/download> offers one version of Java, and that is the supported version 1.8 of Java. It is the very first version tested for in the script that JMRI ships. I can’t help it if your modifications do not behave the way the scripts we have extensively tested behave. Use our scripts and report issues against them.
> Despite not having received the debugging information requested
> 
> Which one do you miss?
> 

The results of running a JMRI installation using the JMRI provided launcher. I do not benefit from the results of a script I don’t have and can’t test. However, unless someone else can demonstrate that the current launcher in JMRI’s repo does not behave as documented within the launcher itself, I don’t need it anymore, since the issues described are either not bugs or are already in code for the next version of JMRI.

bobjacobsen, 2019-10-27:
JMRI attempting to use an alternate JVM isn't a bug.  I'm closing this.

@Tenderschieber, if you're still having a problem with the updated scripts (with #7542 and #7544) please open another Issue that can focus on that.

PR #7544 (2019-10-27): fix: better protect against unexpected heap values
This protects in two ways:
- Increases possiblity that a number will be returned at initial `heap` assignment
- Handles heap being empty later than done in #7542

Fixes #7541 by ignoring `SoftMaxHeapSize` HotSpot JVM setting for ZGC garbage collector.
closed 2019-10-27 by bobjacobsen