JMRI/JMRI #7554: DecoderPro xml: nested Qualifiers buggy

svatoun, 2019-10-29:
Suppose there are nested `<qualifier>` elements, see *diff* file attached that modifies `Comprehensive / Basic Speed` pane; `System.err` logging is added to trace `Qualifier` executions for issues described here. The intent of this is simple: disable (hide) controls that can't be used if CV29/bit 4 is `0` or `1`, depending on the current state.

The qualified items will load OK (hidden, or visible, depending on variable), but a change to the referenced variable will *never* hide them, just show. The code in `QualifierAdder` suggests that qualifiers on parent elements are taken into account, thought the behaviour does not seem to be documented. During debugging, I have found the following defects:

* on load, Qualifiers are processed from inner to outer, **the outermost**  Qualifier's `currentDesiredState` will be used (it will override effects for more nested ones, that are `update()`d first). QualifierCombiner is created, but **not** updated. Because of another defect, the initial appearance is actually fine.
* VariableTableModel `QualifierAdder` subclass attaches `QualifierCombiner` as listener to the `qualifiedVal` **instead of** the watched variables of the constituent Qualifiers. This means the QualifierCombiner will not be notified when one of the contained Qualifiers change. They will act independently, the **outermost** wins, wich will mix with `QualifierCombiner` behaviour (essentially logical AND) if ever applied. Result is buggy, non-determistic behaviour to the user.
* Other `QualifierAdder` anonymous classes attach to the `JPanel` (newGroup, newGridGroup, newColumn, newRow, newGrid, newGridItem) or `JLabel`. But **none** of those `JComponent`s fire `visible` property change (see JComponent documentation), it's fired as a different Swing event. In effect, the `QualifierCombiner` never receives the property change event with broken or erratic behaviour as seen above.
* `QualifierCombiner.propertyChange` is invoked for unrelated Swing property `ancestor` and processes the sub-qualifiers. This happens when the component goes on screen after fully constructed, so it accidentally compensates the "not updated" Qualifier bug mentioned as the first item above. But the intent from the code is clearly to watch out for "variable change" (not hierarchy change) to process the logical AND over the Qualifier chain, which never occurs.

**Additionl non-fatal issues:**
* on change, an individual JComponent's `visible` state might turn `false`, then `true` multiple times: each time for each nested Qualifier, causing unnecessary listeners to be called and the change propagated. Visually this is OK, since all happens within one AWT event, repaint being done in a different event after the component's state settles. Performance impacted, possible unnecessary layout (true -> false -> true, visibility change invalidates layout) performed.
* each `row` and `column` is realized as a JPanel. If the JPanel is hidden (Qualifier = false), it does not matter whether its children are hidden or not; they will disappear from the screen. For entire first `<row>`, one Listener (Qualifier) would be enough; but an instance is attached for each item nested at arbitrary depth in the `row`. So the `CV29' is now watched by 8-9 listeners, instead of 1, each processing CV value. 
* the duplication is especially visible on the "min, avg, max" section, which has `<qualifier>` (equals to `QualifierCombiner`) attached basically to the `grid` element, which equals to `JPanel`. But a `Qualifier` (+outer Qualifier's new instance + QualifierCombiner and a corresponding number of Listeners) is attached to each child in its hierarchy.  Instead of ~1-3 Qualifiers, there is 24 Qualifiers and the appropriate number of Listeners.

I did not log the case for the speed table or other matrix UIs where the number of listeners would explode.

The code in `EnumVariableValue`:
```
int oldVal = getIntValue();
```
does not actually fetch the **old** value, as the value is already changed. No issues *at the moment*, as that (accidentally new) value is passed to `newValue` event parameter. But naming should be cleared, or the code fixed to avoid further confusion.
The attached files:
* [debug.zip](https://github.com/JMRI/JMRI/files/3785952/debug.zip) - patch to the Programmer file + diagnostic output
* [decoder-open.txt](https://github.com/JMRI/JMRI/files/3785958/decoder-open.txt) -  log from decoder window creation; multiple Qualifier executions are logged, as described above.
* [decoder-change.txt](https://github.com/JMRI/JMRI/files/3785957/decoder-change.txt) - log from variable change to "0". Note that no `QualifierCombiner` is logged.

---
Although the code suggests that the qualifiers can be nested, and are combined as AND 
I have scanned all decoder definitions, and found but **single** occurence of nested `Qualifiers` in `RR-CirKits-MotorMan`; in that case, qualifiers watch for fixed decoder ID, so changes are not relevant.

bobjacobsen, 2019-10-29:
Thanks for the detailed work!  Haven't have time yet to go through carefully, but one quick thought:
 
Qualifiers are not just about display, but about the functional existence of a variable:  If de-qualified, the variable no longer takes part in read/write operations.  (This is so you can i.e. have different variables for Mode A and Mode B with different default values) That might be related to how the columns & rows are handled.

dheap, 2019-10-29:
@svatoun Interesting. Many years ago I started working on a unified `SpeedControlPane.xml` intended to replace/merge `BasicSpeedControlPane.xml` and `SpeedTablePane.xml` but I ran into problems with qualifiers (I still have the xml code parked on one of my Macs).

I remember having email conversations with @bobjacobsen at the time, but we didn't come to any resolution and I moved on to other more pressing tasks.

However, I also later came across a (maybe similar) problem in the ESU Function Outputs pane and had to use a dummy variable to get the desired results.

svatoun, 2019-10-29:
The question is whether *nested* Qualifiers should be really supported: could save typing (i.e. outer `exists` qualifier enables UI, while nested `eq`/`ge` enables variants). Or if it is weird enough to be dropped (just one decoder uses it in fact) as nested qualifiers can be replaced by 2 qualifiers at the same level.

I need to test how the change event work in that case - if `QualifierCombiner` does not get processed (that is what I need to test), again the last-processed Listener will win; but that's not AND combination. The mechanism of combining nested qualifiers nad same-level ones should be the same (looking at the code); but need to prove that with a test

dheap, 2019-10-30:
My (hazy) memory is that the `SpeedControlPane.xml` problem I encountered was platform-specific. I tested Mac, Ubuntu and Windows and on one (or more) platform it was necessary to lose focus on the entire pane (not just the changed variable) and then regain it for the pane to refresh. But this may be totally unrelated to the current issue. I don't have time at present to retest...

dheap, 2019-10-30:
> However, I also later came across a (maybe similar) problem in the ESU Function Outputs pane and had to use a dummy variable to get the desired results.

A quick look at `v4fnOutCVs.xml` reveals that I probably used the unqualified dummy variables to force the associated CV to be always included in read/write operations (in contrast to the default behaviour described by @bobjacobsen). So that comment was actually irrelevant to the current issue and `v4fnOutCVs.xml`is behaving as intended with ESU decoders.

svatoun, 2019-10-30:
@bobjacobsen noted. In this case I use purely UI layer; the `Speed Table Definition` variable actually exists defined by in`nmra` folder.
I suppose the upwards traversal in Element hierearchy by `QualifierAdder` is mainly intended for `pane` element hierarchies (it actually stops at `pane` element, variables are IMHO defined in a sibling `decoder` element with no `pane` above them).

closed 2020-04-15 by svatoun