JMRI/JMRI #7561: Impossible to create a transit

nopxor, 2019-10-31:
Hello,
I tried with 4.16 and 4.17.4
I can add the first section to my Transit.
But when I want to add another Section, The list menu is empty.

[image]

Pugwash1, 2019-10-31:
> ---------- Original Message ----------
> creation](https://user-images.githubusercontent.com/4682276/67979513-b952b580-fc1c-11e9-9ec0-e1f08973190e.jpg)

Hi nopxor
The next section(s) that may be added are determined from the section table
entry  in the "Table Of Entry Points", there should be if the Layout Panel is
constructed correctly, at least one  "Block Entering From Entry" that contains a
block in the section adjacent to the first one you added to the transit. Its
likely that either there is a disjointed block, non-contiguos block or unblocked
track.. You can find these on your Layout Panel by using tools->Check-> and then
the type of check you want, one of the first three items should show you the
error.
Steve G.

nopxor, 2019-10-31:
Hi Steve,
Thank you for this quick response.
I will check these blocks.

nopxor, 2019-11-01:
Thanks a lot for your help Steve.
Everything is fine now. I can create transits.
My mistake was that I had not assigned blocks to turnouts.

closed 2019-11-01 by nopxor