JMRI/JMRI #7566: Layout Editor WebServer Occupied Alternative track color 4.17.4

sidlo64, 2019-11-02:
WebServer doesn't show Occupied and Alternative track color in Panel in web browser.
Last OK version is 4.17.3.
How to reproduce:
use version 4.17.4 or 4.17.5
start PanelPro
start WebServer
open panel TestNX.xml
open panel in web browser
set all sensors to inactive
set sensor **S5** to **Active** - in panel clock **B5** become **Red**, in web browser nothing happens
set path from **2** to **12** - in panel blocks **B2**, **B6**, **B7** become **White**, in web browser nothing happens

[Test_NX.xml.zip](https://github.com/JMRI/JMRI/files/3800640/Test_NX.xml.zip)

geowar1, 2019-11-02:
I suspect that this may be caused by #7359 ("replace usernames with systemnames in web xml"); I'll look into it if someone want's to assign it to me.

geowar1, 2019-11-02:
How do I "set path from 2 to 12"?

sidlo64, 2019-11-02:
First buttom blue 2, second button blue 12.
[image]

geowar1, 2019-11-03:
PR# 7567 has passed all tests and is now just waiting on a merge… It would be nice to get the bug fix into 4.17.5.

sidlo64, 2019-11-04:
Good job.
[image]

closed 2019-11-04 by sidlo64