JMRI/JMRI #7590: New Decoder config file for Tams LD-G-32.2

optonix, 2019-11-09:
Hello JMRI community,

I created a new Decoder config file for Tams LD-G-32.2
This decoder is based on the old LD-G-32 but has one electrical output more.
I´m using this decoder quite often and was always missing the possibility configuring the additional output with the LD-G-32 config file.
Hopefully you can add this file to the next releases.

Thanks Martin

[TAMS_LD-G-32_2.zip](https://github.com/JMRI/JMRI/files/3827542/TAMS_LD-G-32_2.zip)

bobjacobsen, 2019-11-10:
Just to confirm, this is an additional decoder definition for a new decoder type, and the old one should be left in place.  Is that correct?  Thanks!

optonix, 2019-11-10:
Yes, correctly this a new decoder definition and the old shall as well stay
in place.
I just found some small last minute bugs in the first version.
Please find attached the corrected version.

closed 2020-01-08 by bobjacobsen