JMRI/JMRI #7598: Window Positions not Remembered on Dual Screens

crosland, 2019-11-10:
Windows 10 with dual screens.

Windows (e.g. console, programmer) do not reopen in their previous position if they were moved to the 2nd screen. Instead they open in the default top left position.

mattharris, 2019-11-10:
Andrew, can you post the content of 'Help > Context' as that will contain the various dimensions of what JMRI knows of the displays.

Thinking that this could be related to window coordinates with negative numbers.

crosland, 2019-11-10:
You may be on to something. My displays are "swapped", by which I mean display 2 is actually my main display (where the Windows task bar, etc., are located). There was a reason for it at the time...:)

JMRI Version: 4.16+R6f9aced   
JMRI configuration file name: null   (from org.jmri.apps.Apps.configFilename system property)
JMRI main window name: DecoderPro: All Entries   
JMRI Application: DecoderPro   
Connection 0: SPROG DCC connected via SPROG on COM6 Disabled false   
Available Communication Ports:
 Found 3 serial ports
 Port: COM6 - in use by: JMRI app   
 Port: COM2 - not in use   
 Port: COM7 - not in use   
Active profile: SPROG II Prog COM6   
Profile location: C:\Users\Daddy\JMRI\SPROG_II_Prog_COM6.jmri   
Profile ID: SPROG_II_Prog_COM6.3eaf763c   
JMRI Network ID: jmri-0A0027000006-3eaf763c
JMRI Storage ID: 56402b6d-a3c8-4947-997e-a70150b1f006
Preferences directory: C:\Users\Daddy\JMRI\SPROG_II_Prog_COM6.jmri\   
Program directory: C:\Program Files (x86)\JMRI4.16   
Roster index location: C:\Users\Daddy\JMRI\SPROG_II_Prog_COM6.jmri\roster.xml   
Current panel file: [none]   
Audio factory type: [not initialised]   
java.version: 1.8.0_151   
java.vendor: Oracle Corporation   
java.home: C:\Program Files\Java\jre1.8.0_151   
java.vm.version: 25.151-b12   
java.vm.vendor: Oracle Corporation   
java.vm.name: Java HotSpot(TM) 64-Bit Server VM   
java.specification.version: 1.8   
java.specification.vendor: Oracle Corporation   
java.specification.name: Java Platform API Specification   
java.class.version: 52.0   
java.class.path: .;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\commons-compress-1.18.jar;lib\commons-io-2.6.jar;lib\commons-lang3-3.7.jar;lib\commons-text-1.2.jar;lib\gluegen-rt.jar;lib\hid4java-0.5.0.jar;lib\jackson-annotations-2.9.8.jar;lib\jackson-core-2.9.8.jar;lib\jackson-databind-2.9.8.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.4.7.v20170914.jar;lib\jetty-io-9.4.7.v20170914.jar;lib\jetty-security-9.4.7.v20170914.jar;lib\jetty-server-9.4.7.v20170914.jar;lib\jetty-servlet-9.4.7.v20170914.jar;lib\jetty-util-9.4.7.v20170914.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jlfgr-1_0.jar;lib\jmdns-3.5.5.jar;lib\jna-4.4.0.jar;lib\jna-platform-4.4.0.jar;lib\joal.jar;lib\json-schema-validator-1.0.10.jar;lib\jsplitbutton-1.3.1.jar;lib\jul-to-slf4j-1.7.25.jar;lib\jython-standalone-2.7.0.jar;lib\libusb4java-1.2.0-linux-arm.jar;lib\libusb4java-1.2.0-linux-x86.jar;lib\libusb4java-1.2.0-linux-x86_64.jar;lib\libusb4java-1.2.0-osx-x86.jar;lib\libusb4java-1.2.0-osx-x86_64.jar;lib\libusb4java-1.2.0-windows-x86.jar;lib\libusb4java-1.2.0-windows-x86_64.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\mqtt-client-0.4.0.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\purejavacomm-1.0.1.jar;lib\slf4j-api-1.7.25.jar;lib\slf4j-log4j12-1.7.25.jar;lib\smtp.jar;lib\thumbnailator-0.4.8.jar;lib\usb-api-1.0.2.jar;lib\usb4java-1.2.0.jar;lib\usb4java-javax-1.2.0.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.4.7.v20170914.jar;lib\websocket-client-9.4.7.v20170914.jar;lib\websocket-common-9.4.7.v20170914.jar;lib\websocket-server-9.4.7.v20170914.jar;lib\websocket-servlet-9.4.7.v20170914.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.1.jar;lib\xercesImpl.jar   
java.library.path: .;lib\x64;lib   
java.compiler: null   
java.ext.dirs: C:\Program Files\Java\jre1.8.0_151\lib\ext;C:\WINDOWS\Sun\Java\lib\ext   
file.encoding: UTF-8   
os.name: Windows 10   
os.arch: amd64   
os.version: 10.0   
python.home: C:\Users\Daddy\JMRI\systemfiles   
python.path: null   
python.cachedir: null   
python.cachedir.skip: null   
python.startup: null   
user.name: Daddy   
user.home: C:\Users\Daddy   
user.dir: C:\Program Files (x86)\JMRI4.16   
user.country: GB   
user.language: en   
user.timezone: Europe/London   
jmri.log.path: C:\Users\Daddy\JMRI\log\   
FileSystemView#getDefaultDirectory(): C:\Users\Daddy\Documents
FileSystemView#getHomeDirectory(): C:\Users\Daddy\Desktop
Default JFileChooser(): C:\Users\Daddy\Documents
Screen size h:1080, w:1920 Inset t:0, b:40; l:0, r:0
Screen resolution: 96
Environment max bounds: java.awt.Rectangle[x=0,y=0,width=1920,height=1040]
bounds[0] = java.awt.Rectangle[x=0,y=0,width=1920,height=1080]
bounds[1] = java.awt.Rectangle[x=0,y=0,width=1920,height=1080]
bounds[2] = java.awt.Rectangle[x=0,y=0,width=1920,height=1080]
bounds[3] = java.awt.Rectangle[x=0,y=0,width=1920,height=1080]
bounds[4] = java.awt.Rectangle[x=0,y=0,width=1920,height=1080]
bounds[5] = java.awt.Rectangle[x=0,y=0,width=1920,height=1080]
Device: \Display0 bounds = java.awt.Rectangle[x=0,y=0,width=1920,height=1080] sun.awt.Win32GraphicsConfig@2466f539[dev=Win32GraphicsDevice[screen=0],pixfmt=182]
bounds[0] = java.awt.Rectangle[x=1920,y=0,width=1920,height=1080]
bounds[1] = java.awt.Rectangle[x=1920,y=0,width=1920,height=1080]
bounds[2] = java.awt.Rectangle[x=1920,y=0,width=1920,height=1080]
bounds[3] = java.awt.Rectangle[x=1920,y=0,width=1920,height=1080]
bounds[4] = java.awt.Rectangle[x=1920,y=0,width=1920,height=1080]
bounds[5] = java.awt.Rectangle[x=1920,y=0,width=1920,height=1080]
Device: \Display1 bounds = java.awt.Rectangle[x=1920,y=0,width=1920,height=1080] sun.awt.Win32GraphicsConfig@316c9c8c[dev=Win32GraphicsDevice[screen=1],pixfmt=182]
JmriInsets t:0, b:40; l:0, r:0
[... cut ...]
ZeroConfService host: jmri-0a0027000006-3eaf763c-1 running 0 service(s)
ZeroConfService host: jmri-0a0027000006-3eaf763c-1 running 0 service(s)
ZeroConfService host: jmri-0a0027000006-3eaf763c-2 running 0 service(s)
ZeroConfService host: jmri-0a0027000006-3eaf763c-2 running 0 service(s)
ZeroConfService host: jmri-0a0027000006-3eaf763c running 0 service(s)
ZeroConfService host: jmri-0a0027000006-3eaf763c running 0 service(s)
ZeroConfService host: jmri-0a0027000006-3eaf763c running 0 service(s)
ZeroConfService host: jmri-0a0027000006-3eaf763c-1 running 0 service(s)

dheap, 2019-11-11:
@crosland I think the problem is more general than you  assume.

I've never been able to retain window positions on the second display of any of my multiple-monitor Macs.

I don't have the main displays swapped.

mattharris, 2019-11-12:
Seems the negative numbers thing was a red-herring.

In one of my multi-monitor setups, JMRI successfully records the positions, but this is a side-effect of some subtleties of the current implementation.

Essentially, we only look at the size of the Default display and then ensure that the proposed location is not greater than or equal to the extent. Where the Default display is to the right-hand side of the secondary display, this calculation means that all window positions are restored.

On one of my multi-monitor setups, I have 3 screens with the Default in the centre. In this case, windows on the left-hand screen have there position saved and restored, but those on the right-hand screen are not - these end up back on the Default (centre) screen.

Given that there are use cases where we wouldn't necessarily want to 'force' a window to be solely contained within the context of one screen, we should purely look to loop through the various screens available to verify that the proposed top-left position of the window is a valid coordinate for at least one screen and, only when no available screens can handle this, fall-back to the Default screen.

I'll take a deeper look into this.

mattharris, 2019-11-13:
Consider also #7613 whilst looking into this problem as it's the same area of the code.

mattharris, 2019-11-13:
@crosland & @dheap would it be possible for you both to provide context information from your multi-monitor setups with the latest HEAD (that includes the updates from #7614)

This will help to understand the way that the various screen dimensions are being reported.

Thanks.

mattharris, 2019-11-13:
#7598 has a potential fix for this.

dheap, 2019-11-14:
@mattharris Still not working as I think  it should. All windows go back to primary screen when reopened.
[MacPro context.txt](https://github.com/JMRI/JMRI/files/3844208/MacPro.context.txt)
[iMac context.txt](https://github.com/JMRI/JMRI/files/3844209/iMac.context.txt)

mattharris, 2019-11-14:
Hmmm, there's a few more subtleties going on as what's reported back in the context is roughly what's expected.

For a point of comparison, here's context from my 3-screen Windows 10 setup.
[Win10-3-screens.context.txt](https://github.com/JMRI/JMRI/files/3845349/Win10-3-screens.context.txt)

To gain a better understanding of what's going on, could you try running with:

`log4j.category.jmri.util.JmriJFrame=DEBUG`

added to `default.lcf` and provide back the Console log? There are a whole bunch of debug statements added that might aid us in this.

Thanks again for your help.

dheap, 2019-11-14:
Mac Pro
[MacPro context.txt](https://github.com/JMRI/JMRI/files/3845818/MacPro.context.txt)
[session 1.txt](https://github.com/JMRI/JMRI/files/3845819/session.1.txt)
[session 2.txt](https://github.com/JMRI/JMRI/files/3845821/session.2.txt)

dheap, 2019-11-14:
In the first session I moved "JMRI System Console" and "NCE Simulator: Command Monitor" to secondary display, arranged side by side, closed and opened them. Both moved themselves back to primary display on reopen, with some loss of dimensions.

Opened second session and both had shrunk to same smaller size.

crosland, 2019-11-14:
My main screen is screen 2 in Windows, screen 1 is to the right of it.

What I see now:

Drag a window to the right, onto screen 1 and close the application.
Reopen and the vertical position is remembered, but the window always reverts to the right hand side of the main screen.

Context and log pasted below

Context:

JMRI Version: 4.17.6ish+Daddy+20191114T1128Z+R9d25976   
JMRI configuration file name: null   (from org.jmri.apps.Apps.configFilename system property)
JMRI main window name: PanelPro   
JMRI Application: PanelPro   
Connection 0: SPROG DCC connected via SPROG on COM6 Disabled false   
Available Communication Ports:
 Found 3 serial ports
 Port: COM6 - in use by: JMRI app   
 Port: COM2 - not in use   
 Port: COM7 - not in use   
Active profile: SPROG II Prog COM6   
Profile location: C:\Users\Daddy\JMRI\SPROG_II_Prog_COM6.jmri   
Profile ID: SPROG_II_Prog_COM6.3eaf763c   
JMRI Network ID: jmri-0A0027000006-3eaf763c
JMRI Storage ID: 56402b6d-a3c8-4947-997e-a70150b1f006
Preferences directory: C:\Users\Daddy\JMRI\SPROG_II_Prog_COM6.jmri\   
Program directory: D:\JMRI   
Roster index location: C:\Users\Daddy\JMRI\SPROG_II_Prog_COM6.jmri\roster.xml   
Current panel file: [none]   
Audio factory type: [not initialised]   
java.version: 1.8.0_25   
java.vendor: Oracle Corporation   
java.home: C:\Program Files\Java\jdk1.8.0_25\jre   
java.vm.version: 25.25-b02   
java.vm.vendor: Oracle Corporation   
java.vm.name: Java HotSpot(TM) 64-Bit Server VM   
java.specification.version: 1.8   
java.specification.vendor: Oracle Corporation   
java.specification.name: Java Platform API Specification   
java.class.version: 52.0   
java.class.path: D:\JMRI;D:\JMRI\lib\purejavacomm-1.0.1.jar;D:\JMRI\lib\xercesImpl.jar;D:\JMRI\lib\xml-apis-1.4.01.jar;D:\JMRI\lib\gluegen-rt.jar;D:\JMRI\lib\javacsv.jar;D:\JMRI\lib\jdom-2.0.5.jar;D:\JMRI\lib\jdom.jar;D:\JMRI\lib\jhall.jar;D:\JMRI\lib\jinput.jar;D:\JMRI\lib\jmdns-3.5.5.jar;D:\JMRI\lib\openlcb.jar;D:\JMRI\lib\jlfgr-1_0.jar;D:\JMRI\lib\joal.jar;D:\JMRI\lib\jython-standalone-2.7.1.jar;D:\JMRI\lib\log4j-1.2.17.jar;D:\JMRI\lib\javax.servlet-api-3.1.0.jar;D:\JMRI\lib\vecmath-1.5.2.jar;D:\JMRI\lib\mailapi.jar;D:\JMRI\lib\smtp.jar;D:\JMRI\lib\xAPlib.jar;D:\JMRI\lib\jna-4.4.0.jar;D:\JMRI\lib\jna-platform-4.4.0.jar;D:\JMRI\lib\commons-lang3-3.7.jar;D:\JMRI\lib\commons-text-1.2.jar;D:\JMRI\lib\commons-compress-1.18.jar;D:\JMRI\lib\jackson-core-2.10.0.jar;D:\JMRI\lib\jackson-annotations-2.10.0.jar;D:\JMRI\lib\jackson-databind-2.10.0.jar;D:\JMRI\lib\jetty-http-9.4.7.v20170914.jar;D:\JMRI\lib\jetty-security-9.4.7.v20170914.jar;D:\JMRI\lib\jetty-server-9.4.7.v20170914.jar;D:\JMRI\lib\jetty-servlet-9.4.7.v20170914.jar;D:\JMRI\lib\jetty-util-9.4.7.v20170914.jar;D:\JMRI\lib\jetty-io-9.4.7.v20170914.jar;D:\JMRI\lib\websocket-api-9.4.7.v20170914.jar;D:\JMRI\lib\websocket-client-9.4.7.v20170914.jar;D:\JMRI\lib\websocket-common-9.4.7.v20170914.jar;D:\JMRI\lib\websocket-server-9.4.7.v20170914.jar;D:\JMRI\lib\websocket-servlet-9.4.7.v20170914.jar;D:\JMRI\lib\slf4j-api-1.7.25.jar;D:\JMRI\lib\slf4j-log4j12-1.7.25.jar;D:\JMRI\lib\jul-to-slf4j-1.7.25.jar;D:\JMRI\lib\xbee-java-library-1.2.1.jar;D:\JMRI\lib\pi4j-core-1.2.jar;D:\JMRI\lib\commons-io-2.6.jar;D:\JMRI\lib\bluecove-2.1.1-SNAPSHOT.jar;D:\JMRI\lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;D:\JMRI\lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;D:\JMRI\lib\jhidrawplugin.jar;D:\JMRI\lib\beansbinding-1.2.1.jar;D:\JMRI\lib\libusb4java-1.2.0-linux-arm.jar;D:\JMRI\lib\libusb4java-1.2.0-linux-x86.jar;D:\JMRI\lib\libusb4java-1.2.0-linux-x86_64.jar;D:\JMRI\lib\libusb4java-1.2.0-osx-x86_64.jar;D:\JMRI\lib\libusb4java-1.2.0-windows-x86.jar;D:\JMRI\lib\libusb4java-1.2.0-windows-x86_64.jar;D:\JMRI\lib\usb-api-1.0.2.jar;D:\JMRI\lib\usb4java-1.2.0.jar;D:\JMRI\lib\usb4java-javax-1.2.0.jar;D:\JMRI\lib\hid4java-0.5.0.jar;D:\JMRI\lib\mqtt-client-0.4.0.jar;D:\JMRI\lib\json-schema-validator-1.0.10.jar;D:\JMRI\lib\jinputvalidator-0.8.0.jar;D:\JMRI\lib\jsplitbutton-1.3.1.jar;D:\JMRI\lib\thumbnailator-0.4.8.jar;D:\JMRI\target\classes   
java.library.path: D:\JMRI;D:\JMRI\lib\windows\x64;D:\JMRI\lib\windows;D:\JMRI\lib   
java.compiler: null   
java.ext.dirs: C:\Program Files\Java\jdk1.8.0_25\jre\lib\ext;C:\WINDOWS\Sun\Java\lib\ext   
file.encoding: UTF-8   
os.name: Windows 8.1   
os.arch: amd64   
os.version: 6.3   
python.home: null   
python.path: null   
python.cachedir: null   
python.cachedir.skip: null   
python.startup: null   
user.name: Daddy   
user.home: C:\Users\Daddy   
user.dir: D:\JMRI   
user.country: GB   
user.language: en   
user.timezone: Europe/London   
jmri.log.path: C:\Users\Daddy\JMRI\log\   
FileSystemView#getDefaultDirectory(): C:\Users\Daddy\Documents
FileSystemView#getHomeDirectory(): C:\Users\Daddy\Desktop
Default JFileChooser(): C:\Users\Daddy\Documents
Screen size h:1080, w:1920 Inset t:0, b:40; l:0, r:0
Screen resolution: 96
Environment max bounds: java.awt.Rectangle[x=0,y=0,width=1920,height=1040]
Device: \Display0 virtualBounds = java.awt.Rectangle[x=0,y=0,width=1920,height=1080]
Device: \Display0 virtualInsets = java.awt.Insets[top=0,left=0,bottom=40,right=0]
Device: \Display0 bounds = java.awt.Rectangle[x=0,y=0,width=1920,height=1080] D3DGraphicsConfig[dev=D3DGraphicsDevice[screen=0],pixfmt=0]
[... 106 lines cut ...]
2019-11-14 11:35:39,173 util.JmriJFrame                       DEBUG - getPreferredSize "SPROG Console" returns width 797 height 597 [AWT-EventQueue-0]
[... cut ...]
2019-11-14 11:35:39,188 util.JmriJFrame                       DEBUG - getPreferredSize "SPROG Console" returns width 797 height 597 [AWT-EventQueue-0]
2019-11-14 11:35:39,188 util.JmriJFrame                       DEBUG - reSizeToFitOnScreen sets height 597 width 797 [AWT-EventQueue-0]
2019-11-14 11:35:55,585 util.JmriJFrame                       DEBUG - dispose "SPROG Console - Connected to SPROG IIv4  v4.4" [AWT-EventQueue-0]
2019-11-14 11:35:55,601 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]
apps.PanelPro.PanelPro finished with return code 0
BUILD SUCCESSFUL (total time: 37 seconds)

mattharris, 2019-11-14:
@dheap @crosland  thanks for those traces - seem to confirm my thoughts that it's the `reSizeToFitOnScreen` that needs some tweaking. The basic storage and retrieval code now works as when moving the initial PanelPro frame to a second screen it stays there on subsequent launches.

Just need to find why that's not the case for other classes of window...

Pugwash1, 2019-11-14:
The code keeps looking at the size of the physical screen, not the size of the desktop. 
`// First, ask for the physical screen size
                Dimension screen = getToolkit().getScreenSize();`
It needs to ask the size of the desktop, and possible the viewportal into the desktop. If you have unequal size screens, life can get very complecated. The treatment varies by operating system, and sometimes window manager.

Pugwash1, 2019-11-14:
replace getToolkit().getScreenSize() with
`    private Dimension getMonitorSizes() {        
        GraphicsEnvironment ge = GraphicsEnvironment.getLocalGraphicsEnvironment();
        Rectangle fred = ge.getMaximumWindowBounds();  // this doesnt do what you think, it still return max single screen size.
        Dimension dim = new Dimension(0,0);
        // construct the desktop. 
        for (GraphicsDevice gd : ge.getScreenDevices()) {
            Rectangle bounds = gd.getDefaultConfiguration().getBounds();
            dim.width = (int) bounds.getWidth()+ bounds.x > dim.width ? (int) bounds.getWidth()+ bounds.x : dim.width;
            dim.height = (int) bounds.getHeight()+ bounds.x > dim.height ? (int) bounds.getHeight()+ bounds.x : dim.height;
        }    
        return dim;
    }
`

See if it works for Mac and WIndows. Works for X  and Wayland using GTK

mattharris, 2019-11-15:
Related item to consider whilst looking at this (from jmriusers):

https://groups.io/g/jmriusers/message/165869

> Group:
> 
> Using Raspberry PI (Raspbian) with JMRI I'd like to size the WiThrottle pane large enough to show 10 or so throttles... No keyboard or mouse running "headless" so I can't resize on the fly.
> 
> If re-sized and JMRI closed and restarted yes.,.. power down and restarted no. With Raspbian (or Mint 19.2) JMRI doesn't remember... Anyone working  on this? Reason is I want to see/show non tech savvy club members just what's going on with their WiThrottle server throttles. I have a Pi with a nice large display attached...
> 
> I can do that on my Mac using prefs while on the pane and saving while there and that then works nicely.
> 
> Jim Albanowski

dheap, 2019-11-15:
Window size and position is saved into the (profile and machine)-specific user-interface.xml file. My understanding is that this happens as part of the window close/dispose procedure.

Jim seems to be saying that the size and position are saved if JMRI is shut down properly but not if the PI is simply powered off (without JMRI shutting down properly). If I'm reading his post right, it's inevitable that the size and position will not be saved with a hard power-off.

dsand47, 2019-11-15:
The location and size for JmriJFrame windows in user-interface.xml  is updated **_whenever_** the location or size is changed.  This does depend on the saveSize and savePosition booleans when the window is created.

Whether a component uses the user-interface.xml location and size during window creation depends on the  code.

Pugwash1, 2019-11-16:
As I Stated. The problem is that code is looking at the size of the default screen, not the size of the desktop. The code change  I suggest works in Linux, does it work for Apple and Microsoft.

dheap, 2019-11-16:
@Pugwash1 

> replace getToolkit().getScreenSize() with
... 
> See if it works for Mac and WIndows. Works for X and Wayland using GTK

It partially worked for me on Mac. It correctly returned:
`DEBUG - getScreenSize returns W:4480.0, H:1440.0 [AWT-EventQueue-0]`
`DEBUG - Detected 2 screens. [AWT-EventQueue-0]`

Instead of the incorrect:
`DEBUG - getScreenSize returns W:2560.0, H:1440.0 [AWT-EventQueue-0]`
`DEBUG - Detected 2 screens. [AWT-EventQueue-0]`

But it didn't stop all my windows moving back to display 1 on restart.

Pugwash1, 2019-11-16:
@dsand47 As @dsand47 pointed out the second part of the problem is not all windows save size and/or position. And the third issue is not all windows appear to use Jmrijframe correctly.

mattharris, 2019-11-16:
This algorithm returns the dimensions of a single rectangle fully encompassing the sum of the individual screen extents. This rectangle may include non-visible portions depending on position and geometry of each individual display.

Whilst this algorithm may be ok for certain use cases, the majority of cases we need to consider each screen extent separately.

I've got the basics of an algorithm worked out, just need to fully implement it.

In essence, we find which display would be the one that contains the window (based upon top-left origin) and then use the extent of that display to perform our resize & repositioning.

Pugwash1, 2019-11-16:
@dheap @dsand47 It would appear that some windows you need to close  the window and then close jmri, the position and size is remembered, but if you just close jmri with the windows open it doesnt...

dheap, 2019-11-16:
@Pugwash1 My tests with that code included the JMRI System Console and I monitored the contents of user-interface.xml in real time. The changes to the recorded size of  occurred instantly, not on closure. The recoded size/position were correct at the time and persisted through shutdown and the first part of restart. The size/position was messed up as the window was reopened.

mattharris, 2019-11-16:
There's an updated version of the algorithm in #7628 which considers the extents of the display upon which the origin of the window is detected.

This will likely not cope fully with windows that span multiple screens, but is (hopefully) an improvement upon previous behaviour.

dheap, 2019-11-17:
See my comment in #7628. It now works for me on Mac.

crosland, 2019-11-26:
Apologies for the delay, but it's working on Windows too.

mattharris, 2019-11-26:
@crosland thanks for the confirmation. Will close this issue now.

closed 2019-11-26 by mattharris