JMRI/JMRI #7607: DecoderPro 4.17.5 delete roster.xml

sidlo64, 2019-11-12:
DecoderPro 4.17.5 delete roster.xml
version 4.17.4 is OK
How to reproduce
Install 4.17.5 or 4.17.6ish
Start DecoderPro, add one or more record to roster.
Stop DecoderPro.
Start DecoderPro. Table Roster is empty.
Stop DecoderPro. This operation delete roster.xml.

bobjacobsen, 2019-11-18:
I haven't been able to recreate this.  Could you please attach your `messages.log` file to this Issue please?  (You might have to .zip it or add .txt to the filename to get GitHub to accept it)

sidlo64, 2019-11-20:
I did a small research.

JMRI-4.17.5

Windows 7 Home Premium Service Pack 1 -- not affected
OS X Yosemite 10.10.5 -- not affected
Ubuntu 18.04.3 LTS -- has the problem described above

Attached files are preference directory.
First file after add one record to roster.
Second file deleted record after second start DecoderPro.
There is nothing strange in the log file.

[jmri-1-created-record.zip](https://github.com/JMRI/JMRI/files/3871264/jmri-1-created-record.zip)
[jmri-2-deleted-record-Linux.zip](https://github.com/JMRI/JMRI/files/3871265/jmri-2-deleted-record-Linux.zip)

sidlo64, 2020-01-07:
In version JMRI-4.18 it works.

closed 2020-01-07 by sidlo64