JMRI/JMRI #7612: Block and Memory Variables Values show TagID not User Name on Web Server

mstevetodd, 2019-11-13:
On a Layout editor panel, when a memory variable or block icon is included, and linked to an RFID Reporter, RFIDs tags show in the block or MV with their user names defined in the ID Tags table.
However, when that same panel is displayed on the web it shows the System Name in the form ID04D1949A74. This is the value shown in the Value column of the Blocks table. However, this is converted on the native panel into the User Name defined in the ID Tags table.
see https://groups.io/g/jmriusers/topic/49451297 for initial request and discussion.

mstevetodd, 2019-11-13:
Method to recreate is as follows:

Open panel using CAN Simulation
Ensure Web Server started
Open web browser and show panel (http://localhost:12080/panel/Layout/Block%20on%20Web%20Problem )
Open Merg->Merg Send CAN Frame dialog
Complete the Frame Packets 1 ,2 and 3 with:
X FA 000B 0470949A74
X FA 000B 0000000002
X FA 000B 0000000003
Tick the checkboxes next to frames
Un-tick the "Send as outgoing CanMessage
Tick the "Send as incoming CanReply
Click 'Start'
Note how 'native' panel shows the User Names set up in id Tags table
Note how browser panel shows the System Names without translation

mstevetodd, 2019-11-13:
Attaching screenshots and example files provided by Andy Brown:
[BlockOnWeb.zip](https://github.com/JMRI/JMRI/files/3842264/BlockOnWeb.zip)

geowar1, 2019-11-14:
This may be related to #7335, #7338 and/or #7359. Read discussion at [https://groups.io/g/jmriusers/topic/layout_editor_panel_with_web/32919542](url)

mstevetodd, 2019-11-22:
This issue is much deeper than expected. 
The json Reporter logic didn't match what was shown in Tools, Tables, Reporter.  Fixed.
The memory value shown in Tools, Tables, Memory Variables shows the "system name" of the tag, not what is displayed on the panel. I don't see (yet) how the panel "knows" to display something else for these items. I expected the Memory to be set to the Reportable string, so downstream logic could "just work". Out of time for now.

mstevetodd, 2019-11-26:
Memories can store various object types, so the panel display logic handles each type individually. Supported types are String, Number, ImageIcon, IdTag, Reportable.
I copied this logic from jmri.jmrit.display.layoutEditor.MemoryIcon into the json memory code, and the client-side panel now matches the JMRI panel for the example provided. Skipping ImageIcon until I get an example panel using that.

danielb987, 2019-11-27:
> Seems like there should be a single place to put this code....

My suggestion is to put it in `jmri.Memory` if possible. Like adding something like `getIndirectValue()` to Memory. Maybe even adding two methods. `getIndirectValue()` that handles String, Number, IdTag and Reportable. And `getIndirectGUIValue()` that also handles ImageIcon.

I'm working on LogixNG and currently working on reading and writing Memory in LogixNG. If Memory has a `getIndirectUserName()`, it could make sense to not only be able to check if the Memory itself is equal to a value, but also be able to check if the report in the Memory is equal to a value.

So if you add this method to Memory, I would be able to use it in LogixNG.

rhwood, 2019-11-27:
The JSON client should be able to get and manipulate (if possible) the JSON representation of the *actual* memory value if that Memory is an object that can be represented in JSON, not what one specific use case of the Memory may need for its purposes.

I.e., if the Memory contains an IdTag, the JSON Memory object should contain an IdTag, and it should be the JSON client's responsibility, not the JMRI server's responsibility to correctly render those contents. This is for multiple reasons:
- The JSON client *may be* a connection between JMRI and a layout
- The web panel JavaScript is not the only JSON client
- The JSON client *is not* expected to be the same locale as the JMRI server

We don't want the JSON code in Memory, for the same reason we don't have the XML code in and NamedBean (we only load the XML or JSON code if/when needed and can allow those classes to be GCed when not needed).

mstevetodd, 2019-12-09:
Closed via #7670 and will be in 4.17.7 and later.

closed 2019-12-09 by mstevetodd