JMRI/JMRI #7613: Save window state (minimized, maximized, normal)

mstevetodd, 2019-11-13:
When saving JMRI window positions, is it possible (or desirable?) to also save each window's "state" (minimized, maximized, normal)? And then use that info to restore it to the same state when starting up?

See https://groups.io/g/jmriusers/topic/hiding_web_only_panels/53431959 for background discussion.

mattharris, 2019-11-13:
Seems like a good idea. As I'm looking into stuff in this area, I'll see what possibilities there are.

Guessing that we'd want it to be a configurable option.

bobjacobsen, 2019-11-13:
> When saving JMRI window positions, is it possible (or desirable?) to also save each window's "state" (minimized, maximized, normal)?

There’s an unresolved question as to whether this should be done via panel file content (stored intentfully and only occasionally) vs preferences (stored automatically every time).

Originally, it worked solely via panel files, but the current behavior is some kind of hybrid.

geowar1, 2019-11-14:
The conclusion I thought was reached last time this was discussed was for the panel file to have the default size & position but for the machine specific preferences to have priority.

Pugwash1, 2019-11-14:
Hmmm. I can understand an action specified during start up to open minimized. But if I open a window, that was minimized when I last exited, and it doesn't show up I'm going to get annoyed. When I open  a window I expect to see it where it was the last time it was visible.
I thought all the windows inherited from jmrijframe, or something similarly named, where all this save and restore took place.

dsand47, 2019-11-14:
> The conclusion I thought was reached last time this was discussed was for the panel file to have the default size & position but for the machine specific preferences to have priority.

Correct.  The machine specific preferences (user-interface.xml) are updated as windows are moved and take precedence during panel loading.  The panel definitions have the last **_saved_** location and size which are used when the machine specific preferences do not exist.

mstevetodd, 2021-03-30:
I don't believe this has been addressed, but still seems to be a valid request.

closed 2021-05-29 by stale[bot]