JMRI/JMRI #7615: JMRI Panels support HTML in text fields, but client-side web panels do not

mstevetodd, 2019-11-13:
HTML can be included in text fields in Java, and JMRI users are using it to display rich text data in panel items such as memory variables. Note: the string must be wrapped in an <html> tag. 
However, the JMRI client-side web panel currently outputs the HTML string instead of rendering it.

This issue is to determine if this behavior can be changed to match the JMRI panel.

See https://groups.io/g/jmriusers/topic/53014832 for background discussion.

mstevetodd, 2019-11-14:
Adding example panel provided by Andy Brown.
[HtmlInPanel.zip](https://github.com/JMRI/JMRI/files/3846694/HtmlInPanel.zip)

rhwood, 2019-11-14:
The web panels support labels with invalid HTML characters by escaping the contents of the label (always have since those characters “just work” in panel labels.

Basically any fix should be tested to correctly escape the labels “A -> B” and “C & D”.

Randall

mstevetodd, 2019-11-14:
Closing via #7620

closed 2019-11-15 by mstevetodd