JMRI/JMRI #7621: xsd does not enforce "specificappearance" aspect as one of aspectnames[]

devel-bobm, 2019-11-14:
According to the [Defining your own sigal system] (https://www.jmri.org/help/en/html/tools/signaling/DefineNewAspects.shtml) web page, the `specificappearances` section requires that any defined `aspect` value be a valid `aspectname` for the mast.  `appearancetable.xsd` seems unable to enforce this requirement.

This can be seen via passing `Debug->Validate XML` of "SOU-1981/appearance-CL-2-arm-permissive.xml".  That mast definition _does not_ define "Stop" as a valid `<aspectname>` but does reference it as the `<aspect>` for the `<danger>` and `<held>` specific appearances.  That XML _ought_ to fail based on the statement in the help file referenced above.  (And, for this specific set of cases, the `<aspect>` for `<danger>` and `<held>` _ought_ to be "Restricted Proceed".)

The same sort of problem appears in each of the "*permissive*" mast defintions for SOU-1981.  It is possible that other signal systems may have similar problems.

I am mostly "schema-illiterate".  Is this something tha XSD _can_ identify?  If so, the schema should be updated appropriately by someone with proper knowledge.  Then signaling systems _should_ be validated and updated as appropriate.

bobjacobsen, 2019-11-14:
Thanks for creating this Issue with a specific case!

XML Schema 1.1 can do this. Unfortunately, the parser JMRI is currently using only does 1.0 checks. 

I’ve wanted to do the update for a while as part of the unique-system-name work but was putting it off to the next release series. I’ll see if I can do this part first.

bobjacobsen, 2019-11-17:
Is there a reason for a signal definition to define an appearance twice?  E.g.
```
	<appearance>
      <aspectname>Restricted Proceed</aspectname>
      <show>flashred</show>
      <reference>Rule 241</reference>
      <imagelink>../../../resources/icons/smallschematics/aspects/BN-1989/SL-1-high/rule-241-b.gif</imagelink>
	</appearance>

	<appearance>
      <aspectname>Restricted Proceed</aspectname>
      <show>flashred</show>
      <reference>Rule 241</reference>
      <imagelink>../../../resources/icons/smallschematics/aspects/BN-1989/SL-1-high/rule-241-c.gif</imagelink>
	</appearance>
```
(from xml/signals/BN-1989/appearance-SL-1.xml)  It looks like that could be handled with a single definition that has two `<imagelink>` elements.  Is there some other reason to do this?

Duplicates make it harder to check the aspect mapping in the schema.

devel-bobm, 2019-11-17:
I _suspect_ that this occurence likely comes from the prototype rulebook.  And that JMRI's signal system definition help does not seem to give specific guidance for a boundary condition.  

I'll give an example I recently encountered.  Consider the following snippet of aspects for the "Restricting" indication, taken from the Consolidated Code of Operating Rules of 1967:

[image]

Note in the image that Fgures 2 and 8 are show differing aspects for otherwise identical masts for this indication.  Similar sets include , Figures 3&10&13, Figures 4&11&14, Figures 5&6, Figures 9&12.

How should this be implemented in a JMRI signal system?  Help page doesn't give appropriate guidance.  So I'll go to the prototype mast.  Historically, a prototype searchlight or colorlight head has a fixed set of spectacle/lens colors.  So a given mast install can ONLY generate one set of aspects, and its available set of aspects forces it to implement only one of the aspects from 240N.

**To my mind,** this means that to allow a layout to include some masts which display 240N-5, and some masts which display 240N-6, one needs to include two different two-head dwarf mast definitions in the signal system- one with "Restricting" defined as Red/Lunar, the other with "Restricting" defined as Lunar/Red.  Each of these masts would use the `<aspectname>` of "Restricting", regardless of which of the 240N "figures" is implemented. 

Or one could implement an additional `<aspectname>` name, such as "Restriciting Lunar Over Red".  And update aspects.xml to include that new aspect name variant.  And update all of the masts in the definition to include the new aspect name in the aspect mapping.  And, if a signal from some other system must protect that mast, that signal system would need updates to reflect the new `<aspectname>`.  That sounds like FAR too much work to me, and probably non-trivial user confusion.

Or one could simply ignore the alternate aspect figures, and implement only one mast displaying only one of the optional aspects (i.e. 240N Figure 4 but not 240N Figure 11 and not 240N Figure 14, above).  (I've done this for one signaling system for a layout I'm going to work with, but the layout owner has not yet weighed-in on this decision...)

If any of the approaches described above is taken, then schema can enforce a "no duplicate `<aspectname>` definitions" rule within a given mast definition.  And requiring a single mast to have unique, singly-defined `<aspectname>`s, with a single appearance, and a single `<imagelink>`

@bobjacobsen, do you have a feel for how many mast implementations have similar duplicate `<aspectname>` definitions?

KenC57, 2019-11-17:
I would agree with Bob M that a single file for a mast would only have one
aspect name to appearance. Multiple files would be the cleaner way for these
implementations. As the layout/signal designer, you would know which fig is
being installed on the layout. Now sensible naming conventions to make it
clear which one you want when, that might be the trickier case. But in that
display of rule 240N, I would expect that each of the fig come from a
different mast file that is a mix of head and arm configurations. And that
relates to any single physical mast on the prototype only used one of the 14
figs listed, which might be related to history of that mast as what hardware
was there. #3 and #9 are curious as some railroads tried to implement
standards where the offset of the head mattered. What I've heard is the
superintendent who started that idea never stayed long enough for it to
become implemented in all places so you ended up with the physical mix still
meaning the same thing.

devel-bobm, 2019-11-17:
Slightly off-topic:  @KenC57, I've discussed "staggered" ("offset") signal heads ("arms") with Mike Burgett.  He states that in the era before highly-reflective materials and before bright headlamps, number plates were ineffective, so offset arms were used specifically to identify "permissive" signals.  Vertically-aligned arms were used to identify "absolute" signals.  Staggered signals are still being installed in modern railroading.

bobjacobsen, 2019-11-17:
There are currently 74 errors when matching aspect names in the _same_ file (i.e. not matching from a specific `appearance*.xml` file to the common `aspect.xml` file)  See the attached list.

[SignalSchemaErrors.txt](https://github.com/JMRI/JMRI/files/3856019/SignalSchemaErrors.txt)

bobjacobsen, 2019-11-17:
I think it would be fine to handle a single mast type (i.e. one file) with duplicate definitions by splitting it into separate mast types (i.e. multiple files) each with one of those definitions in each.  

It would be great if somebody would edit the existing files to do that.  

The BN-1989 system has duplicate definitions of Restricted Proceed in:
 - xml/signals/BN-1989/appearance-SL-1.xml

and duplicate definitions of Clear in 
 - xml/signals/BN-1989/appearance-SL-2.xml
 - xml/signals/BN-1989/appearance-SL-3.xml

The SLSF-1973 system has duplicate definitions of Restricting in:
 - xml/signals/SLSF-1973/appearance-SL-3-triangle.xml
 - xml/signals/SLSF-1973/appearance-SL-2-triangle.xml
 - xml/signals/SLSF-1973/appearance-SL-2-pbs.xml
 - xml/signals/SLSF-1973/appearance-SL-2-grade.xml
 - xml/signals/SLSF-1973/appearance-SL-2-abs.xml
 - xml/signals/SLSF-1973/appearance-SL-3.xml

bobjacobsen, 2019-11-17:
You can find a schema that does additional checks (and looks a bit better online) in the bobjacobsen/JMRI repository as the signal-schema-update branch.  To play with this:

```
git fetch https://github.com/bobjacobsen/JMRI.git
git checkout bobjacobsen/signal-schema-update
ant clean tests
./runtest.csh  jmri/implementation/SignalSystemFileCheckTest
```

bobjacobsen, 2019-12-01:
I'm going to edit the files in the BN-1989 and SLSF-1973 definitions (see above) to move the 2nd definition of Restricted Proceed, Clear and Restricting to a -b file, or if necessary a -c or -d file.  This is consistent with how the icons are named.

bobjacobsen, 2019-12-30:
The duplicates have been resolved, so now we're back to the specificappearance issue.  The files with problems are:
```
xml/signals/B&O-Semaphore/appearance-bumper.xml line 129 value 'Restricting'
xml/signals/NW-1981/appearance-CPL-1-arm-permissive.xml line 146 value 'Not Lit'
xml/signals/SOU-1981/appearance-CL-2-arm-permissive.xml line 203 value 'Stop'
xml/signals/CROR-2008/appearance-CROR-2OL-Hi.xml line 371 value 'Stop Signal'
xml/signals/CROR-2008/appearance-CROR-2O-Hi.xml line 368 value 'Stop Signal'
xml/signals/CROR-2008/appearance-CROR-2OD-Hi.xml line 369 value 'Stop Signal'
xml/signals/B&O-2009/appearance-med1.xml line 228 value 'Permissive'
xml/signals/B&O-2009/appearance-fast.xml line 248 value 'Permissive'
xml/signals/B&O-2009/appearance-meds.xml line 244 value 'Permissive'
xml/signals/B&O-2009/appearance-full.xml line 309 value 'Permissive'
xml/signals/B&O-2009/appearance-junc.xml line 252 value 'Permissive'
xml/signals/B&O-2009/appearance-fasts.xml line 225 value 'Permissive'
xml/signals/B&O-2009/appearance-block.xml line 206 value 'Permissive'
xml/signals/B&O-2009/appearance-fast1.xml line 214 value 'Permissive'
xml/signals/B&O-2009/appearance-med.xml line 252 value 'Permissive'
xml/signals/B&O-2009/appearance-boths.xml line 277 value 'Permissive'
xml/signals/B&O-2009/appearance-fastm.xml line 237 value 'Permissive'
xml/signals/B&O-2009/appearance-slow.xml line 182 value 'Permissive'
xml/signals/B&O-2009/appearance-bothm.xml line 287 value 'Permissive'
xml/signals/B&O-2009/appearance-medm.xml line 241 value 'Permissive'
xml/signals/B&O-2009/appearance-junm.xml line 279 value 'Permissive'
xml/signals/WM-1980/appearance-USS-R2-1-arm-low-marker-permissive.xml line 140 value 'Not Lit'
xml/signals/WM-1980/appearance-USS-R2-1-arm-permissive.xml line 134 value 'Not Lit'
xml/signals/SLSF-1973/appearance-SL-3-c.xml line 194 value 'Stop'
xml/signals/BR-2003/appearance-3-i.xml line 134 value 'Off'
xml/signals/B&O-1980/appearance-CPL-High-permissive.xml line 212 value 'Stop'
```
Note that this is just the first missing value; there might be others in the files too.

closed 2021-04-14 by stale[bot]