JMRI/JMRI #7627: Handling unreliable CI tests

bobjacobsen, 2019-11-16:
I've restarted 34 CI test runs in the past week due to specific transient test failures.  See also #7594, #7510, #6910, #6912, and probably others.

I believe that passing CI tests is valuable, that just looking at the output and deciding it's an "expected failure" without restarting is dangerous, but the number of restarts now needed is a ridiculous use of time.  

Note that these intermittent tests essentially never fail in the Jenkins SeparateJobs task.  The problem seems to be running them together, and with other tests.

There are two goals:
 - We need to be able to run these transient-failure tests locally and in CI
 - We need a way to run them a _lot_ of times until they pass (at least once) cleanly

But we don't have the ability to get somebody to actually fix this tests to run in a joint-test CI environment, and we have to be realistic in that respect

So I propose:
 - Add an environment variable `jmri.skipKnownIntermittent` tests, default false, set true in CI
 - Create a new `scripts/run_intermittent_tests_separately` script that runs all the known-intermittent tests with a large number of retries.  
 - Create a Jenkins job that runs just the `run_intermittent_tests_separately ` script once a day
 - Add a fourth phase to the Travis job (but not the AppVeyor job) that will run `run_intermittent_tests_separately` script separately

 - When a test is known to be seriously intermittent, 
   - document that failure mechanism in the the test class and class under test Javadoc
   - have the test class honor `jmri.skipKnownIntermittent`
   - add the test class to the `run_intermittent_tests_separately` script

pabender, 2019-11-16:
I am not sure why #6910 and 6912 were tagged on this.  Those are both long closed PRs.

What I see is that #7510 and #7594 both refer to linkedWarsntTest failing.  I would not be opposed to having that test disabled until we can figure out why it is frequently failing.  It has been doing this for a while.  Also, if you really look at this test, it isn’t a unit test.  It is better classified as an integration test.

One other note, this can be an argument for moving to JUnit 5.  JUnit 5 has the ability to tag and filter tests.  ( see https://www.baeldung.com/junit-filtering-tests for an example ).  I would rather see us move to a standard method of filtering out failing tests rather than inventing something of our own.

bobjacobsen, 2019-11-16:
@pabender  Sorry, but I consider a move to JUnit5 off-topic.  The problem is _today_.  

It's not just one class, though LinkedWarrantTest is certainly one of the most common. Here are some of the restarts from this past week that I've seen more than once:
 - jmri.swing.NamedBeanComboBoxTest
 - jmri.jmrit.logix.NXFrameTest
 - jmri.jmrit.logix.LearnWarrantTest
 - jmri.jmrit.display.controlPanelEditor.CircuitBuilderTest
 - jmri.jmrit.display.layoutEditor.LayoutTrackEditorsTest

rhwood, 2019-11-17:
I think every test is now JUnit5 ready since the migration to JUnit4 is complete, so maybe the tests can be filtered *today* by migrating the test framework to JUnit5.

geowar1, 2019-11-17:
I just completely rewrote LayoutTrackEditorsTest in PR #7575 that was merged yesterday; were any of its restarts since then?

bobjacobsen, 2019-11-17:
Several occurrences today of 
```
1001[ERROR]   LayoutTrackEditorsTest.testEditDoubleSlipDone:403 » TestTimedOut test timed ou...
1002[ERROR]   LayoutTrackEditorsTest.testEditRHTurnoutDone:297 » TestTimedOut test timed out...
1003[ERROR]   LayoutTrackEditorsTest.testEditSingleSlipDone:461 » TestTimedOut test timed ou...
1004[ERROR]   LayoutTrackEditorsTest.testEditTrackSegmentDone:96 » TestTimedOut test timed o...
1005[ERROR]   LayoutTrackEditorsTest.testEditTurnoutDone:204->Object.wait:502->Object.wait:-2 » TestTimedOut
1006[ERROR]   LayoutTrackEditorsTest.testEditTurntableDone:693 » TestTimedOut test timed out...
1007[ERROR]   LayoutTrackEditorsTest.testEditXingDone:552 » TestTimedOut test timed out afte...
1008[ERROR]   MemoryIconTest>MemoryIconTest.testShowBlank:88->MemoryIconTest.getColor:282 » Jemmy
1009[ERROR]   MemoryIconTest>MemoryIconTest.testShowReportable:229 » TimeoutExpired Wait eve...
```

geowar1, 2019-11-17:
Thanks… I’ll take alook.

geowar1, 2019-11-17:
All of the LayoutTrackEditorsTest ones are timing out on a JComboBoxOperator.selectItem(idx) call… against a NamedBeanComboBox.

geowar1, 2019-11-17:
It’s possible that some of the travis runs that took too long were caused by accumulated timeouts that failed repeatedly… it may be possible to improve this by adjusting the timeout and retry rules for those tests. Those NamedBeanComboBox selectItem calls ether complete immediately or fail after 10 seconds… we might try adjusting that timeout down to 3 (or even 1) so that they’ll fail faster… and not so drastically effect the overall runtime of the tests.

bobjacobsen, 2019-11-17:
I changed the various names to focus on separately running the tests.

It's hard to diagnose what's going on, but `LayoutTrackEditorsTest` runs fine on my laptop and fails almost all the time in CI.  I _think_ some window is being left open too long along the way, causing a cascade of failures (extending even to MemoryIconTest right after)  Running that test class separately is being tried now in PR #7630 

Jenkins AllTest is still running the separate tests, and seems to (so far) be doing fine.  Not sure what the difference is w.r.t. Travis and AppVeyor.

bobjacobsen, 2019-11-18:
Closed by PR #7630

closed 2019-11-18 by bobjacobsen