JMRI/JMRI #7633: Does JMRI require access to a JSON schema server?

bobjacobsen, 2019-11-17:
Recently, several CI jobs have failed with messages pointing to a "Failed to load json schema!" problem  and/or problems with the format of the JSON schema. IIRC, this has happened at least once before.

Does JMRI require online access to a JSON schema source when running tests? In normal operation?

If so, we probably need a fall-back mechanism.

rhwood, 2019-11-17:
JMRI does not need access to an external server during normal operations (there is a preference that can be set to require that, but it is not enabled by default and has UI to enable it).

Tests *do* need access to an external site, but since #6721, have be ignored, not failed, if that site is not available.

rhwood, 2019-11-17:
Actually, I need to clarify, *some tests* have been ignored, not failed, on those failures since #6721, but not all.

rhwood, 2019-11-17:
What's also interesting is that if there was a general failure to get to a schema, there would be significantly more tests failing than just those three.

rhwood, 2019-11-17:
I've found the problem and am working on a PR for it.

rhwood, 2019-11-18:
The problem is twofold:
1. The json-schema.org site now redirects all http requests to https and Java's URLConnection class does not follow http->https redirects.
  networknt/json-schema-validator#216 addresses this, and I expect can be incorporated soon.
2. Our rests are, for some reason, ignoring the json-schema-validator configuration set in `jmri.server.json.schema.JsonSchemaServiceCache` that should cause requests for `http://json-schema.org/draft-04/schema#` to be changed to `resource:jmri/server/json/schema/json-schema.draftv4.json`. This is somewhat more worrisome, and what I am working to fix.

stevehu, 2019-11-18:
@rhwood I have released 1.0.27 with this issue resolved by @andersonf. Please try it out and see if it works. Thanks.

rhwood, 2019-12-02:
@stevehu Version 1.0.28 of json-schema-validation works. Thank you.

stevehu, 2019-12-02:
@rhwood Cool. Thanks for the confirmation.

PR #7691 (2019-12-03): fix: Update JSON Schema handling
Fixes #7633 

- Updates json-schema-validator to current version to get around HTTP->HTTPS redirection failures in Java
- Adds an additional URL mapping to the JSON schema URL mapping to use a local resource instead of a network resource
- Resumes testing of tests ignored to workaround 7633
- Perform more schema testing
- Fix an issue noted with `JsonReporterHttpService`
commit ffa2b0b1 (2019-12-03): test: Resume testing

Fixes #7633
closed 2019-12-03 by bobjacobsen