JMRI/JMRI #7649: Create Test Release 4.17.7

bobjacobsen, 2019-11-20:
This is the next release in the 4.17 cycle. It's intended to be created around November 24 from the `HEAD` of the `master` branch.

bobjacobsen, 2019-12-05:
The release-4.17.7 branch has been created. 

The master branch on GitHub is currently locked.  Once we are sure that 4.17.7 has the right contents, hopefully quite soon, we can unlock it and start merging PRs again toward the 4.19.1, the first test release of the next series.

Jenkins will be creating files shortly at the [CI server](http://builds.jmri.org/jenkins/job/TestReleases/job/4.17.7/)

Pugwash1, 2019-12-06:
Took it for a spin. Open Layout Panel. Save Panel, and it keeps defaulting to the panel title, not the file name I used to open it, or the file name last used to save it. This is very recent

geowar1, 2019-12-07:
Write up an incident and assign to me… this was working… only used panel title if no default/load file.

geowar1, 2019-12-09:
Bugfix:Save Panel uses panel title instead of load/last saved name PR #7705

bobjacobsen, 2019-12-11:
JMRI 4.17.7 has been released. Files are available in the GitHub release section.

closed 2019-12-11 by bobjacobsen