JMRI/JMRI #7652: Loconet: Function keys F9-F28 cleared when F0-F8 pressed

mstevetodd, 2019-11-21:
Had a user (Patrick) report strange Fkey behavior that I was able to duplicate on each of my hardware systems.
Connected to Loconet DCS240 using the "Loconet PR3" connection.
No roster entry.
Open JMRI throttle.
Acquire a loco with short address between 64 and 127.
Press a function key between F9 and F28. e.g. F15.
Function stays selected as expected.
Press a function key between F0 and F8. e.g. F1
F1 stays selected as expected.
F15 immediately changes to NOT selected. Not as expected.
This occurs for any combinations. Press F15, F16, F17, then F1, and F15, F16, F17 go unselected.
Watching the Loconet Monitor, I see only the expected commands, I do not see anything on the monitor to cause the issue.

Same behavior using a roster entry, if Fkey marked as "Lock".
Same behavior seen with EngineDriver (not surprising).
Problem does NOT occur using Digitrax Simulator.
Problem does NOT occur with short addresses below 64 or with long addresses. (!)

Any thoughts? Anyone want to chase this?

bobjacobsen, 2020-01-21:
The problem seems to be that the Slot is not getting updated with the function states for higher numbered functions when those messages are handled. With the Slot stuck at false for these, any update clears the throttle. 

Not clear what’s missing.  LocoNetSlot.functionMessage(long pkt) is supposed to do that.

devel-bobm, 2020-01-22:
This issue is an odd one.  I find it to be perplexing.

The original comment that changing F1 causes F15 to unexpectely change suggests a problem in decoding the packet type.  Perhaps an incorrect invocation of LocoNetSlot.functionMessage() when it should _not_ be invoked.  But I do not see (at this time) any code which would cause this to occur.

Another oddity is the statement that this happens specifically for short addresses between 64 and 127...  Why just when bit 6 of the short address is 1?  Why not long addresses?  Seems like some sort of a masking error if it really is a problem specific to an address range.  There isn't _any_ "short/long" address sensitivity for F0 thru F8, as this is handled solely thru slot numbering, not addressing.

I wonder if focusing on the jmrix.loconet code is a red-herring...

bobjacobsen, 2020-01-22:
My bet:
 - sending e.g. F15 sends an immediate packet but does not yet change the Slot
 - that message comes back to the Slot manager and has to be routed to the right Slot object
 - but for addresses 65-128, something goes wrong so that the wrong Slot is updated
 - later the throttle refreshes from the Slot (a valid thing to do), but gets the wrong contents

It would be really helpful to get the actual raw LocoNet messages to write a test that checks this, but I think it's likely in `SlotManager.getDirectFunctionAddress`, perhaps the mask with 0x40 in the second line below:

```
        // check long address
        if ((m.getElement(5) & 0x40) != 0) {
            addr = (m.getElement(5) & 0x3F) * 256 + (m.getElement(6) & 0xFF);
            if ((m.getElement(4) & 0x02) != 0) {
                addr += 128;  // and high bit
            }
        } else {
            addr = (m.getElement(5) & 0xFF);
            if ((m.getElement(4) & 0x01) != 0) {
                addr += 128;  // and high bit
            }
        }
```

devel-bobm, 2020-01-22:
I would think that address masking, if it were incorrect, would cause control of one loco's functions to update the wrong slot data information, not cause it to update the right slot but the wrong function numbers...

I don't see any significant changes in `LocoNetSlot.functionMessage()` since 2015, so I don't think that function is at fault.  And I do not see any coding issues in `LocoNetSlot` where _address_ information is conflated with _function_ information.

But I do remember that NMRA DCC packets have different lengths for short/long address operations.  Perhaps the code that creates the `long pkt` parameter used by `LocoNetSlot.functionMessage()` is at fault, specifically for the case where a NMRA function control packet for a shot loco address is encoded in the OPC_IMM_PACKET message.

And I wonder about using type `int` for the return value from `SlotManager.getDirectDccPacket(LocoNetMessage m)`.  I have not studied JAVA data rages, so I wonder whether an `int` is reliably able to hold up to 7 bytes of packet data, and how the `int` might be corrupted, in a JRE-specific way, if a longer-than-int packet is passed to `getDirectDccPacket()`.

**note that the return value from `SlotManager.getDirectDccPacket()` (of type `int`) is passed as an argument to `LocoNetSlot.functionMessage(long pkt)`.  Note the type mismatch** - `int` vs. `long`.  In some programming languanges, `int` and `long` are different types, and in some programming languages, `int` might be identical to `long` on some platforms but not on other platforms.  I do not know the JAVA-related details related to these types.

I agree that a specific LocoNet message stream (including raw Hex values!) will be useful in tracking this issue to ground.

It also seems to me that `LocoNetSlot.functionMessage()` does _not_ attempt to differentiate NMRA packets with long addresses from packets with short addresses.  Perhaps this is part of the problem?

bobjacobsen, 2020-01-22:
> I would think that address masking, if it were incorrect, would cause control of one loco's functions to update the wrong slot data information, 

Exactly my point.  And because it's updating the _wrong_ slot, when the LocoNetThrottle object retrieves the upper function values, they're still all set to "off" - some other slot got the right values instead.

> not cause it to update the right slot but the wrong function numbers...

I don't see any evidence that's happening.

devel-bobm, 2020-01-22:
The more I look at this, the more I think it is improper handling of the "raw packet" information (previously-extracted from the OPC_IMM_PACKET message and passed in via the method's parameter) in `LocoNetSlot.functionMessage()`.  That code seems to _ignore_ that the incoming packet might be to _either_ a long address _or_ a short address.  It seems to extract data from the `long` value from a single set of positions, regardless of whether one or two bytes of address are present.

Perhaps when an OPC_IMM_PACKET to a "short" address between 64 and 128 is seen, it is passed to `LocoNetSlot.functionMessage()` and interpreted _assuming_ that there are two bytes of address information rather than one, and so the "command" bits are checked from the wrong byte, and thus the wrong interpretation is had.

(We really need a LocoNet message log where the problem occurs; based on the suppositions above, I strongly suspect that the initial statement of problem is not quite an accurate statement of the problem.)

Regards,
Billybob

mstevetodd, 2020-01-22:
>  I strongly suspect that the initial statement of problem is not quite an accurate statement of the problem.

@devel-bobm What part do you feel is inaccurate? I can reverify any of this tonight. I can also provide a loconet log.
Are you not able to reproduce this on your hardware?

Pugwash1, 2020-01-23:
It's a dcs240. If the throttle is chatting expanded slots either by accident or deliberately this is the expected result.

mstevetodd, 2020-01-23:
The original reporter saw the symptoms on a DCS51 and a DCS240.

devel-bobm, 2020-01-23:
@mstevetodd,

All of my suppositions are based on code review, and I _cannot_ justify any problem in LocoNet code where accessing a function in the range F0 thru F8 would have _any_ effect on any function(s) in the range of F13 thru F28.  This is because:

- the LocoNet messaging for F0 thru F4 uses one LocoNet message and specifies a slot number not a loco address

- the LocoNet messaging for F5 thru F8 uses another Loconet message, and again, it specifies a slot number not a loco address.

- The LocoNet messaging for F9 thru F28 is based on a third type of LocoNet message, and this message format specifies the loco address (and includes a difference between long and short loco address).

So if actuating F1 causes problems with F15, I cannot justify it from LocoNet messaging and/or JMRI's LocoNet-specific coding.  If actuating F1 causes problems with F15, I _suspect_ the problem would be outside of JMRI's LocoNet functionality, and would affect systems other than LocoNet-based ones.

devel-bobm, 2020-01-23:
@Pugwash1, the fact that it is a DCS240 ought to be irrelevant.  JMRI does not understand expanded slots, but it _does_ understand OPC_IMM_PACKET, and has had coding for that message.  (There is some question in my mind whether that message is being properly handledin JMRI, though.)

The symptom originally stated is that changing f1 causes an unintended change to F15.  Given that, under the "standard slot" protocol, F15 is _only_ controlled by OPC_IMM_PACKET, and F1 is (typically) not controlled by OPC_IMM_PACKET, the problem does not seem likely to be related to the DCS240 (or even the LocoNet code).

devel-bobm, 2020-01-23:
I am very curious whether someone has _changed_ the way JMRI behaves for F0 thru F8 (or made some sort of a JMRI configuration change), such that JMRI is creating OPC_IMM_PACKET messages instead of OPC_DIRF for F0 thru F4 and OPC_SND for F5 thru F8.  

If such a code change (or JMRI configuration change) has been made, it might justify the reported problem, based on my review of the LocoNet code.

bobjacobsen, 2020-01-23:
> So if actuating F1 causes problems with F15, I cannot justify it from LocoNet messaging and/or JMRI's LocoNet-specific coding. If actuating F1 causes problems with F15, I _suspect_ the problem would be outside of JMRI's LocoNet functionality, and would affect systems other than LocoNet-based ones.

Except it’s also seems clear that the F15 message _is_ going back into the JMRI code, and it _is_ necessary for that to be properly handled so that the new on state of F15 is remembered.  If that is not correct, for whatever reason, then later that F15 will be found to still be in the off state when the values are read from the slot.  That provides a LocoNet-specific mechanism..

One test that might add some information is to try setting F15 on, then setting F17 to see if F15 changes, then also try F22.

devel-bobm, 2020-01-23:
Is there any evidence that this problem does or does not occur for non-LocoNet JMRI connections?

Pugwash1, 2020-01-23:
This is truley bonkers. The F15 gets unset on speed change, direction change, slot read...
remember for regular old slots, only f0-f8 is tracked. Throttles need to track the status internally. for an address of 48 it does decode imm packets correctly. for f15 the loconetslot never gets the function message, so its higher up the food chain.

Pugwash1, 2020-01-23:
getDirectDccPacket seems to be the problem. It returns x04 instead of x0de04
for this packet  [ED 0B 7F 33 02 42 5E 04 00 00 4F]  Send packet immediate: Locomotive 66 set F13=Off, F14=Off, F15=On, F16=Off, F17=Off, F18=Off, F19=Off, F20=Off.

Pugwash1, 2020-01-23:
And getDirectFunctionAddress also seems to have trouble

bobjacobsen, 2020-01-23:
(sorry to be slow, traveling)

> Is there any evidence that this problem does or does not occur for non-LocoNet JMRI connections?

I think @SteveG is hot on the trail of this, so perhaps this is redundant, but my basis for thinking its not in the GUI code or AbstractThrottle is from the original report which says that a simulator connection doesn’t show the problem.  I confirmed that with LocoNet and NCE simulated connections.

Since those use the same GUI and core code, but have different message-processing code than real LocoNet connections, I thought it pointed toward the latter.

Pugwash1, 2020-01-23:
It definately loconet code. Working on it now. When looking at the imm packet its only checking bit 6 not bit 7 and 6 for the long stuff.. Of course loconet stuffs those bits in different bytes just to make my hangover worse!

mstevetodd, 2020-01-26:
Closed by #7994

closed 2020-01-26 by mstevetodd