JMRI/JMRI #7660: Panel files that work in JMRI 4.16 and broken in 4.17.6

bobjacobsen, 2019-11-24:
After JMRI 4.17.6 was released, a [user posted in jmriusers](https://groups.io/g/jmriusers/message/166182) that 

> So many errors showed up with panels that worked under 4.16, that I gave up and went back to 4.16 Most of these errors were due to signal mast heads not being found and invalid signal head numbering.

He kindly provided the files, which are attached.  The first presenting problem is that the panel file defines and uses internal turnouts with system names of IT5 and IT005 which are intended to be (and _always_ _have_ _been_) separate, but JMRI 4.16.7 thinks should be the same.

```
    [java] 2019-11-22 07:12:22,358 managers.AbstractManager              ERROR - systemName is already registered. Current system name: IT005. New system name: IT5 [AWT-EventQueue-0]
    [java] 2019-11-22 07:12:22,360 configurexml.ErrorHandler             ERROR - Unexpected error (Exception) while load(/Users/jake/Library/Containers/com.apple.mail/Data/Library/Mail%20Downloads/00B6EBF9-B4DF-47CF-BD53-BA128E515B85/Bondale%2020191101%200903%20L2.xml) in adaptor of type jmri.jmrix.internal.configurexml.InternalTurnoutManagerXml Exception: jmri.NamedBean$DuplicateSystemNameException: systemName is already registered. Current system name: IT005. New system name: IT5
    [java] See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
    [java] jmri.NamedBean$DuplicateSystemNameException: systemName is already registered. Current system name: IT005. New system name: IT5
    [java] 	at jmri.managers.AbstractManager.register(AbstractManager.java:235)
    [java] 	at jmri.managers.AbstractTurnoutManager.newTurnout(AbstractTurnoutManager.java:124)
    [java] 	at jmri.managers.ProxyTurnoutManager.makeBean(ProxyTurnoutManager.java:50)
    [java] 	at jmri.managers.ProxyTurnoutManager.makeBean(ProxyTurnoutManager.java:18)
    [java] 	at jmri.managers.AbstractProxyManager.newNamedBean(AbstractProxyManager.java:320)
    [java] 	at jmri.managers.ProxyTurnoutManager.newTurnout(ProxyTurnoutManager.java:114)
    [java] 	at jmri.managers.configurexml.AbstractTurnoutManagerConfigXML.loadTurnouts(AbstractTurnoutManagerConfigXML.java:223)
    [java] 	at jmri.jmrix.internal.configurexml.InternalTurnoutManagerXml.load(InternalTurnoutManagerXml.java:35)
    [java] 	at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:658)
```
…
```
    [java] 2019-11-22 07:12:36,208 managers.AbstractManager              ERROR - systemName is already registered. Current system name: IT005. New system name: IT5 [AWT-EventQueue-0]
    [java] 2019-11-22 07:12:36,209 nfigurexml.SingleTurnoutSignalHeadXml WARN  - Failed to provide Turnout "IT5" in loadTurnout [AWT-EventQueue-0]
    [java] 2019-11-22 07:12:36,210 configurexml.ErrorHandler             ERROR - Unexpected error (Exception) while load(/Users/jake/Library/Containers/com.apple.mail/Data/Library/Mail%20Downloads/00B6EBF9-B4DF-47CF-BD53-BA128E515B85/Bondale%2020191101%200903%20L2.xml) in adaptor of type jmri.managers.configurexml.AbstractSignalHeadManagerXml Exception: java.lang.NullPointerException
    [java] See http://jmri.org/help/en/package/jmri/configurexml/ErrorHandler.shtml for possibly more information. [AWT-EventQueue-0]
    [java] java.lang.NullPointerException
    [java] 	at jmri.implementation.SingleTurnoutSignalHead.initialize(SingleTurnoutSignalHead.java:73)
    [java] 	at jmri.implementation.SingleTurnoutSignalHead.<init>(SingleTurnoutSignalHead.java:42)
    [java] 	at jmri.implementation.configurexml.SingleTurnoutSignalHeadXml.load(SingleTurnoutSignalHeadXml.java:108)
    [java] 	at jmri.managers.configurexml.AbstractNamedBeanManagerConfigXML.loadInAdapter(AbstractNamedBeanManagerConfigXML.java:439)
    [java] 	at jmri.managers.configurexml.AbstractSignalHeadManagerXml.loadSignalHeads(AbstractSignalHeadManagerXml.java:121)
    [java] 	at jmri.managers.configurexml.AbstractSignalHeadManagerXml.load(AbstractSignalHeadManagerXml.java:98)
    [java] 	at jmri.configurexml.ConfigXmlManager.loadOnSwingThread(ConfigXmlManager.java:658)
    [java] 	at jmri.configurexml.ConfigXmlManager.lambda$load$1(ConfigXmlManager.java:580)
    [java] 	at jmri.util.ThreadingUtil.runOnGUIwithReturn(ThreadingUtil.java:148)
    [java] 	at jmri.configurexml.ConfigXmlManager.load(ConfigXmlManager.java:578)
```

At a minimum, these files should go through the load-store-and-compare tests without significant change. Once that's done, it would be good to check back with that user to make sure that his layout operates properly.

[CR-test-files.zip](https://github.com/JMRI/JMRI/files/3883904/CR-test-files.zip)

bobjacobsen, 2019-11-24:
What might be a report of a related problem, from a different user.  Will attach files if I get them.

>  I named my C/MRI turnouts like CT0009 for the first node.  With 4.17.6, these get converted to CT9 in the table.  But the routeOutputTurnout for my routes still refer to CT0009

(I'm not sure what the presenting problem is here, trying to get details)

rhwood, 2019-11-24:
Do we have a sense that 4.17.5 did not have these problems?

danielb987, 2019-11-24:
The problem is probably due to #7278.
`AbstractManager` stores the beans in the variable `_beans` which is a `TreeSet` with a `NamedBeanComparator` as comparator and in the map `_tsys` which uses the system name as the key.

The solution, as discussed in the jmri-developers list, is to let `AbstractNamedBean.compareSystemNameSuffix()` use `String.compareTo()` instead of `AlphanumComparator`. And then change the system specific beans, like LtTurnout, to use `AlphanumComparator`.

PR #7666 (2019-12-01): fix: Systems must provide the comparator for system names
This fixes #7660 by requiring that the SystemConnectionMemo for each connection explicitly define the Comparator that will be used for system names in Managers. This way no Manager will possibly inherit an incorrect-for-its-system Comparator.

Marked as WIP since I am submitting this design for solving #7660 prior to updating all SystemConnectionMemos and Managers to implement it.
commit 3432f624 (2019-12-01): fix: Systems must provide the comparator for system names

This fixes #7660 by requiring that the SystemConnectionMemo for each connection explicitly define the Comparator that will be used for system names.
closed 2019-12-01 by bobjacobsen