JMRI/JMRI #7661: createNewSensor to throw IAE, not return null

silverailscolo, 2019-11-24:
The ``createNewSensor`` abstract method is annotated as ``@Nonnull`` and the javadoc states ``@return never null``. Since PR #1408 this can be improved further, using name validation before calling this method.
Many jmrix implementations still return ``null`` if eg. an invalid name was provided. That makes Spotbugs flag a warning for overriding the base annotation. (Same applies to createNewTurnout, createNewMemory etc.)

@danielb987 added: If an invalid name is provided, it's better to throw an ``IllegalArgumentException`` than returning ``null``.

danielb987, 2021-07-03:
I removed the Pending closure label since @icklesteve is working on this.

danielb987, 2021-08-17:
This issue is work in progress by @icklesteve.

closed 2022-01-14 by stale[bot]