JMRI/JMRI #7673: Fatal Crash Script Output Frame

icklesteve, 2019-11-27:
Not system dependant, reproducible with CBUS Sim + Loconet Sim

Launch PanelPro, no panel reqd.
Debug Menu > Display Message Log
Panels > Script Output - causes crash.

Reported via https://groups.io/g/jmriusers/message/166374

```
2019-11-27 10:45:00,221 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2019-11-27 10:45:00,274 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Steve\JMRI\log\messages.log [main]
2019-11-27 10:45:00,277 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Steve\JMRI\log\session.log [main]
2019-11-27 10:45:00,294 apps.Apps                             INFO  - PanelPro version 4.17.7ish+Steve+20191127T1044Z+R27e0b249d3 starts under Java 1.8.0_191 on Windows 7 amd64 v6.1 at Wed Nov 27 10:45:00 GMT 2019 [main]
2019-11-27 10:45:05,057 apps.Apps                             INFO  - Starting with profile Sim.3f3148fc [main]
2019-11-27 10:45:05,463 node.NodeIdentity                     INFO  - Using jmri-cD1mPaFlWhOiaaFbJZmpKW-3f3148fc as the JMRI storage identity for profile id 3f3148fc [AWT-EventQueue-0]
2019-11-27 10:45:06,582 configurexml.ConnectionConfigXml      INFO  - CAN Simulator Started [main]
2019-11-27 10:45:07,240 simulator.CbusSimulator               INFO  - Starting CBUS Network Simulation Tools [AWT-EventQueue-0]
2019-11-27 10:45:07,298 simulator.CbusDummyCS                 INFO  - Simulated Command Station: Standard Command Station [AWT-EventQueue-0]
2019-11-27 10:45:07,498 simulator.CbusDummyNode               INFO  - Simulated CBUS Node: SLIM [AWT-EventQueue-0]
2019-11-27 10:45:07,509 simulator.CbusEventResponder          INFO  - Simulated Event Responses: Random On / Off [AWT-EventQueue-0]
2019-11-27 10:45:07,682 eventtable.CbusEventTableDataModel    INFO  - Starting MERG CBUS Event Table [main]
2019-11-27 10:45:13,328 util.FileUtilSupport                  INFO  - File path program: is C:\Users\Steve\Documents\GitHub\JMRI\ [main]
2019-11-27 10:45:13,332 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Steve\JMRI\Sim.jmri\ [main]
2019-11-27 10:45:13,335 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Steve\JMRI\Sim.jmri\ [main]
2019-11-27 10:45:13,344 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Steve\JMRI\ [main]
2019-11-27 10:45:13,344 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Steve\ [main]
2019-11-27 10:45:13,345 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Steve\Documents\GitHub\JMRI\jython\ [main]
2019-11-27 10:47:46,919 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0_191 [AWT-EventQueue-0]
2019-11-27 10:47:46,962 jmri.InstanceManager                  ERROR - Exception creating auto-default object for jmri.script.JmriScriptEngineManager [AWT-EventQueue-0]
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at jmri.InstanceManager.getInstance(InstanceManager.java:289)
	at jmri.InstanceManager.getNullableDefault(InstanceManager.java:233)
	at jmri.InstanceManager.getDefault(InstanceManager.java:204)
	at jmri.script.JmriScriptEngineManager.getDefault(JmriScriptEngineManager.java:172)
	at jmri.script.ScriptOutput.getOutputArea(ScriptOutput.java:45)
	at jmri.jmrit.jython.JythonWindow.actionPerformed(JythonWindow.java:66)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:842)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:886)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
[... 79 lines cut ...]
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
2019-11-27 10:47:47,640 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.Error: Cannot call invokeAndWait from the event dispatcher thread
	at java.awt.EventQueue.invokeAndWait(EventQueue.java:1331)
	at java.awt.EventQueue.invokeAndWait(EventQueue.java:1324)
	at javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1353)
	at jmri.util.JTextPaneAppender.append(JTextPaneAppender.java:129)
	at org.apache.log4j.AppenderSkeleton.doAppend(AppenderSkeleton.java:251)
	at org.apache.log4j.helpers.AppenderAttachableImpl.appendLoopOnAppenders(AppenderAttachableImpl.java:66)
	at org.apache.log4j.Category.callAppenders(Category.java:206)
	at org.apache.log4j.Category.forcedLog(Category.java:391)
	at org.apache.log4j.Category.log(Category.java:856)
	at org.slf4j.impl.Log4jLoggerAdapter.error(Log4jLoggerAdapter.java:576)
	at jmri.util.exceptionhandler.UncaughtExceptionHandler.uncaughtException(UncaughtExceptionHandler.java:34)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1057)
	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1052)
	at java.awt.EventDispatchThread.processException(EventDispatchThread.java:227)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:219)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

mattharris, 2019-11-27:
Not sure what `Debug > Message Log` gives that's there in `Help > System Console` but this looks to be a threading issue.

I've got a fix that seems to resolve the repro scenario - PR to come shortly.

mattharris, 2019-11-27:
@icklesteve see #7674 for a possible solution.

icklesteve, 2019-11-27:
Thanks for such a quick fix @mattharris 
Have tested #7674 and does the trick :-)

Steve

icklesteve, 2019-11-28:
O/p also confirms fix has worked via https://groups.io/g/jmriusers/message/166479

closed 2019-11-28 by icklesteve