JMRI/JMRI #7681: Issues with Panel Pro and LoconetBuffer-USB - LocoNet Freezes

dacorreia, 2019-11-29:
We have come across a strange thing that we cannot explain. I do not know if it is related to other Panel Pro issues I have seen. I have a Panel Pro file that drives some of the signals on our layout. I update this file because of new crossovers and signals that were added on the layout. We also have updated the computer running Windows 10 and it has JMRI 4.17 installed on it.
When I load the new panel version in Panel Pro, we get a LocoNet error on the LocoBuffer-USB (the LocoNet LED on the LocoBuffer-USD turns orange) and no LocoNet Traffic is seen on the computer. We have the LocoBuffer-USD plugged into a USB 2.0 port running at 9600 baud.

I was able to duplicate this issue using another computer with the same version of JRMI and using a different LocoBuffer-USB.

 I also had the same issue occur on my laptop (same Version of JMRI) and another LocoBuffer-USB using my Digitrax Zephyr 50 (!) not even connected to the railroad’s signaling LocoNet cable. In this case the panel started but the LocoNet Error occurred when I ran the “SetUnknownSensorsToInactive” script when connected to the laptop. 

We do not have this issue using an older LocoBuffer Version 2.07 or a Digitrax PR3. The signaling Panel started and responded to inputs from the railroad. 

Have we discovered some issue with the LocoBuffer-USD using the current version of JMRI? Or have we discovered a limit to the size of a panel or the number of elements in the panel that the LocoNetBuffer-USD can handle? Why did the PR3 or the older LocoBuffer-USB allow the panel to load and run correctly? PLease let me know how to send the Panel Pro file to the group.

Thank you for helping us with this issue.

bobjacobsen, 2019-11-29:
What do you have the flow-control set to in the JMRI preferences? It’s in the advanced preferences under the Connection definition

dacorreia, 2019-11-29:
That I do not know, but I send a email to club members who will be there tonight. I it strange that older LocoNetBuffer-USB (version 0) is having the issue but not LooNetBuffer-USB version 2.0.7.0.

David Correia

Northern Virginia Model Railroads

devel-bobm, 2019-11-29:
This issue raises a number of questions in my mind:

1. What LocoNet error was reported?  We need to know the exact wording!

2. 4.17 is not a released version of JMRI.  Either it is 4.16 or it is 4.17.someNumber.  It is important to know exactly which version you are using, as some recent JMRI versions have problems configuring the connection to the LocoBuffer devices (as well as a bunch of other computer interface devices, not just LocoNet devices).  Have you tried 4.17.5?

3. Where are you setting a baud rate to 9600 for the LocoBuffer-USB?  In Device Manager?  JMRI should automatically set the baud rate at 57600 when it opens a LocoNet connection via a LocoBuffer-USB.

4. Have you configured JMRI for a connection via "LocoBuffer" or for a connection via "LocoBuffer-USB"?  The difference IS important in JMRI and the configuration profile should specify the type of LocoBuffer device which is actually being used.

5. Exactly which LocoBuffer device are you working with?  I am not familiar with "LocoBuffer-USD".  And I am not familiar with a LocoBuffer-USB device having an orange LED.  Look at the label on the LocoBuffer device and report it here.  It should note a "revision letter", something like "LocoBuffer-USB rev. F".

6. Do you have the correct drivers installed?  Are you using any other USB devices which operate as a COM port on the machines?  It is not uncommon for one device's drivers to "supercede" the drivers for another device and cause that other device to become inoperable.

devel-bobm, 2019-11-29:
@dacorreia, so far as I know, the "LocoBuffer-USB Version 'O'" (not version '0'!) is the very latest LocoBuffer version available, not an older model.

dacorreia, 2019-11-29:
Bob, you are correct this is a LocoBuffer-USB, I mistyped. The LocoBuffer-USB rev O. As the error, we do not see anything in JMRI that states that a error has occurred, other than the Loconet activity LED on the LocoNetBuffer turns a solid orange and no LocoNet traffic is observed on the computer.

I am going to pass your questions along to our DCC chairman who might be at the club tonight and get some responses. We are currently using a older LocoBuffer-USD Version 2.0.7.0 without any issues. Thank you for your quick reply.

Sincerely,

David Correia

Northern Virginia Model Railroaders

devel-bobm, 2019-11-29:
@dacorreia, note that the LocoBuffer-USB Rev "N" and LocoBuffer-USB Rev "O" draw more current from the "RailSync" wires (in the LocoNet cable) than previous LocoBuffer-USB models.  And the DCS240 provides less current on the "RailSync" wires than previous Digitrax command stations (see below).

This can cause any LocoNet devices which depend on RailSync to fail, including those throttles which are pulling power from the RailSync wires.  While I have never seen a LocoNet failure where LocoNet data is totally blocked, it is a possible failure scenaro.  

Suggested debugging technique: If simply unplugging the LocoBuffer-USB Rev "O" from LocoNet allows other LocoNet devices to use LocoNet "normally", then you may well have overloaded the ability of the DCS240 to power "RailSync".  And if this is the case, be ABSOLUTELY SURE that any LocoNet device which can run on batteries actually has a fully-charged battery installed.  And, try unplugging any batteryless throttles from LocoNet.

Another recommendation - check your LocoNet cabling using the LT1 (and the proper LT1 test methodolgy as described in the Digitrax documentation!) at all locations between the LocoBuffer-USB and the command station.  A simple broken RailSync conneciton, or even a broken "LocoNet Ground" connection can cause excessive current draw and prevent the DCS240 from passing appropriate power to LocoNet devices.

If I recall correctly, in my testing a year ago, the DCS240 plus a LocoBuffer Rev "O" could not provide enough current to power up a DT402D.  If I unplugged the LocoBuffer-USB Rev "O", the DT402D would power up.  If I tried the DCS100 with a LocoBuffer Rev "O" and the DT402D, it did power up.  This test is the basis of my statement that the DCS240 provides less RailSync power than a DCS100.

dacorreia, 2019-11-29:
Bob, okay will we do the steps that you outline. Maybe the test with the Zephyr was not the best method. Our command station is a  DCS 100s, which we keep up to date if that helps. I am also keeping our DCC Chairman in the loop.

David Correia

Northern Virginia Model Railroaders

danielb987, 2019-11-29:
If there is not enough power on the Loconet railsync to power the LocoNet devices, there are severall possibilities to add power to railsync. Either using a separate 12 volt supply, if the booster signal is not needed on the LocoNet, or to use a RailSync Current Limiting Device, RSCLD.
http://mollehem.se/index.php/en/control-and-electronics/rscld-detail

devito0824, 2019-12-04:
Our club has the same issue when using a Locobuffer-USB Rev-O. I worked with Dick and we were never able to find the cause. For us, only the locobuffer would lock up and you couldn't use JMRI to talk to the layout anymore or view loconet traffic. The layout by itself continued to function, as in you could run trains and throttles would work. I went down as far as only having the computer, locobuffer, and command station (DCS 240) connected and still had the same issue. We were using a PR3 before and did not have this issue. We have since gone back to the PR3 as we could not find a solution. The PR3 with JMRI and our layout does not have the same problem. We have turnouts, block detection, and signaling.

dacorreia, 2019-12-05:
Sounds like the same issue that we are having. I wonder if there is another bug in JMRI that was discovered with version 4.16. Dick has a notice on the RR-CirKits website about that bug and to upgrade to 4.17. We did that are still having the issue that you described above and that I also posted on this board. We are using a older LocoBuffer-USB Version 2.0.7.0 and are not having any issues. We also tried using the PR3 and did not have any issues like your club.

David Correia

Northern Virginia Model Railroaders

closed 2021-04-14 by stale[bot]