JMRI/JMRI #7713: Z21 - 1 second timeout sending turnout commands

Johan-Dahlberg, 2019-12-10:
Hello,

I am experiencing a problem very similar to issue #6400.
This time I get long delays when trying to control turnouts.
The log messages below are recorded when activating a two-turnout route.

I am running JMRI version 4.17.6 on an Ubuntu machine.

Console log:
```
2019-12-10 21:31:49,173 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: 08 00 A2 00 B0 66 20 09 consecutive timeouts = 0 in z21.Z21TrafficController [z21.Z21TrafficController Transmit thread]
2019-12-10 21:31:51,215 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: 08 00 A2 00 B0 67 20 08 consecutive timeouts = 0 in z21.Z21TrafficController [z21.Z21TrafficController Transmit thread]

```

Z21 traffic window:
```
21:31:47.157:  LocoNet Tunnel Message: Requesting Switch at LT103 (Gba1) to Closed (Output On).

21:31:47.157:  XpressNet Tunnel Reply: Feedback Response: 0 102
21:31:47.173:  LocoNet Tunnel Message: Requesting Switch at LT103 (Gba1) to Closed (Output Off).

21:31:49.212:  LocoNet Tunnel Message: Requesting Switch at LT104 (Gba2) to Closed (Output On).

21:31:49.214:  XpressNet Tunnel Reply: Feedback Response: 0 103
21:31:49.214:  LocoNet Tunnel Message: Requesting Switch at LT104 (Gba2) to Closed (Output Off).
```

I hope you can help me out.

Best regards,
/Johan

pabender, 2019-12-14:
@Johan-Dahlberg 

Your log is interesting for several reasons.

First, you are sending LocoNet turnout requests, but receiving XpressNet responses.  
Second, the XpressNet response isn’t actually valid, at least not from a standard XpressNet   point of view.

I will look into the response to see if Roco specified thier own.  I don’t recall seeing that they did.

What happens if you use an XPressNet turnout Instead of a LocoNetone?

Johan-Dahlberg, 2019-12-16:
Thanks for your response, Paul!

I tried controlling a single switch and two switches in a route, both using XpressNet and LocoNet.
This was done in a completely fresh profile, to avoid non-default settings creating strange effects.

1. XpressNet single switch using address 5 set to closed:
Z21 monitor
```20:44:06.857:  XpressNet Tunnel Message: 53 00 04 88 DF
20:44:06.921:  XpressNet Tunnel Reply: Feedback Response: Turnout without Feedback Turnout: -1 State: Not Operated; Turnout: 0 State: Thrown Left
20:44:07.174:  XpressNet Tunnel Message: 53 00 04 80 D7
20:44:07.184:  LocoNet Tunnel Tx: Requesting Switch at LT5 to Thrown (Output Off).
```
Message log
```
20:44:12.174 (124696) WARN  [z21.Z21XNetPacketizer Transmit thread] AbstractMRTrafficController.java:579  - Timeout on reply to message: 53 00 04 80 D7 consecutive timeouts = 0 in z21.Z21XNetPacketizer
```

2. LocoNet single switch using address 1 set to closed:
Z21 monitor
```
20:50:26.235:  LocoNet Tunnel Message: Requesting Switch at LT1 (LN1) to Closed (Output On).

20:50:26.293:  XpressNet Tunnel Reply: Feedback Response: Turnout without Feedback Turnout: -1 State: Not Operated; Turnout: 0 State: Not Operated
20:50:26.326:  LocoNet Tunnel Message: Requesting Switch at LT1 (LN1) to Closed (Output Off).
```
Message log
```
20:50:28.326 (500848) WARN  [z21.Z21TrafficController Transmit thread] AbstractMRTrafficController.java:579  - Timeout on reply to message: 08 00 A2 00 B0 00 20 6F consecutive timeouts = 0 in z21.Z21TrafficController
```

3. XpressNet switch addresses 5+6 set to thrown:
Z21 monitor
```
20:53:38.540:  XpressNet Tunnel Message: 53 00 04 89 DE
20:53:38.541:  XpressNet Tunnel Reply: Feedback Response: Turnout without Feedback Turnout: -1 State: Not Operated; Turnout: 0 State: Thrown Left
20:53:39.129:  XpressNet Tunnel Message: 53 00 05 89 DF
20:53:39.131:  XpressNet Tunnel Reply: Feedback Response: Turnout without Feedback Turnout: -1 State: Thrown Left; Turnout: 0 State: Thrown Left
20:53:40.230:  XpressNet Tunnel Message: 53 00 04 81 D6
20:53:40.232:  LocoNet Tunnel Tx: Requesting Switch at LT5 to Closed (Output Off).

20:53:45.333:  XpressNet Tunnel Message: 53 00 05 81 D7
20:53:45.334:  LocoNet Tunnel Tx: Requesting Switch at LT6 to Closed (Output Off).
```
Message log
```
20:53:45.230 (697752) WARN  [z21.Z21XNetPacketizer Transmit thread] AbstractMRTrafficController.java:579  - Timeout on reply to message: 53 00 04 81 D6 consecutive timeouts = 0 in z21.Z21XNetPacketizer
20:53:50.332 (702854) WARN  [z21.Z21XNetPacketizer Transmit thread] AbstractMRTrafficController.java:579  - Timeout on reply to message: 53 00 05 81 D7 consecutive timeouts = 1 in z21.Z21XNetPacketizer
```

4. LocoNet switch addresses 1+2 set to thrown:
Z21 monitor
```
20:56:11.639:  LocoNet Tunnel Message: Requesting Switch at LT1 (LN1) to Thrown (Output On).

20:56:11.639:  XpressNet Tunnel Reply: Feedback Response: Turnout without Feedback Turnout: -1 State: Not Operated; Turnout: 0 State: Not Operated
20:56:11.673:  LocoNet Tunnel Message: Requesting Switch at LT1 (LN1) to Thrown (Output Off).

20:56:13.678:  LocoNet Tunnel Message: Requesting Switch at LT2 (LN2) to Thrown (Output On).

20:56:13.679:  XpressNet Tunnel Reply: Feedback Response: Turnout without Feedback Turnout: -1 State: Thrown Left; Turnout: 0 State: Not Operated
20:56:13.679:  LocoNet Tunnel Message: Requesting Switch at LT2 (LN2) to Thrown (Output Off).
```
Message log
```
20:56:13.673 (846195) WARN  [z21.Z21TrafficController Transmit thread] AbstractMRTrafficController.java:579  - Timeout on reply to message: 08 00 A2 00 B0 00 00 4F consecutive timeouts = 0 in z21.Z21TrafficController
20:56:15.679 (848201) WARN  [z21.Z21TrafficController Transmit thread] AbstractMRTrafficController.java:579  - Timeout on reply to message: 08 00 A2 00 B0 01 00 4E consecutive timeouts = 0 in z21.Z21TrafficController
```

Do you need a dump of the ethernet traffic?

Regards,
/Johan

pabender, 2019-12-17:
No need for a LocoNet trace.  I see the problem.

This is yet another case of Roco deciding the existing XpressNet protocol was not sufficient and creating their own reply.

I will fix that in the next day or two

Johan-Dahlberg, 2020-01-31:
Hello Paul,

I guess you either became very busy, the problem was more complicated than
you first expected, or both... ;-)

Not that I want to make you feel stressed about this issue,
but if you have the time I would appreciate your help in solving it.

Regards,
/Johan

pabender, 2020-02-04:
@Johan-Dahlberg this is still on my todo list, but other things have gotten in the way (not all JMRI related).  I''ve actually had this issue open on my computer for the last week, hoping I would have the hour or so it will take to fix it, but that hasn't happened yet.

pabender, 2020-03-02:
PR #8199 should fix this, or at least reduce any delays to the response time of the Z21.

pabender, 2020-03-16:
@Johan-Dahlberg try the next daily build from https://builds.jmri.org/jenkins/job/development/job/packages/

Any build after #98 will have the code from #8246, which should basically eliminate the timeouts you reported recently on #8199 (there may still be an occasional timeout, but it should not be a regular occurrence).

Johan-Dahlberg, 2020-03-16:
@pabender I tested building from the master branch, which I found included #8246 .

For XpressNet turnouts, this fixes the timeout problem completely!

For LocoNet turnouts, the timeout still remains as shown below where LN1+LN2 are changed to closed manually right after each other:

Z21 traffic:
```
07:06:08.280:  LocoNet Tunnel Message: Requesting Switch at LT1 (LN1) to Closed (Output On).

07:06:08.339:  XpressNet Tunnel Reply: Turnout 1 State Thrown
07:06:08.375:  LocoNet Tunnel Message: Requesting Switch at LT1 (LN1) to Closed (Output Off).

07:06:10.376:  LocoNet Tunnel Message: Requesting Switch at LT2 (LN2) to Closed (Output On).

07:06:10.377:  XpressNet Tunnel Reply: Turnout 2 State Thrown
07:06:10.378:  LocoNet Tunnel Message: Requesting Switch at LT2 (LN2) to Closed (Output Off).
```
Message log:
```
     [java] 2020-03-16 07:06:10,375 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: 08 00 A2 00 B0 00 20 6F consecutive timeouts = 0 in z21.Z21TrafficController [z21.Z21TrafficController Transmit thread]
     [java] 2020-03-16 07:06:12,377 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: 08 00 A2 00 B0 01 20 6E consecutive timeouts = 0 in z21.Z21TrafficController [z21.Z21TrafficController Transmit thread]
```

Regards,
/Johan

cescp, 2020-07-27:
Hi,
I have experienced the same problem with a Roco z21 (white) and a Digikeijs DR4018 accessory decoder.

Some time ago I reported the problem:
https://groups.io/g/jmriusers/message/148423

According to "Figure 5 Example Sequence: Turnout switching" in page 29/54 of "Z21 LAN Protocol Specification"
(https://www.z21.eu/media/Kwc_Basic_DownloadTag_Component/root-en-main_47-1652-959-downloadTag-download/default/d559b9cf/1558675126/z21-lan-protokoll-en.pdf)
a response is expected from z21 when activating output, but no response is expected from z21 after deactivating output.
It seems that JMRI is expecting a response in both cases.

I have made some experiments using a Z21 WLANmaus (WiFi) while logging from JMRI:

a) when controlling a turnout from WLANmaus, there are no delays; I can move a turnout (closed, thrown, closed, ...) as quickly as I want.

b) the first time I start DecoderPro Z21 logs, they are more verbose than the next ones
(the sequence is: push closed button on WLANmaus; push thrown button on WLANmaus; push closed button on WLANmaus):

```
12:31:22.099: [09 00 40 00 43 00 04 02 45 08 00 A1 00 B0 04 30 7B]   XpressNet Tunnel Reply: Turnout 5 State Thrown
12:31:22.128: [08 00 40 00 42 01 80 C3]   XpressNet Tunnel Message: Accessory Decoder/Feedback Encoder Status Request: Base Address 1,Lower Nibble.
12:31:22.130: [08 00 40 00 42 01 06 45]   XpressNet Tunnel Reply: Feedback Response: Turnout without Feedback Turnout: 5 State: Thrown Right; Turnout: 6 State: Thrown Left
12:31:22.136: [08 00 40 00 43 00 04 47]   XpressNet Tunnel Message: Request Turnout 5 State
12:31:22.141: [09 00 40 00 43 00 04 02 45]   XpressNet Tunnel Reply: Turnout 5 State Thrown
12:31:22.170: [08 00 40 00 42 01 80 C3]   XpressNet Tunnel Message: Accessory Decoder/Feedback Encoder Status Request: Base Address 1,Lower Nibble.
12:31:22.172: [08 00 40 00 42 01 06 45]   XpressNet Tunnel Reply: Feedback Response: Turnout without Feedback Turnout: 5 State: Thrown Right; Turnout: 6 State: Thrown Left
12:31:22.182: [08 00 40 00 43 00 05 46]   XpressNet Tunnel Message: Request Turnout 6 State
12:31:22.186: [09 00 40 00 43 00 05 01 47]   XpressNet Tunnel Reply: Turnout 6 State Closed
12:31:22.257: [08 00 A1 00 B0 04 20 6B]   LocoNet Tunnel Tx: Requesting Switch at LT5 to Closed (Output Off).

12:31:38.376: [09 00 40 00 43 00 04 01 46 08 00 A1 00 B0 04 10 5B]   XpressNet Tunnel Reply: Turnout 5 State Closed
12:31:38.543: [08 00 A1 00 B0 04 00 4B]   LocoNet Tunnel Tx: Requesting Switch at LT5 to Thrown (Output Off).

12:31:54.301: [09 00 40 00 43 00 04 02 45 08 00 A1 00 B0 04 30 7B]   XpressNet Tunnel Reply: Turnout 5 State Thrown
12:31:54.522: [08 00 A1 00 B0 04 20 6B]   LocoNet Tunnel Tx: Requesting Switch at LT5 to Closed (Output Off).
```

c) if I start with an empty turnout table in JMRI, every time I use WLANmaus, some turnouts are automatically added:
- WLANmaus: T5 set to closed
- JMRI Turnout table: LT5, XT5, XT6 have been added

Is this behaviour correct?

I'm using JMRI v4.20 on a Linux computer.

Francesc

pabender, 2020-07-27:
> Hi,
> I have experienced the same problem with a Roco z21 (white) and a Digikeijs DR4018 accessory decoder.
>
> Some time ago I reported the problem:
> https://groups.io/g/jmriusers/message/148423 [...]

In JMRI, are you controlling the turnouts as XPressNet or LocoNet devices?  

XPressNet devices are definitely preferred for the Z21.  I would think it would be the XPressNet, since (if I recall correctly) the white Z21 doesn't have a physical LocoNet connection like the black Z21 does (I have a Black Z21)).

In the XPressNet case, there should always be a reply.  Sometimes the Z21 replies with a LocoNet message instead of an XPressNet message, but the documentation doesn't tell us when a LocoNet response will be made instead of an XPressNet one.
 
> I have made some experiments using a Z21 WLANmaus (WiFi) while logging from JMRI:
>
> a) when controlling a turnout from WLANmaus, there are no delays; I can move a turnout (closed, thrown, closed, [...]

We ask for the status of the turnout pair (5 and 6 in your log) when we create them.  We don't need to ask for the status after that point because the command station will update position even if we don't change it ourselves.  

> ```
> 12:31:22.099: [09 00 40 00 43 00 04 02 45 08 00 A1 00 B0 04 30 7B]   XpressNet Tunnel Reply: Turnout 5 State Thrown
> 12:31:22.128: [08 00 40 00 42 01 80 C3]   XpressNet Tunnel Message: Accessory [...]

Yes.

When you change the position of T5 on your WLANmaus, the Z21 sends both LocoNet and XPressNet messages indicating there was a change.  JMRI listens to those and creates the turnouts when it sees them.  The XPressNet feedback messages are always for a pair of turnouts, hence the reason you see XT5 and XT6.

Paul

cescp, 2020-07-27:
Hi,
You are right, z21 (white) does not have a physical LocoNet connector.

XNet turnouts only work the first time I use them. Afterwards, they can only be operated as LocoNet turnouts.
This is the sequence I folowed.

I create an XNet turnout XT5 (Turnout table, tab Z21XpressNet):

```
19:24:25.920: [08 00 40 00 42 01 80 C3]   XpressNet Tunnel Message: Accessory Decoder/Feedback Encoder Status Request: Base Address 1,Lower Nibble.
19:24:25.922: [08 00 40 00 42 01 0A 49]   XpressNet Tunnel Reply: Feedback Response: Turnout without Feedback Turnout: 5 State: Thrown Right; Turnout: 6 State: Thrown Right
19:24:25.926: [08 00 40 00 43 00 04 47]   XpressNet Tunnel Message: Request Turnout 5 State
19:24:25.926: [09 00 40 00 43 00 04 02 45]   XpressNet Tunnel Reply: Turnout 5 State Thrown
19:24:25.931: [08 00 40 00 42 01 80 C3]   XpressNet Tunnel Message: Accessory Decoder/Feedback Encoder Status Request: Base Address 1,Lower Nibble.
19:24:25.931: [08 00 40 00 42 01 0A 49]   XpressNet Tunnel Reply: Feedback Response: Turnout without Feedback Turnout: 5 State: Thrown Right; Turnout: 6 State: Thrown Right
19:24:25.934: [08 00 40 00 43 00 05 46]   XpressNet Tunnel Message: Request Turnout 6 State
19:24:25.935: [09 00 40 00 43 00 05 02 44]   XpressNet Tunnel Reply: Turnout 6 State Thrown
```

Then I click on its state button in Turnout Table, tab Z21XpressNet, to change its state:

```
19:24:32.997: [09 00 40 00 53 00 04 88 DF]   XpressNet Tunnel Message: Request Turnout 5 activate Output 0 with Queue false
19:24:32.998: [09 00 40 00 43 00 04 01 46 08 00 A1 00 B0 04 10 5B]   XpressNet Tunnel Reply: Turnout 5 State Closed
19:24:33.095: [08 00 40 00 52 01 80 D3]   XpressNet Tunnel Message: Accessory Decoder Operations Request: Turnout Address 5(Base Address 1,Sub Address 0) Turn Output 0 Off.
19:24:33.096: [08 00 A1 00 B0 04 00 4B]   LocoNet Tunnel Tx: Requesting Switch at LT5 to Thrown (Output Off).

```
On Z21LocoNet tab, a new turnout entry has been created: LT5 (as you mentioned, this is normal).

From now on, I cannot control the XT5: it does not respond to clicks on XT5 state button.
I can only control the turnout if I click on LT5 state button on Z21LocoNet tab.

Another question: on logs at 19:24:32.998, why LocoNet packet "08 00 A1 00 B0 04 10 5B" is not shown in a separate row?
It is not shown in Z21LocoNet Monitor either.

Francesc

PR #8876 (2020-08-05): use the right off message
fixes #7713
closed 2020-08-05 by bobjacobsen