JMRI/JMRI #7724: Create Test Release 4.19.1

bobjacobsen, 2019-12-15:
This is the first release in the 4.19 cycle. 

It is being made via a new process.  See [this jmri-developers thread](https://jmri-developers.groups.io/g/jmri/topic/68643252) for background info.

Developers, when you create PRs that you want included here (4.19.1, i.e. after the 4.17/4.18 cycle) please create the PR against the `development` branch instead of the `master` branch.  Until the 4.18 production release is out, please either mark these PRs with the 4.19.1 release number and/or the "After Next Production Release" label so we know how to handle them:
 - The "After Next Production Release" label means they are not intended for 4.17/4.18
 - A 4.19.1 milestone means it's ready to be included in the development branch for 4.19.1.  

The 4.19.1 milestone implies "after the next production release", but it would be good to set that label. The "After Next Production Release" label without the 4.19.1 milestone means something is not ready to merge for 4.19.1 yet.

Maintainers, if you find a new PR that is against `master` but should be against the `development` branch, please
 - Use the edit button at the top-right of the PR page to change the branch 
 - Then close and reopen the PR with a message "Close and reopen to reset the CI branch"

bobjacobsen, 2019-12-15:
To establish a local `development` branch, should you need it, assuming you don't have one by that name:
```
git checkout --track origin/development
```

You can then access and update that the same way you use `master`, via `git merge` and `git pull` commands, etc.

rhwood, 2019-12-15:
So if developers have a queue of branches waiting that we’re based off master, should they expect to be able to simply rebase off development?

rhwood, 2019-12-15:
Please note that only people who can commit to JMRI/JMRI can set labels on PRs.

bobjacobsen, 2019-12-15:
No need to rebase, though you can if you want to.  Just set the PR to merge _to_ `development` instead of `master`.  Or we can set that for you, if the PR already exists.  C.f. PR #7709:

[image]

danielb987, 2019-12-15:
> No need to rebase, though you can if you want to. Just set the PR to merge _to_ `development` instead of `master`. Or we can set that for you, if the PR already exists. C.f. PR #7709:

I tried to do that on my PR, but it seems only maintainers can change which branch to merge to. If I click on the Edit button, I can only change the title.

https://github.com/JMRI/JMRI/pull/6463

bobjacobsen, 2019-12-15:
> If I click on the Edit button, I can only change the title.

Could you try making a new PR that targets the `development` branch, please?  

If new PRs can select `development`, I'm happy to fix up all the existing one.  I'm slowly working through those already.

danielb987, 2019-12-15:
Yes, creating a new PR that targets development works fine.

bobjacobsen, 2019-12-15:
Installation packages for the `development` branch are now [building in Jenkins](http://builds.jmri.org/jenkins/job/TestReleases/job/development/)

bobjacobsen, 2019-12-31:
The release-4.19.1 branch has been created.

Maintainers, please set the 4.9.2 milestone on pulls from now on, as that will be the next test release.

Jenkins has created files in the usual place.

bobjacobsen, 2020-01-01:
JMRI 4.19.1 has been released.

closed 2020-01-01 by bobjacobsen