JMRI/JMRI #7725: Last Operated time is in ISO8601 long form.

BitFiddler27, 2019-12-15:
JMRI version 4.17.7 running on Windows 10. A minor issue that has been present for the last several test releases is that the LastOperated timestamp as is displayed in the roster is currently being displayed in the ISO8601 long form. ie 2019-10-11T18:03:903+0000 rather than the short form 2019-10-11T18:03:90Z. I do not find any way I can control the display format. It used to display the short format.

Thank you and Merry Christmas
Peter Jerkewitz

rhwood, 2019-12-15:
It should be in your locale’s preferred time format.

BitFiddler27, 2019-12-17:
Thanks for the reply, but I do not believe I can control how JMRI puts a time stamp into an XML file.

Peter

rhwood, 2019-12-17:
The timestamp should always be an ISO date in the XML; the displayed date should match the preferred format for your Locale (as set either in your OS, or in JMRI if overriding the OS).

My comments are as much notes describing what should be happening so I know what a fix looks like.

BitFiddler27, 2019-12-17:
Randall,

You are correct that the DateTime format in an XML file is recommended to be in ISO8601 format, and both long and short forms are acceptable.
The timestamp stored in the XML file is under control of JMRI, I’m guessing that a  Java SimpleDateFormat class is most often used to format the date-Time. While the Locale of my Windows 10 system does specify how windows displays date and time, there is no requirement for an application to honor that specification. And unfortunately windows does not let me specify the ISO8601 format. There were significant changes in Java 8 in regards to timestamp handling and support of timestamps. It could be that if the LastOperated timestamp module were compiled using Java 8, or later, the behavior could have changed. While both long and short forms of ISO8601 specification are interchangeable and acceptable, the short form just takes up less column width.

Thank you and Merry Christmas Randall
Peter

rhwood, 2019-12-18:
I was mistaking the last operated time for the last modified time. I should have a fix that makes the last operated time appear the same as the last modified time in the roster tables.

rhwood, 2019-12-18:
This is turning out to be non-trivial to fix, so its going to be in 4.19.x.

BitFiddler27, 2019-12-18:
Thank you ever so much Randall for your efforts and time spent working on JMRI. I know the several of us will appreciate your efforts.

Merry Christmas and a very Happy New Year
Peter

BitFiddler27, 2021-03-30:
Folks,

This issue has not been fixed. Although minor, I suspect that lots of users are un-familiar with ISO8601 format, especially with the zone offset at the end. The bigger issue is the amount of real estate it takes on the display.

Thank you
Peter Jerkewitz

closed 2021-05-29 by stale[bot]